displaylink-rpm/displaylink-rpm #2: libebvdi.so cannot be found

sjoerdmullender, 2016-05-23:
I compiled and installed the latest displaylink driver using this package (I built the RPM myself using the provided sources), but the driver didn't work.  My external monitory wasn't found.  I then asked about this problem on the displaylink forum and got a very helpful reply.  See [1].
Per the reply I modified the spec file and added a line to the %install section:
cp evdi-%{version}/library/libevdi.so $RPM_BUILD_ROOT/usr/libexec/displaylink
With this change, the driver works and finds my external monitor.
(For a complete change, the comment about not copying libevdi.so should also be removed.)

[1] http://www.displaylink.org/forum/showthread.php?p=80215

hadess, 2016-05-24:
And thanks for testing again :)

Let me know if there's any more problems.

closed 2016-05-24 by hadess

sjoerdmullender, 2016-05-26:
This version works.  Thanks.

commit 9da43040 (2016-05-24): Add missing libevdi.so library

When we started using the upstream evdi sources instead of the binaries
shipped by DisplayLink, we said:
Don't copy libevdi.so, we compiled it from source

But forgot to actually install it from the evdi sources. Do that now.

Closes #2