displaylink-rpm/displaylink-rpm #4: Host RPMs/SRPMs

ssaavedra, 2016-05-31:
It would be nice if the RPMs or SRPMs could be downloaded from somewhere without having to build the RPM oneself.

(I mean, I have no problem with that, but it can be a hassle for a greater user base)

We can use GitHub's releases (you can see it also in my own fork of the repo), or we can just host here the SRPM and build the different RPMs using copr for each Fedora release. Although it seems we do not depend on anything else than `libdrm-devel` for building, so a single RPM might suffice, but not sure about that. `libdrm-devel` versions are scattered between 2.4.59 and 2.4.66 for most recent Fedora/CentOS-based releases.

oxyoptiadamo, 2016-06-18:
The best way to go about this is to use the Fedora copr service imho. I planned on doing this but never got around to it yet. My plan was actually to get the kernel driver upstream and re-implement the user space blob in mesa but I am not clear what is the story with that nsa user blob yet.. but that is another story :)

victoredwardocallaghan, 2016-06-18:
The best way to go about this is to use the Fedora copr service imho. I planned on doing this but never got around to it yet. My plan was actually to get the kernel driver upstream and re-implement the user space blob in mesa but I am not clear what is the story with that nsa user blob yet.. but that is another story :)

victoredwardocallaghan, 2016-06-18:
fyi - https://copr.fedorainfracloud.org/

hadess, 2016-06-18:
> My plan was actually to get the kernel driver upstream and re-implement the user space blob in mesa

That's actually the major part of the story if you want to host this on Fedora's COPR. You can only build in COPR packages that would be accepted by Fedora. Half of this package is proprietary, so no chance of getting it into Fedora proper, or COPR.

hogarthj, 2016-09-19:
However ... that's what RPM Fusion exists for. If the blob is still redistributable then they are likely to accept.

That would be the best home for them potentially ... unless/until DisplayLink get patches submitted into the relevant upstreams for kernel etc

gebidesign, 2016-09-20:
For someone who can't build themselves, check out this guy's blog, I've got my DL working on Fedora just because of him.
http://nothen.com.ar/en/support-for-displaylink-adapters-on-linux/

ssaavedra, 2016-09-28:
Would anyone help with getting this on RPMFusion, or building in Travis?

It seems that the license agreement may prevent us altogether from hosting the driver:

> 1.4. You may distribute the Software (other than the Open Source Elements) only (i) without
> modification, (ii) in an unobfuscated form which allows the Software to be identified, and (iii) subject
> to the terms and conditions of this EULA, a copy of which shall be included with any such
> distribution. You may use, copy, modify and distribute the Open Source Elements in accordance
> with the terms and conditions applicable to them.

We seem to be allowed to redistribute verbatim copies, so we can indeed host the src.rpm files, because the proprietary .zip fulfills the conditions.

However, I am not sure on whether the build process we are performing can be seen as "without modification", which would prevent us from legally hosting the files on rpmfusion, even though it seems that there is intention from DisplayLink Corp. for the community to provide support for other distros, as per http://support.displaylink.com/knowledgebase/articles/679060.

I honestly think companies should stop shooting at their feet by saying they would like this repackaging process to happen and then locking it out in their licensing terms.

waterfoul, 2016-10-11:
Since we are just running/repackaging their installer I think it should be fine, Best way to find out would be to post on the displaylink forums and ask

ssaavedra, 2016-10-11:
We should try to get written confirmation from DisplayLink Ltd. or, even better, have them explicitly grant so in the license agreement.

waterfoul, 2016-10-11:
I posted on the forums to see if we get any bites
http://displaylink.org/forum/showthread.php?p=81520#post81520

waterfoul, 2016-10-12:
Here is the reply from the displaylink folks

> The text of the license was written to ensure the driver can be repacked by other distros and redistribute. As long as the text of license file is replicated, and binaries are redistributed as provided in the .run package (unmodified), and it is still possible to identify what component is what and where it came from, the license terms are met IMO. Note however that the original package now has a much more sophisticated udev rules than the project you linked, so I'd recommend to stick to them, rather than rolling your own.
> 
> Cheers,
> Michal

So it appears that we are good to go

PR #8 (2016-10-17): Travis CI + 1.2.64
Bump downloaded version to 1.2.64
Add support for automatically downloading the .zip
Add travis CI support

Fixes #4 

For this to work you will need to enable travis ci for the project and either give me access to this repo or change the api_key entry on the travis file. Once all of that is setup any time you create a release in github it will look something like [this](https://github.com/waterfoul/displaylink-rpm/releases/tag/test8). The travis file is setup to allow easy adds of other versions/distros, they only need to have a docker container in dockerhub. Once you find  [... cut]
closed 2016-10-17 by ssaavedra