displaylink-rpm/displaylink-rpm #5: Current evdi fails to build on kernel >= 4.7

ssaavedra, 2016-09-05:
The currently released evdi version fails on kernels >=4.7 due to the drm interface having changed.

Updating to master evdi should solve the issue (see displaylink/evdi#32) while DisplayLink does not release a packaged update.

Steps to reproduce:
1. Update to kernel >=4.7 (in my case, 4.7.2-101.fc23)
2. dkms install evdi/1.1.65

Expected result:
evdi gets installed

Observed output:
dkms fails with:

```
DKMS make.log for evdi-1.1.65 for kernel 4.7.2-101.fc23.x86_64 (x86_64)
lun sep  5 15:42:51 CEST 2016
make KBUILD_VERBOSE=1 SUBDIRS=/var/lib/dkms/evdi/1.1.65/build SRCROOT=/var/lib/dkms/evdi/1.1.65/build CONFIG_MODULE_SIG= -C /lib/modules/4.7.2-101.fc23.x86_64/build modules
make[1]: Entering directory '/usr/src/kernels/4.7.2-101.fc23.x86_64'
[...]
[unimportant lines]
[...]
  gcc -Wp,-MD,/var/lib/dkms/evdi/1.1.65/build/.evdi_stats.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-redhat-linux/5.3.1/include -I./arch/x86/include -Iarch/x86/include/generated/uapi -Iarch/x86/include/generated  -Iinclude -I./arch/x86/include/uapi -Iarch/x86/include/generated/uapi -I./include/uapi -Iinclude/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -funit-at-a-time -maccumulate-outgoing-args -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_FXSAVEQ=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_CRC32=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -pipe -Wno-sign-compare -fno-asynchronous-unwind-tables -fno-delete-null-pointer-checks -O2 --param=allow-store-data-races=0 -Wframe-larger-than=2048 -fstack-protector-strong -Wno-unused-but-set-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fvar-tracking-assignments -g -pg -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -Werror=date-time -Werror=incompatible-pointer-types -DCC_HAVE_ASM_GOTO -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_stats"'  -DKBUILD_MODNAME='"evdi"' -c -o /var/lib/dkms/evdi/1.1.65/build/evdi_stats.o /var/lib/dkms/evdi/1.1.65/build/evdi_stats.c
/var/lib/dkms/evdi/1.1.65/build/evdi_main.c: In function ‘evdi_driver_unload’:
/var/lib/dkms/evdi/1.1.65/build/evdi_main.c:82:2: error: implicit declaration of function ‘drm_connector_unplug_all’ [-Werror=implicit-function-declaration]
  drm_connector_unplug_all(dev);
  ^
/var/lib/dkms/evdi/1.1.65/build/evdi_fb.c: In function ‘evdi_fb_user_fb_create’:
/var/lib/dkms/evdi/1.1.65/build/evdi_fb.c:503:30: error: passing argument 1 of ‘drm_gem_object_lookup’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  obj = drm_gem_object_lookup(dev, file, mode_cmd->handles[0]);
                              ^
In file included from /var/lib/dkms/evdi/1.1.65/build/evdi_drv.h:23:0,
                 from /var/lib/dkms/evdi/1.1.65/build/evdi_fb.c:22:
include/drm/drm_gem.h:241:24: note: expected ‘struct drm_file *’ but argument is of type ‘struct drm_device *’
 struct drm_gem_object *drm_gem_object_lookup(struct drm_file *filp, u32 handle);
                        ^
/var/lib/dkms/evdi/1.1.65/build/evdi_fb.c:503:35: warning: passing argument 2 of ‘drm_gem_object_lookup’ makes integer from pointer without a cast [-Wint-conversion]
  obj = drm_gem_object_lookup(dev, file, mode_cmd->handles[0]);
                                   ^
In file included from /var/lib/dkms/evdi/1.1.65/build/evdi_drv.h:23:0,
                 from /var/lib/dkms/evdi/1.1.65/build/evdi_fb.c:22:
include/drm/drm_gem.h:241:24: note: expected ‘u32 {aka unsigned int}’ but argument is of type ‘struct drm_file *’
 struct drm_gem_object *drm_gem_object_lookup(struct drm_file *filp, u32 handle);
                        ^
/var/lib/dkms/evdi/1.1.65/build/evdi_fb.c:503:8: error: too many arguments to function ‘drm_gem_object_lookup’
  obj = drm_gem_object_lookup(dev, file, mode_cmd->handles[0]);
        ^
In file included from /var/lib/dkms/evdi/1.1.65/build/evdi_drv.h:23:0,
                 from /var/lib/dkms/evdi/1.1.65/build/evdi_fb.c:22:
include/drm/drm_gem.h:241:24: note: declared here
 struct drm_gem_object *drm_gem_object_lookup(struct drm_file *filp, u32 handle);
                        ^
cc1: some warnings being treated as errors
scripts/Makefile.build:289: recipe for target '/var/lib/dkms/evdi/1.1.65/build/evdi_fb.o' failed
make[2]: *** [/var/lib/dkms/evdi/1.1.65/build/evdi_fb.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/evdi/1.1.65/build/evdi_gem.c: In function ‘evdi_gem_mmap’:
/var/lib/dkms/evdi/1.1.65/build/evdi_gem.c:211:30: error: passing argument 1 of ‘drm_gem_object_lookup’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  obj = drm_gem_object_lookup(dev, file, handle);
                              ^
In file included from /var/lib/dkms/evdi/1.1.65/build/evdi_drv.h:23:0,
                 from /var/lib/dkms/evdi/1.1.65/build/evdi_gem.c:11:
include/drm/drm_gem.h:241:24: note: expected ‘struct drm_file *’ but argument is of type ‘struct drm_device *’
 struct drm_gem_object *drm_gem_object_lookup(struct drm_file *filp, u32 handle);
                        ^
/var/lib/dkms/evdi/1.1.65/build/evdi_gem.c:211:35: warning: passing argument 2 of ‘drm_gem_object_lookup’ makes integer from pointer without a cast [-Wint-conversion]
  obj = drm_gem_object_lookup(dev, file, handle);
                                   ^
In file included from /var/lib/dkms/evdi/1.1.65/build/evdi_drv.h:23:0,
                 from /var/lib/dkms/evdi/1.1.65/build/evdi_gem.c:11:
include/drm/drm_gem.h:241:24: note: expected ‘u32 {aka unsigned int}’ but argument is of type ‘struct drm_file *’
 struct drm_gem_object *drm_gem_object_lookup(struct drm_file *filp, u32 handle);
                        ^
/var/lib/dkms/evdi/1.1.65/build/evdi_gem.c:211:8: error: too many arguments to function ‘drm_gem_object_lookup’
  obj = drm_gem_object_lookup(dev, file, handle);
        ^
In file included from /var/lib/dkms/evdi/1.1.65/build/evdi_drv.h:23:0,
                 from /var/lib/dkms/evdi/1.1.65/build/evdi_gem.c:11:
include/drm/drm_gem.h:241:24: note: declared here
 struct drm_gem_object *drm_gem_object_lookup(struct drm_file *filp, u32 handle);
                        ^
cc1: some warnings being treated as errors
scripts/Makefile.build:289: recipe for target '/var/lib/dkms/evdi/1.1.65/build/evdi_gem.o' failed
make[2]: *** [/var/lib/dkms/evdi/1.1.65/build/evdi_gem.o] Error 1
cc1: some warnings being treated as errors
scripts/Makefile.build:289: recipe for target '/var/lib/dkms/evdi/1.1.65/build/evdi_main.o' failed
make[2]: *** [/var/lib/dkms/evdi/1.1.65/build/evdi_main.o] Error 1
  if [ "-pg" = "-pg" ]; then if [ /var/lib/dkms/evdi/1.1.65/build/evdi_stats.o != "scripts/mod/empty.o" ]; then ./scripts/recordmcount  "/var/lib/dkms/evdi/1.1.65/build/evdi_stats.o"; fi; fi;
Makefile:1461: recipe for target '_module_/var/lib/dkms/evdi/1.1.65/build' failed
make[1]: *** [_module_/var/lib/dkms/evdi/1.1.65/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.7.2-101.fc23.x86_64'
Makefile:21: recipe for target 'all' failed
make: *** [all] Error 2

```

dswebbthg, 2016-09-21:
it looks this was solved in upstream, see:

https://github.com/DisplayLink/evdi/issues/32#issuecomment-236555337

ssaavedra, 2016-09-22:
We should update the package to release the newest DisplayLink driver (packaged as 1.2; but actually with released evdi 1.2.55 and daemon 1.2.68), which should officially support a fixed evdi.

I tried to update the package, but #6 needs more information before the merge. I am not really sure on the difference that may be between the different DisplayLinkManager.

Anyway, this issue should be closed as #6 gets merged.

hadess, 2016-09-27:
@ssaavedra #6 got merged, does that fix it?

ssaavedra, 2016-09-28:
Yes :smile:

closed 2016-09-28 by ssaavedra