displaylink-rpm/displaylink-rpm #9: LICENSE?

ssaavedra, 2016-10-14:
The code on this repo lacks a LICENSE, which should be an important thing to have.

I am not sure this is even copyright-able, and for what is worth, I send all my contributions to the Public Domain through CC0.

This is not about DisplayLink's license, but about the *packaging* scripts and metadata license.

hadess, 2016-10-14:
I would put my contributions under the same license as the Fedora spec files:
https://fedoraproject.org/wiki/Licensing:MIT#Modern_Style_with_sublicense

But we'd need to know whether @enothen thinks. The only original content left from his RPM files are the scriptlets.

enothen, 2016-10-14:
I agree on putting my bits under the same MIT license used by the Fedora spec files

waterfoul, 2016-10-15:
I like MIT for this too

chuma, 2016-10-18:
I agree with putting this under the MIT license as well.

commit 564697ae (2016-11-07): Add MIT License

That was agreed on issue #9 on GitHub.

GitHub: close #9
closed 2016-11-07 by ssaavedra