displaylink-rpm/displaylink-rpm #11: Issue connecting USB screen Lenovo LT1421

bowerth, 2016-11-05:
Hello,
many thanks for making the RPMs available. I managed to compile this from source but unfortunately have the same result as with the release RPMs:
as soon as I connect the USB screen, it shows an image, freezes, turns black and keeps re-connecting

I'm running on a Dell XPS 13 with kernel 4.8.4-200.fc24.x86_64

`dmesg | grep "usb 1-1"` returns

```
[18493.675392] evdi: [D] evdi_detect:72 Painter is disconnected
[18493.675403] evdi: [D] evdi_detect:72 Painter is disconnected
[18494.019604] usb 1-2: USB disconnect, device number 11
[18494.310695] usb 1-2: new high-speed USB device number 12 using xhci_hcd
[18494.466722] usb 1-2: Device not responding to setup address.
[18494.672663] usb 1-2: Device not responding to setup address.
[18494.880645] usb 1-2: device not accepting address 12, error -71
[18532.614273] usb 1-2: new high-speed USB device number 14 using xhci_hcd
[18532.770295] usb 1-2: device descriptor read/64, error -32
[18533.045073] usb 1-2: New USB device found, idVendor=17e9, idProduct=03e0
[18533.045075] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[18533.045077] usb 1-2: Product: Lenovo LT1421 wide
[18533.045078] usb 1-2: Manufacturer: DisplayLink
[18533.045078] usb 1-2: SerialNumber: 6V9CATN2
[18533.126862] evdi: [W] evdi_painter_disconnect:478 (dev=-1) An unknown connection to ffff8a9116073000 tries to close us
[18533.126863] evdi: [W] evdi_painter_disconnect:479  - ignoring
[18533.150553] evdi: [W] evdi_painter_disconnect:478 (dev=0) An unknown connection to ffff8a9116073000 tries to close us
[18533.150555] evdi: [W] evdi_painter_disconnect:479  - ignoring
[18533.418368] evdi: [D] evdi_painter_connect:448 (dev=-1) Connected with           (null)
[18533.418370] evdi: [D] evdi_detect:69 (dev=1) Painter is connected
[18533.418399] evdi: [D] evdi_painter_crtc_state_notify:374 (dev=1) Notifying crtc state: 3
[18533.420757] evdi: [D] evdi_detect:72 Painter is disconnected
[18533.420805] evdi: [D] evdi_detect:69 (dev=1) Painter is connected
[18533.420808] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff
[18533.525273] evdi: [D] evdi_painter_mode_changed_notify:391 (dev=1) Notifying mode changed: 1366x768@60; bpp 32; 
[18533.525274] evdi: [D] evdi_painter_mode_changed_notify:392 pixel format 875713112
[18533.525281] evdi: [D] evdi_painter_crtc_state_notify:374 (dev=1) Notifying crtc state: 0
[18533.525283] evdi: [D] evdi_painter_dpms_notify:361 (dev=1) Notifying dpms mode: 0
[18533.726488] evdi: [D] evdi_detect:72 Painter is disconnected
[18533.726501] evdi: [D] evdi_detect:69 (dev=1) Painter is connected
[18533.726503] evdi: [D] evdi_painter_get_edid_copy:192 (dev=1) 00 ff ff
[18534.103659] traps: SourceEngine_00[26798] general protection ip:4d57f4 sp:7f15c57fa5b0 error:0 in DisplayLinkManager[400000+66d000]
[18534.302793] evdi: [D] evdi_painter_disconnect:499 (dev=1) Disconnected from ffff8a9172f92200
[18534.302795] evdi: [D] evdi_detect:72 Painter is disconnected
[18534.305741] evdi: [D] evdi_detect:72 Painter is disconnected
[18534.305798] evdi: [D] evdi_detect:72 Painter is disconnected
[18534.329847] evdi: [D] evdi_painter_crtc_state_notify:374 (dev=-1) Notifying crtc state: 3
[18534.329849] evdi: [W] evdi_painter_send_crtc_state:266 Painter is not connected!
[18534.424560] evdi: [D] evdi_detect:72 Painter is disconnected
[18534.424574] evdi: [D] evdi_detect:72 Painter is disconnected
```

ssaavedra, 2016-11-05:
This seems more likely to be related to evdi than us. I'd invite you to
open an issue at https://github.com/evdi/evdi

Or is there any reason to think this is a packaging fault?

bowerth, 2016-11-05:
thanks for suggesting, closing here
https://github.com/DisplayLink/evdi/issues/56

closed 2016-11-05 by bowerth