displaylink-rpm/displaylink-rpm #13: libstdc++.so.6 dependency issues

kmongwa, 2017-01-02:
Running on a Lenovo X1 Carbon (gen 1) with Lenovo Thinkpad USB 3 dock

$ sudo rpm -i centos-7-displaylink-1.2.64-1.x86_64.rpm 
[sudo] password for xxx: 
error: Failed dependencies:
	libstdc++.so.6(CXXABI_1.3.8)(64bit) is needed by displaylink-1.2.64-1.x86_64
	libstdc++.so.6(GLIBCXX_3.4.20)(64bit) is needed by displaylink-1.2.64-1.x86_64
	libstdc++.so.6(GLIBCXX_3.4.21)(64bit) is needed by displaylink-1.2.64-1.x86_64

I am running CentOS 7.3 with kernel updated to 4.9
$ cat /etc/centos-release
CentOS Linux release 7.3.1611 (Core)
$ uname -r
4.9.0-1.el7.elrepo.x86_64

$ yum list | grep libstdc++
compat-libstdc++-33.i686                3.2.3-72.el7                   @base    
libstdc++.i686                          4.8.5-11.el7                   @base    
libstdc++.x86_64                        4.8.5-11.el7                   @anaconda
libstdc++-devel.x86_64                  4.8.5-11.el7                   @base    
libstdc++-static.x86_64                 4.8.5-11.el7                   @base    
compat-libstdc++-33.x86_64              3.2.3-72.el7                   base     
libstdc++-devel.i686                    4.8.5-11.el7                   base     
libstdc++-docs.x86_64                   4.8.5-11.el7                   base     
libstdc++-static.i686                   4.8.5-11.el7                   base

hadess, 2017-01-03:
Tried installing with `yum` instead of `rpm`?

kmongwa, 2017-01-03:
Yum cannot resolve those dependencies either.  Is there a specific repo with those packages that I should include?

hadess, 2017-01-03:
I have no idea how @ssaavedra generated those. They might not be installable if the proprietary binaries are built against newer versions.

ssaavedra, 2017-01-04:
I did not generate those. [Travis does](https://github.com/displaylink-rpm/displaylink-rpm/blob/master/.travis.yml).

I really have no clue on why does CentOS not work. @hadess maybe the libstdc++ compatibility is automatically picked up by the rpm linter from the binaries' ldd result?

hadess, 2017-01-04:
> maybe the libstdc++ compatibility is automatically picked up by the rpm linter from the binaries' ldd result?

That's exactly what it does. You'll need to get newer versions of the libstdc++ library into the package for them to work.

ssaavedra, 2017-01-04:
What do you mean, into the package?

I don't have the sources from DisplayLink, I can do nothing about the binary itself. Should we bundle an older stdlibc++ or what kind of workaround are you thinking of?

ssaavedra, 2017-01-05:
To be clear, does this mean that displaylink needs a newer libstdc++ than it is currently provided in centos repos?

What should be the desirable approach? Bundle the newer stdlib?

hadess, 2017-01-05:
> To be clear, does this mean that displaylink needs a newer libstdc++ than it is currently provided in centos repos?

Yes.

> What should be the desirable approach? Bundle the newer stdlib?

Yes as well. I'd go with bundling the new libstdc++ in `/usr/libexec/displaylink/` as we already have a library in there.

ssaavedra, 2017-01-06:
Do we need to fix the LD_LIBRARY_PATH for that binary (without affecting
the rest of the system), or does it get picked up automatically by being in
the same folder?

I don't really have much experience with overriding the dynamic loader.

Also, how do we avoid rpmlint from introducing the bundled dependency?

hadess, 2017-01-10:
> Do we need to fix the LD_LIBRARY_PATH for that binary (without affecting the rest of the system), or does it get picked up automatically by being in the same folder?

I thought that the launcher for `DisplayLinkManager` already did this to load libevdi.so, but turns out it does not.

The easiest is to add the libstdc++ library to the path already mentioned, and add a `Environment=LD_LIBRARY_PATH=/usr/libexec/displaylink` line to the .service file. This will tell the dynamic linker where to find the library.

> I don't really have much experience with overriding the dynamic loader.
>
> Also, how do we avoid rpmlint from introducing the bundled dependency?

I don't understand this. `rpmlint` "verifies" packages, it doesn't add dependencies. `rpmbuild` would have add both the Requires (which it already has) and the provides for that library.

ssaavedra, 2017-01-11:
>> Also, how do we avoid rpmlint from introducing the bundled dependency?
>
> I don't understand this. rpmlint "verifies" packages, it doesn't add dependencies. rpmbuild would have add both the Requires (which it already has) and the provides for that library.

I mean, in the specfile there is no "Requires:" specifying libstdc++,
but it is included as a dependency in the rpm anyway, so if we bundle
our own libstdc++, will rpmbuild be smart enough not to add it as a
dependency?

hadess, 2017-01-11:
rpmbuild would add both the Requires (which it already has) and the provides for that library.

ssaavedra, 2017-01-11:
But then wouldn't that be a problem if CentOS decided to upgrade their own libstdc++, or if another package like ours also did the same trick (we'd have two packages providing the same library version)?

hadess, 2017-01-11:
Pretty sure that's not in the plans. It's supposed to be ABI stable. Centos 8/RHEL8 will have a different version.

neithere, 2017-06-21:
Hi, any update on this? I'm stuck with CentOS 7 and I don't think it would make sense to wait until CentOS 8 :)

ssaavedra, 2017-06-21:
I don't know what's the best way to bundle libstdc++ whilst keeping only one spec file.

p1lgr1m, 2017-09-21:
I've just managed to get DisplayLink going on CentOS (7.4). I now have a laptop, with 2 external monitors connected via a DisplayLink USB3 dock.

Here's what I found:
- The last DisplayLink driver version (the Ubuntu one that is this project depends on) that is useable on CentOS is version 1.2. Versions 1.2.1 and higher all depend on a version of libstdc++ that doesn't come with CentOS (and I don't know if it's possible to somehow get the necessary version of libstdc++, and use that in isolation for the DisplayLink driver without messing up the rest of the system). DisplayLink driver 1.2 appears to be no longer available from the Displaylink website - but it comes bundled release 1.2.55-2 of this project. Therefore, https://github.com/displaylink-rpm/displaylink-rpm/releases/tag/v1.2.55-2 is the latest version that works with CentOS 7 .
- Initially, on the monitors connected to the DisplayLink dock, all I got was a blank screen with a mouse cursor. Could not get any background or window to show on those monitors. The issue was that apparently the driver of the laptop screen (Intel HD4000 graphics) was interfering with DisplayLink. This fix worked for me: https://github.com/AdnanHodzic/displaylink-debian/issues/68#issuecomment-301232674 .

knumskull, 2017-10-09:
@p1lgr1m Could you give a bit more insight, what you've did, to install the RPM and which version.
You mentioned, that v1.2.55-2 is working in CentOS-7.4. Are you talking about `centos-7-displaylink-1.2.55-2.x86_64.rpm`?

I tried to install this rpm on my rhel-7.3 system, but still getting dependency issue.
~~~
sudo rpm -ivh centos-7-displaylink-1.2.55-2.x86_64.rpm                                                                                                                                                                                     
error: Failed dependencies:
        dkms is needed by displaylink-1.2.55-2.x86_64
        kernel > 4.7 is needed by displaylink-1.2.55-2.x86_64
        kernel-devel > 4.7 is needed by displaylink-1.2.55-2.x86_64
        libstdc++.so.6(CXXABI_1.3.8)(64bit) is needed by displaylink-1.2.55-2.x86_64
        libstdc++.so.6(GLIBCXX_3.4.20)(64bit) is needed by displaylink-1.2.55-2.x86_64
        libstdc++.so.6(GLIBCXX_3.4.21)(64bit) is needed by displaylink-1.2.55-2.x86_64
~~~

neithere, 2017-10-09:
Just to confirm that package 1.2.55 would **not** install on CentOS 7.3:
```
$ lsb_release -d
Description:	CentOS Linux release 7.3.1611 (Core) 
$ uname -r
4.11.4-1.el7.elrepo.x86_64
$ sudo rpm -ivh centos-7-displaylink-1.2.55-2.x86_64.rpm 
error: Failed dependencies:
	libstdc++.so.6(CXXABI_1.3.8)(64bit) is needed by displaylink-1.2.55-2.x86_64
	libstdc++.so.6(GLIBCXX_3.4.20)(64bit) is needed by displaylink-1.2.55-2.x86_64
	libstdc++.so.6(GLIBCXX_3.4.21)(64bit) is needed by displaylink-1.2.55-2.x86_64
```
@p1lgr1m, are you sure that the package 1.2.55 corresponds to version <1.2.1 of the driver? It's a bit counter-intuitive.

p1lgr1m, 2017-10-09:
I admit that, in figuring this out, I didn't install the binary rpm. I ended up doing a manual install (and, since that worked, I figured that the binary install would work too - but maybe it doesn't).

I believe that this is what I did:

```
wget https://github.com/displaylink-rpm/displaylink-rpm/releases/download/v1.2.55-2/centos-7-displaylink-1.2.55-2.src.rpm
mkdir displaylink
cd displaylink
rpm2cpio ../centos-7-displaylink-1.2.55-2.src.rpm | cpio -idmv
unzip "DisplayLink USB Graphics Software for Ubuntu 1.2.zip"
./displaylink-driver-1.2.58.run 

```

knumskull, 2017-10-09:
~~~
sudo ./displaylink-driver-1.2.58.run
Verifying archive integrity... All good.
Uncompressing DisplayLink Linux Driver 1.2.58  100% 
DisplayLink Linux Software 1.2.58 install script called: install
Distribution discovered: "Red Hat Enterprise Linux Workstation release 7.3 (Maipo)"
Unsatisfied dependencies. Missing component: Kernel version 3.10.0-514.21.2.el7.x86_64 is too old. At least 3.14 is required.
This is a fatal error, cannot install DisplayLink Linux Software.
~~~

p1lgr1m, 2017-10-09:
You do need to upgrade your kernel... "At least 3.14 is required".

I'm running 3.18.44-20.el7.x86_64 . I believe I got it from here: https://cbs.centos.org/koji/buildinfo?buildID=13755

This should do it:
```
wget http://cbs.centos.org/kojifiles/packages/kernel/3.18.44/20.el7/x86_64/kernel-3.18.44-20.el7.x86_64.rpm
rpm -i kernel-3.18.44-20.el7.x86_64.rpm
````

You'll also need the kernel-devel package :

```
wget http://cbs.centos.org/kojifiles/packages/kernel/3.18.44/20.el7/x86_64/kernel-devel-3.18.44-20.el7.x86_64.rpm
rpm -i kernel-devel-3.18.44-20.el7.x86_64.rpm
```

knumskull, 2017-10-09:
damn ... this is something I can't do :(
Do you know, why this is needed?

p1lgr1m, 2017-10-09:
Sorry - no, I don't know why the kernel requirements for the driver are what they are.

The standard kernel that comes with CentOS (3.10) was first released in 2013... so I guess it's not unreasonable for external software to have a bit more up-to-date requirements.

hadess, 2017-10-10:
Centos 7 and RHEL7's kernel aren't new enough to support the latest versions of the evdi driver, so this might not ever work, but there's a patch below to make the package provide a newer libstdc++.

[displaylink-rpm-centos-patch.txt](https://github.com/displaylink-rpm/displaylink-rpm/files/1372513/displaylink-rpm-centos-patch.txt)

hadess, 2017-10-10:
I think that our best bet might be to split off the kernel module from the daemon, if it's possible at all to use a newer DisplayLinkManager with an older evdi, that which I do not know...

ACiDGRiM, 2017-10-15:
It could be possible to make the Centos 7 package dependant on the epel-kernel 4.13+ kernels. Not ideal, as they cause some selinux conflicts needing to run in permissive mode, or create the required policies by the end user/admin.

However getting displaylink support in Centos would be worth the additional effort to use the newer kernels.

hadess, 2017-10-15:
Something like this:
```
diff --git a/displaylink.spec b/displaylink.spec
index 676d814..33fdae2 100644
--- a/displaylink.spec
+++ b/displaylink.spec
@@ -1,5 +1,11 @@
 %global debug_package %{nil}
 
+%if 0%{?rhel}
+%kernel_pkg_name kernel-mt
+%else
+%kernel_pkg_name kernel
+%endif
+
 Name:          displaylink
 Version:       %{_version}
 Release:       %{_release}
@@ -19,7 +25,7 @@ Source7:      http://mirrors.kernel.org/ubuntu/pool/main/g/gcc-6/libstdc++6_6.3.0-12u
 ExclusiveArch: i386 x86_64
 
 BuildRequires: libdrm-devel
-Requires:      dkms, kernel > 4.7, kernel-devel > 4.7
+Requires:      dkms, %{kernel_pkg_name} > 4.7, %{kernel_pkg_name}-devel > 4.7
 
 %if 0%{?rhel}
 # Avoid provides/requires from private libraries
```

On the top of the patch above. Ideally, this would support both the `kernel-lt` and `kernel-mt` from elrepo.org. I'll let @ssaavedra take it from here.

Also, it would be useful for the CI to test-compile the driver.

pnenov, 2017-10-23:
i do not know what is the point to announce RPM for CentOs which is not working ...
kernel 4.13.8
libstdc++ 4.8.5

yum install centos-7-displaylink-1.5.0-1.x86_64.rpm

Error: Package: displaylink-1.5.0-1.x86_64 (/centos-7-displaylink-1.5.0-1.x86_64)
           Requires: libstdc++.so.6(GLIBCXX_3.4.20)(64bit)
Error: Package: displaylink-1.5.0-1.x86_64 (/centos-7-displaylink-1.5.0-1.x86_64)
           Requires: libstdc++.so.6(CXXABI_1.3.8)(64bit)
Error: Package: displaylink-1.5.0-1.x86_64 (/centos-7-displaylink-1.5.0-1.x86_64)
           Requires: libstdc++.so.6(GLIBCXX_3.4.21)(64bit)

so basically this is not a RPM neither for Centos 7 nor for RedHat 7

hadess, 2017-10-23:
> i do not know what is the point to announce RPM for CentOs which is not working ...

Because it was working for a period of time. Maybe you'd like to finish fixing the problems I mentioned above. It would be much more useful than this sort of comment.

ssaavedra, 2017-10-23:
I'm sorry, I couldn't fix time to catch up with this issue last weekend, I'll try next weekend to allocate some time for this. If anyone else cares to test the patches hadess compiled into a PR, Travis can test it in the meanwhile.

adamel, 2017-10-30:
You cannot just drop in the libstdc++ library, you also need the libc it depends on, and the libc needs the correct version of the ld-linux dynamic linker, and that requires that the hardcoded paths in the libraries are modified to point to the ld-linux included in the RPM. Here is a patch starting that work (that goes on top of @hadess displaylink-rpm-centos-patch.txt) which makes DisplayLinkManager crash with a segmentation fault instead of complaining about missing dependencies. I don't have time right now to debug this any further, but feel free to continue the work. You might also get some inspiration from https://github.com/adamel/chrome-deps/blob/master/chrome-deps.spec

```
diff --git a/Makefile b/Makefile
index 16b982a..5f7a546 100644
--- a/Makefile
+++ b/Makefile
@@ -16,6 +16,8 @@ EVDI_PKG   := v$(VERSION).tar.gz
 SPEC_FILE  := displaylink.spec
 LIBSTDCXX_I386 := libstdc++6_6.3.0-12ubuntu2_i386.deb
 LIBSTDCXX_X86_64 := libstdc++6_6.3.0-12ubuntu2_amd64.deb
+LIBC_I386 := libc6_2.21-0ubuntu4_i386.deb
+LIBC_X86_64 := libc6_2.21-0ubuntu4_amd64.deb
 
 # The following is a little clunky, but we need to ensure the resulting
 # tarball expands the same way as the upstream tarball
@@ -23,7 +25,7 @@ EVDI_DEVEL_BRANCH   := devel
 EVDI_DEVEL_BASE_DIR := /var/tmp
 EVDI_DEVEL          := $(EVDI_DEVEL_BASE_DIR)/evdi-$(VERSION)
 
-BUILD_DEPS := $(DAEMON_PKG) $(EVDI_PKG) $(LIBSTDCXX_I386) $(LIBSTDCXX_X86_64) $(SPEC_FILE)
+BUILD_DEPS := $(DAEMON_PKG) $(EVDI_PKG) $(LIBSTDCXX_I386) $(LIBSTDCXX_X86_64) $(LIBC_I386) $(LIBC_X86_64) $(SPEC_FILE)
 
 #
 # Targets
@@ -121,6 +123,12 @@ $(LIBSTDCXX_X86_64):
 $(LIBSTDCXX_I386):
 	wget http://mirrors.kernel.org/ubuntu/pool/main/g/gcc-6/libstdc++6_6.3.0-12ubuntu2_i386.deb
 
+$(LIBC_X86_64):
+	wget https://mirrors.kernel.org/ubuntu/pool/main/g/glibc/libc6_2.21-0ubuntu4_amd64.deb
+
+$(LIBC_I386):
+	wget  https://mirrors.kernel.org/ubuntu/pool/main/g/glibc/libc6_2.21-0ubuntu4_i386.deb
+
 BUILD_DEFINES =                                                     \
     --define "_topdir `pwd`"                                        \
     --define "_sourcedir `pwd`"                                     \
diff --git a/displaylink.spec b/displaylink.spec
index 3315a89..a30060b 100644
--- a/displaylink.spec
+++ b/displaylink.spec
@@ -16,6 +16,8 @@ Source4:	DisplayLink USB Graphics Software for Ubuntu %{_daemon_version}.zip
 Source5:	20-displaylink.conf
 Source6:	http://mirrors.kernel.org/ubuntu/pool/main/g/gcc-6/libstdc++6_6.3.0-12ubuntu2_i386.deb
 Source7:	http://mirrors.kernel.org/ubuntu/pool/main/g/gcc-6/libstdc++6_6.3.0-12ubuntu2_amd64.deb
+Source8:	https://mirrors.kernel.org/ubuntu/pool/main/g/glibc/libc6_2.21-0ubuntu4_i386.deb
+Source9:	https://mirrors.kernel.org/ubuntu/pool/main/g/glibc/libc6_2.21-0ubuntu4_amd64.deb
 ExclusiveArch:	i386 x86_64
 
 BuildRequires:	libdrm-devel
@@ -93,6 +95,23 @@ tar xvJ --strip-components=4 -C $RPM_BUILD_ROOT/usr/libexec/displaylink/ -f data
 	./usr/lib/*/libstdc++.so.6.0.22
 mv $RPM_BUILD_ROOT/usr/libexec/displaylink/libstdc++.so.6.0.22 \
 	$RPM_BUILD_ROOT/usr/libexec/displaylink/libstdc++.so.6
+%ifarch %ix86
+ar x %{SOURCE8}
+%endif
+%ifarch x86_64
+ar x %{SOURCE9}
+%endif
+tar xv --strip-components=3 -C $RPM_BUILD_ROOT/usr/libexec/displaylink/ -f data.tar.gz \
+	./lib/*/libc.so.6* \
+	./lib/*/libc-2* \
+	./lib/*/ld-2*
+%ifarch x86_64
+mv $RPM_BUILD_ROOT/usr/libexec/displaylink/ld-2.*.so \
+	$RPM_BUILD_ROOT/usr/libexec/displaylink/ld-linux-x86-64.so.2
+%else
+mv $RPM_BUILD_ROOT/usr/libexec/displaylink/ld-2.*.so \
+	$RPM_BUILD_ROOT/usr/libexec/displaylink/ld-linux.so.2
+%endif
 %endif
 
 # Firmwares

```

pnenov, 2017-10-31:
Guys,
excuse my frustration but we are kind of tired listening how busy you are ... all of you 
why don't you just remove this faulty Centos/RedHat RPM, at least the people will not lose hours to update their distros looking to find out why your RPM does not work ....

ssaavedra, 2017-11-25:
That is ok. Can someone try and reply here which is the latest working version for CentOS here? I can remove the versions after that one.

Sorry for not replying back before, but you're right pnenov, we can at least let's not waste everybody's time.

p1lgr1m, 2017-11-25:
@ssaavedra : At this moment, my understanding is that **none** of the builds work with a **stock** CentOS7 distro (not even with the latest version, CentOS 7.4). For anyone that is able to manually upgrade their CentOS7 kernel to (at least) 3.14 , then things can be made to work with version 1.2.55-2. That's what I did (although I used the source RPM to do a manual install. I have not checked whether the binary RPM works).

ssaavedra, 2018-01-19:
I've removed all newer CentOS releases manually from the releases section in GitHub. I've commented CentOS for the while in the travis file so the next versions should not build that by default until we have tackled this.

ACiDGRiM, 2018-01-19:
Personally I've given up on Centos, This works perfectly in Fedora 27.
I'm going to be curious about the release, but maybe RHEL/Centos 8 will be a solution :|

khughitt, 2018-02-01:
@p1lgr1m Did you have to do anything after installing the kernel module with `displaylink-driver-1.2.58.run`? 

Using your suggested steps, I've been able to get the evdi module installed and running with kernel 4.4.114:

```
$ lsmod | grep evdi
evdi                   45056  5 
drm_kms_helper        135168  2 i915,evdi
drm                   356352  8 i915,evdi,drm_kms_helper
syscopyarea            16384  2 evdi,drm_kms_helper
sysfillrect            16384  2 evdi,drm_kms_helper
sysimgblt              16384  2 evdi,drm_kms_helper
```

However, the monitors are still not recognized in xrandr, etc.

Attempting to run DisplayLinkManager as root throws a couple errors, but nothing else happens:

```
./DisplayLinkManager 
[libevdi] ioctl: drop_master error=-1
[libevdi] ioctl: drop_master error=-1
```

Unplugging and replugging the usb-c connection briefly sends a signal to the connected monitors (screen flashes black and then goes back to "no signal detected").

Any thoughts?

p1lgr1m, 2018-02-02:
I don't think those errors you're seeing are anything significant. The same error shows up in my log files, but everything seems to work fine.

A few things come to mind:
1) Have you tried manually executing the following commands (as root)?
```
xrandr --setprovideroutputsource 1 0
xrandr --setprovideroutputsource 2 0
```

2) Do you have the framebuffer driver for X11 loaded?
```
yum list installed | grep fbdev
xorg-x11-drv-fbdev.x86_64             0.4.3-25.el7             @base
```

3) I had to disable acceleration on my existing Intel video card in order to get usable output on the Displaylink monitors (although I don't quite understand why an acceleration setting on 1 videocard should affect output on a different video device - but this was definitely the case)
```
cat /etc/X11/xorg.conf.d/05-display.conf 
Section "Device"
	Identifier "Intel"
	Driver "intel"
#	Option "AccelMethod" "uxa"
EndSection
```

khughitt, 2018-02-02:
Thanks for the suggestions!

I do have the framebuffer drivers installed:

```
yum list installed | grep fbdev
xorg-x11-drv-fbdev.x86_64         0.4.3-25.el7           @base 
```

The xrandr calls, however, result in the following errors:

```
xrandr --setprovideroutputsource 1 0
xrandr --setprovideroutputsource 2 0
```

This led me to look into Xorg.0.log, where I found a potentially problematic errors:

```
[    28.117] (EE) open /dev/dri/card0: No such file or directory
[    28.117] (WW) Falling back to old probe method for modesetting
[    28.117] (EE) open /dev/dri/card0: No such file or directory
[    28.117] (II) Loading sub module "fbdevhw"
[    28.117] (II) LoadModule: "fbdevhw"
[    28.118] (II) Loading /usr/lib64/xorg/modules/libfbdevhw.so
[    28.118] (II) Module fbdevhw: vendor="X.Org Foundation"
[    28.118]    compiled for 1.19.3, module version = 0.0.2
[    28.118]    ABI class: X.Org Video Driver, version 23.0
[    28.118] (EE) open /dev/fb0: No such file or directory
[    28.118] (WW) Falling back to old probe method for fbdev
[    28.118] (II) Loading sub module "fbdevhw"
[    28.118] (II) LoadModule: "fbdevhw"
[    28.118] (II) Loading /usr/lib64/xorg/modules/libfbdevhw.so
[    28.118] (II) Module fbdevhw: vendor="X.Org Foundation"
[    28.118]    compiled for 1.19.3, module version = 0.0.2
[    28.118]    ABI class: X.Org Video Driver, version 23.0
[    28.118] (EE) open /dev/fb0: No such file or directory
[    28.118] (EE) Screen 0 deleted because of no matching config section.
[    28.118] (II) UnloadModule: "modesetting"
[    28.118] (EE) Screen 0 deleted because of no matching config section.
[    28.118] (II) UnloadModule: "fbdev"
[    28.118] (II) UnloadSubModule: "fbdevhw"
...
[    28.847] (II) AIGLX: Screen 0 is not DRI2 capable
[    28.847] (EE) AIGLX: reverting to software rendering
[    28.958] (II) IGLX: enabled GLX_MESA_copy_sub_buffer
[    28.958] (II) IGLX: Loaded and initialized swrast
[    28.958] (II) GLX: Initialized DRISWRAST GL provider for screen 0
...
```

Have you made any changes to your Xorg conf? So far, I haven't touched anything with my Xorg.conf, and the only config in `/etc/X11/xorg.conf.d/` are some keyboard-related settings.

I noticed displaylink also ships with some Xorg settings (20-displaylink.conf), but I haven't tried using those just yet.

I'm going to start looking into the X boot errors, but if you have any suggestions there as well, feel free to share.

Few more potentially relevant pieces of information:

- Dell Latitude 7480
- Intel® Core™ i7-7600U CPU @ 2.80GHz × 4 
- Dell D6000 dock
- 4.4.114-1.el7.elrepo.x86_64

```
$ rpm -qa | grep drv-intel
xorg-x11-drv-intel-2.99.917-26.20160929.el7.x86_64
```

p1lgr1m, 2018-02-02:
I don't have an 'xorg.conf' per se... I have this:
```ll /etc/X11/xorg.conf.d/
total 8
-rw-r--r--. 1 root root 232 Sep 19  2015 00-keyboard.conf
-rw-r--r--. 1 root root  94 Sep 20 18:42 05-display.conf
```

I posted the contents of my 05-display.conf file earlier. And no, I didn't make any other manual changes.

My /dev/dri directory looks like this:
```ll /dev/dri
total 0
crw-rw----+ 1 root video 226,   0 Feb  2 09:07 card0
crw-rw----+ 1 root video 226,   1 Feb  2 09:10 card1
crw-rw----+ 1 root video 226,   2 Feb  2 09:10 card2
crw-------. 1 root video 226,  64 Feb  2 09:07 controlD64
crw-------. 1 root video 226,  65 Feb  2 09:09 controlD65
crw-------. 1 root video 226,  66 Feb  2 09:10 controlD66
crw-rw----+ 1 root video 226, 128 Feb  2 09:07 renderD128
```

If I'm not mistaken, /dev/dri should be populated by the kernel. Perhaps your kernel is too new, and/or missing some options/modules ?

zll11111, 2018-05-31:
@p1lgr1m I have followed all your steps ,including upgrade the kernel ,devel,and even the headers to 3.18.44-20.el7 on my CentOS 7 laptop.But when I ran the displaylink-driver-1.2.58.run ,it still showed a fatal error indicating missing linux headers coresponding to the kernel, which I have installed.
$`uname -r`
`3.18.44-20.el7.x86_64`
`$ rpm -qa kernel-devel`kernel-devel-3.10.0-862.3.2.el7.x86_64
`kernel-devel-3.10.0-693.2.2.el7.x86_64`
`kernel-devel-3.10.0-693.21.1.el7.x86_64`
`kernel-devel-3.10.0-693.5.2.el7.x86_64`
`kernel-devel-3.10.0-693.11.1.el7.x86_64`
`kernel-devel-3.18.44-20.el7.x86_64`
`$ rpm -qa kernel-headers`
`kernel-headers-3.18.44-20.el7.x86_64`
`$ sudo ./displaylink-driver-1.2.58.run `
`Verifying archive integrity... All good.`
`Uncompressing DisplayLink Linux Driver 1.2.58  100%  `
`DisplayLink Linux Software 1.2.58 install script called: install`
`Distribution discovered: "CentOS Linux release 7.4.1708 (Core) "`
`Unsatisfied dependencies. Missing component: Linux headers for running kernel, 3.18.44-20.el7.x86_64.`
`This is a fatal error, cannot install DisplayLink Linux Software.`
Didn't that happen to you when installing ? It's really frustrating.Please help me out,thanks.

p1lgr1m, 2018-05-31:
I believe I had that problem too.

The installer expects a "build" subdirectory to exist in the /lib/modules/<kernel version> directory. You can just put a symlink there.

I see that I also have a symlink named "source" that points to the same place. I don't remember whether that was critical or not.

Make sure you have symlinks set up similar to this:
```
[root@e430 3.18.44-20.el7.x86_64]# pwd
/lib/modules/3.18.44-20.el7.x86_64
[root@e430 3.18.44-20.el7.x86_64]# ls -alF | grep ".->"
lrwxrwxrwx.  1 root root     49 Sep 20  2017 build -> ../../../../usr/src/kernels/3.18.44-20.el7.x86_64/
lrwxrwxrwx.  1 root root      5 Sep 20  2017 source -> build/
```

zll11111, 2018-05-31:
@p1lgr1m After creating symlinks as you advised, the installation is completed. After editing the xorg.conf following [AdnanHodzic/displaylink-debian/issues/68#issuecomment-301232674](https://github.com/AdnanHodzic/displaylink-debian/issues/68#issuecomment-301232674) and reboot,all things work.Thanks a lot.That make all my efforts worthwhile.

There are 2 more issues after that.
First,now my laptop connects an outer screen via my Dell 3000 docker.It works well except the mouse cursor blink all the time. I  doubt it's because of some hardware performance problems.Is there any configurations to make it better? 
The second issue is not a big deal.When I boot my laptop with connecting docker,it really costs a few minutes to boot.The progress bar goes very slow. But when reboot the laptop with unplugging the docker,it's very quick. It just a small defect to mention.

ssaavedra, 2018-06-18:
In this case, do you think it be more maintainable to support CentOS in the same specfile or in a different one? For building through Travis it's mostly the same thing, but if we want to support building through copr, it may be more of an issue.

CooperGerman, 2018-09-22:
Hi guys, i'm trying to get my displaylink driver to work with a D6000 dock on a Inspiron15 700 series laptop on which i'm running Centos 7 with an  updated kernel at 4.18. Unfortunately i have only been able to get the unique HDMI to somewhat work by installing the latest centos7 rpm with the --nodeps option... I'm now trying to get something to work as i plugged both monitors into displayports. 
I am wondering if the issue with centos7 still is a thing as i apparently run into the same kind of problems mentioned above : 
    - libstdc++ not new enough 
    - rpm telling me my kernel is not > 4.7 although i've 4.18 ?
Can anyone tell me if i'm wasting my time trying to get this to work or if i should continue trying ?
Is there any release i could use ? (i feel like non working centos 7 releases have not been removed ?)
I'm fairly new to linux and i feel like i'm trying out some black magic everytime i see someone explaining how to do.
Thanks a lot for your responses.

ssaavedra, 2018-09-28:
@CooperGerman I don't have a setup that can replicate yours, but the currently reported state is that our packages don't work as expected on centos 7. If running centos is a must, then you are probably out of luck until someone with a running centos 7 machine can work out the linking issues there. We can't just "paste the displaylink userland binary" and expect it to work, since it is a dynamically-linked ELF and we need the appropriate linked libraries in the system. Otherwise it's not going to work. If displaylink was a static binary it would be easier, but that's not on our hand.

p1lgr1m, 2018-09-28:
CentOS 7 comes with a version of libstdc++ that is quite old. The latest version of the displaylink driver that works is 1.2 .

I haven't been to get any of the binary distributions to work. If you read through the earlier messages in this thread, you'll see how you can do a manual install of the driver on CentOS 7.

odicha, 2018-12-07:
Last displaylink version works on a vanilla CentOS 7.5, but it needs to be installed following some manual steps. Note that version required for CentOS is Ubuntu 14.04, because of libraries (well known problem with libstdc++)
First step: kernel LT needs to be installed and some additional packages installed
```
rpm --import https://www.elrepo.org/RPM-GPG-KEY-elrepo.org
rpm -Uvh https://www.elrepo.org/elrepo-release-7.0-3.el7.elrepo.noarch.rpm
yum --enablerepo=elrepo-kernel install kernel-lt kernel-lt-devel dkms libdrm-devel
```

After kernel-lt is installed, it can be enabled as default kernel in grub
```
grub2-set-default 0
grub2-mkconfig -o /boot/efi/EFI/centos/grub.cfg
```

After this, workstation needs to be rebooted. Kernel version can can be checked running `uname -r` . It should output something like "4.4.166-1.el7.elrepo.x86_64"
Now, drivers can be downloaded and installed
```
mkdir -p /tmp/displaylink
cd /tmp/displaylink
```
Download displaylink driver from https://www.displaylink.com/downloads/file?id=1261 and paste it in this folder
```
unzip DisplayLink\ USB\ Graphics\ Software\ for\ Ubuntu\ 4.4.zip
./displaylink-driver-4.4.24.run --keep --noexec
cd displaylink-driver-4.4.24/
./displaylink-installer.sh install
rm -rf /opt/displaylink/libusb*
cp x64-ubuntu-1404/DisplayLinkManager /opt/displaylink/DisplayLinkManager
```
*** Accept overwriting old file ***
```
dkms remove evdi/4.4.24 --all
cd /tmp/displaylink
wget https://github.com/DisplayLink/evdi/archive/v1.5.1.tar.gz
tar -zxvf v1.5.1.tar.gz
cd /tmp/displaylink/evdi-1.5.1/library/
make
cp libevdi.so /opt/displaylink/
```
*** Accept overwriting old file ***
 ```
mkdir -p /usr/src/evdi-1.5.1
cp /tmp/displaylink/evdi-1.5.1/module/* /usr/src/evdi-1.5.1
dkms install evdi/1.5.1
```
After this, another system restart needs to be done and docking is operational

ssaavedra, 2018-12-16:
If somebody has time to port this comment to an actual PR, we can make this an automatic process. With conditional statements on the specfile we can probably work this out, although the steps are quite different from Fedora.

pjwelsh, 2019-02-10:
A note to continue this information exchange, there is currently an issue with the kernels kernel-ml-4.19 and kernel-ml-4.20 building as @odicha describes. Kernel-ml-4.18 is still OK at this time. I'm not able to properly use kernel-lt-4.4 for a couple of reasons (servers can't boot when NVME drives are present and Dell 7480s loose wifi and network)

aaFn, 2019-03-13:
Hello,

Correct, I could apparently trace down the issue ito https://lkml.org/lkml/2018/9/11/481
=> there is a need to remove the flag FBINFO_CAN_FORCE_OUTPUT from the EVDI code.

Here is a modified set of instructions, according to that, based on @odicha ones:

Start by getting the right kernel from ELRepo if you do not have it already, and then the package he links to (https://www.displaylink.com/downloads/file?id=1261)

As user, where the file was downloaded:
```
	unzip DisplayLink\ USB\ Graphics\ Software\ for\ Ubuntu\ 4.4.zip
	./displaylink-driver-4.4.24.run --keep --noexec
	wget https://github.com/DisplayLink/evdi/archive/v1.5.1.tar.gz
	tar -zxvf v1.5.1.tar.gz
	cd evdi-1.5.1/library/
	make
```
Then modify ../../displaylink-driver-4.4.24/displaylink-installer.sh by **commenting out** (see `#`) a few lines.
They are not needed since they will fail and anyway we will replace EVDI by another version !
```
      echo "Building EVDI kernel module with DKMS"
      #dkms build evdi/$MODVER -q
      #if [ $? != 0 ]; then
      #  echo "Failed to build evdi/$MODVER. Consult /var/lib/dkms/evdi/$MODVER/build/make.log for details." > $ERRORS
      #  return 3
      #fi

      echo "Installing EVDI kernel module to kernel tree"
      #dkms install evdi/$MODVER -q
      #if [ $? != 0 ]; then
      #  echo "Failed to install evdi/$MODVER to the kernel tree." > $ERRORS
      #  return 4
      #fi
```
Now, since the FBINFO_CAN_FORCE_OUTPUT was dropped from 4.20 kernel, still as user, edit ../module/evdi_fb.c to remove FBINFO_CAN_FORCE_OUTPUT, by commenting it out also (using `/*   */`)
```
		info->flags = FBINFO_DEFAULT /*| FBINFO_CAN_FORCE_OUTPUT*/;
```
	
Then, as root:
```
	cd displaylink-driver-4.4.24/
	./displaylink-installer.sh install
	rm -rf /opt/displaylink/libusb*
	cp x64-ubuntu-1404/DisplayLinkManager /opt/displaylink/DisplayLinkManager
```
*** Accept overwriting old file ***
```
	cp ../evdi-1.5.1/library/libevdi.so /opt/displaylink/
```
*** Accept overwriting old file ***

Verify owner and group of files in /opt/displaylink/, they should be root root (may be needed in case like me you had a previous unsuccessful install attempt ..)

Continue as root:
```
	mkdir -p /usr/src/evdi-1.5.1
	cp ../evdi-1.5.1/module/* /usr/src/evdi-1.5.1
	dkms install evdi/1.5.1
```

and Reboot.

That should be it, at least that did it for me (at last !! I wish the CentOS package in https://github.com/displaylink-rpm/displaylink-rpm/releases/ would not require a higher version of GCC than what is really available in RHEL 7 at install time .... :-( )

Thank you, aaFn.

jryancon140, 2019-04-18:
I am using CentOS 7.6, and trying to follow aaFn's instructions above.  At this point, i get an error
cd evdi-1.5.1/library/
make
`cc -I../module -std=gnu99 -fPIC    -c -o evdi_lib.o evdi_lib.c
evdi_lib.c:5:24: fatal error: libdrm/drm.h: No such file or directory
 #include <libdrm/drm.h>
                        ^
compilation terminated.
make: *** [evdi_lib.o] Error 1
`
Any ideas?

pjwelsh, 2019-04-18:
The "yum" command can help you find these kinds of missing development packages for your build. As root or via sudo to search the package manager to find a pattern(I added the "*/" to your missing .h file):
yum provides */libdrm/drm.h

That should show something like:
libdrm-devel-2.4.91-3.el7.i686 : Direct Rendering Manager development package
Repo : base
Matched from:
Filename : /usr/include/libdrm/drm.h

So, just:
yum install libdrm-devel

You may be missing other dev files. But this process should help get most.
PJ

jryancon140, 2019-04-19:
@pjwelsh - Thanks, that did help me solve the error.

After this i decided to start all over with a clean install, of CentOS 7.6, and follow the instructions from @odicha.  The directions were great, and I was able to get all the way through them without any errors, but I am not getting any video out of the device.

I am using a Startech USB32VGAPRO, which has a DisplayLink DL-3100N.  There was a newer DisplayLink driver, but I choose to stay with the DisplayLink USB Graphics Software for Ubuntu 4.4.zip to follow odicha's instructions.

The kernel is updated, the device is detected, and the service is running without any errors.  Any suggestions?

`[root@localhost ~]# uname -r
4.4.178-1.el7.elrepo.x86_64`

`[root@localhost ~]# systemctl status dlm
  dlm.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/dlm.service; static; vendor preset: disabled)
   Active: active (running) since Thu 2019-05-30 09:30:12 PDT; 1min 57s ago
  Process: 11799 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/4.4.24 && modprobe evdi) (code=exited, status=0/SUCCESS)
 Main PID: 11820 (DisplayLinkMana)
    Tasks: 32
   CGroup: /system.slice/dlm.service
           └─11820 /opt/displaylink/DisplayLinkManager

May 30 09:30:12 localhost.localdomain systemd[1]: Starting DisplayLink Manager Service...
May 30 09:30:12 localhost.localdomain systemd[1]: Started DisplayLink Manager Service.
`

`[    2.220320] usb 2-3: new SuperSpeed USB device number 2 using xhci_hcd
[    2.231825] usb 2-3: New USB device found, idVendor=17e9, idProduct=4300
[    2.231828] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.231830] usb 2-3: Product: USB32VGAPRO
[    2.231832] usb 2-3: Manufacturer: DisplayLink
[    2.231834] usb 2-3: SerialNumber: 100101210019025`

`[  139.936806] evdi: loading out-of-tree module taints kernel.
[  139.941721] evdi: [I] Initialising logging on level 5
[  139.941725] evdi: [I] Atomic driver:no
[  141.647273] evdi: [D] evdi_add_devices:136 Increasing device count to 1
[  141.647640] evdi: [D] evdi_crtc_init:353 drm_crtc_init: 0
[  141.647811] evdi: [W] evdi_painter_crtc_state_notify:486 Painter does not exist!
[  141.647816] evdi: [D] evdi_detect:88 Painter is disconnected
[  141.647823] evdi evdi.0: No connectors reported connected with modes
[  141.647827] [drm] Cannot find any crtc or sizes - going 1024x768
[  141.649892] evdi evdi.0: fb1: evdidrmfb frame buffer device
[  141.649897] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[  141.649899] [drm] No driver support for vblank timestamp query.
[  141.649902] [drm] evdi: evdi_stats_init
[  141.649908] [drm] Initialized evdi 1.5.1 20180731 on minor 1
[  141.660349] evdi: [D] evdi_detect:88 Painter is disconnected
[  141.668151] open /dev/fb1 user=1 fb_info=ffff882fc15c1000 count=1
[  141.694075] evdi: [D] evdi_painter_connect:574 (dev=1) Connected with ffff882fbea94800
[  141.694079] evdi: [D] evdi_detect:85 (dev=1) Painter is connected
[  141.694115] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3
[  141.713918] mmap() framebuffer addr:18446683601024057344 size:3145728
[  141.958504] ip6_tables: (C) 2000-2006 Netfilter Core Team
[  142.142285] Ebtables v2.0 registered
[  142.442849] tun: Universal TUN/TAP device driver, 1.6
[  142.442853] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>
[  142.444032] device virbr0-nic entered promiscuous mode
[  142.679089] nf_conntrack version 0.5.0 (65536 buckets, 262144 max)
[  142.872776] virbr0: port 1(virbr0-nic) entered listening state
[  142.872793] virbr0: port 1(virbr0-nic) entered listening state
[  143.067908] virbr0: port 1(virbr0-nic) entered disabled state
[  144.075076] evdi: [D] evdi_detect:85 (dev=1) Painter is connected
[  144.075083] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
[  144.075192] evdi: [D] evdi_detect:85 (dev=1) Painter is connected
[  144.075194] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
[  144.075682] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3
[  144.080560] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3
[  144.210755] evdi: [D] evdi_detect:85 (dev=1) Painter is connected
[  144.210763] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
[  144.210923] evdi: [D] evdi_detect:85 (dev=1) Painter is connected
[  144.210928] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
[  144.211563] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3
[  144.213750] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3
[  154.063031] released /dev/fb1 user=1 count=0
[  163.905131] fuse init (API version 7.23)
`

pjwelsh, 2019-04-19:
I do not recall being able to use the kernel-lt (kernel 4.4.x) for some reason and changed to kernel-ml-4.18.16-1.el7.elrepo.x86 (and related).
PJ

	[ 139.936806] evdi: loading out-of-tree module taints kernel. [ 139.941721] evdi: [I] Initialising logging on level 5 [ 139.941725] evdi: [I] Atomic driver:no [ 141.647273] evdi: [D] evdi_add_devices:136 Increasing device count to 1 [ 141.647640] evdi: [D] evdi_crtc_init:353 drm_crtc_init: 0 [ 141.647811] evdi: [W] evdi_painter_crtc_state_notify:486 Painter does not exist! [ 141.647816] evdi: [D] evdi_detect:88 Painter is disconnected [ 141.647823] evdi evdi.0: No connectors reported connected with modes [ 141.647827] [drm] Cannot find any crtc or sizes - going 1024x768 [ 141.649892] evdi evdi.0: fb1: evdidrmfb frame buffer device [ 141.649897] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013). [ 141.649899] [drm] No driver support for vblank timestamp query. [ 141.649902] [drm] evdi: evdi_stats_init [ 141.649908] [drm] Initialized evdi 1.5.1 20180731 on minor 1 [ 141.660349] evdi: [D] evdi_detect:88 Painter is disconnected [ 141.668151] open /dev/fb1 user=1 fb_info=ffff882fc15c1000 count=1 [ 141.694075] evdi: [D] evdi_painter_connect:574 (dev=1) Connected with ffff882fbea94800 [ 141.694079] evdi: [D] evdi_detect:85 (dev=1) Painter is connected [ 141.694115] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3 [ 141.713918] mmap() framebuffer addr:18446683601024057344 size:3145728 [ 141.958504] ip6_tables: (C) 2000-2006 Netfilter Core Team [ 142.142285] Ebtables v2.0 registered [ 142.442849] tun: Universal TUN/TAP device driver, 1.6 [ 142.442853] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com (mailto:maxk@qualcomm.com)> [ 142.444032] device virbr0-nic entered promiscuous mode [ 142.679089] nf_conntrack version 0.5.0 (65536 buckets, 262144 max) [ 142.872776] virbr0: port 1(virbr0-nic) entered listening state [ 142.872793] virbr0: port 1(virbr0-nic) entered listening state [ 143.067908] virbr0: port 1(virbr0-nic) entered disabled state [ 144.075076] evdi: [D] evdi_detect:85 (dev=1) Painter is connected [ 144.075083] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff [ 144.075192] evdi: [D] evdi_detect:85 (dev=1) Painter is connected [ 144.075194] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff [ 144.075682] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3 [ 144.080560] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3 [ 144.210755] evdi: [D] evdi_detect:85 (dev=1) Painter is connected [ 144.210763] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff [ 144.210923] evdi: [D] evdi_detect:85 (dev=1) Painter is connected [ 144.210928] evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff [ 144.211563] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3 [ 144.213750] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3 [ 154.063031] released /dev/fb1 user=1 count=0 [ 163.905131] fuse init (API version 7.23)

jryancon140, 2019-04-20:
@pjwelsh from your previous post you stated "I'm not able to properly use kernel-lt-4.4 for a couple of reasons (servers can't boot when NVME drives are present and Dell 7480s loose wifi and network)".  I don't have those issues.  I couldn't find kernel-ml-4.18.16-1.el7.elrepo.x86, so if you have a link to it, let me know.

Am I reading this thread correctly that centos-7-displaylink-1.6.0-2.i386.rpm does not work and the manual steps from odicha and aaFn are used instead of this rpm, not used with it?

What DisplayLink chipsets have you guys gotten to work with the above manual steps from odicha?  I have devices with DL-3100N and DL-6950.

goofy79, 2019-07-16:
THX @odicha your solution work fine for me :)
But i can actual only use the DisplayLink Card, on my Laptop Dsiplay i only see the Startup Logs with
```
[OK] Started GNOME Display Manager
        Starting DisplayLink Manager Service ....
[OK] Started Virtualization daemon.
.
.
.

```
and my second Monitor won't show something.

Is there a possibility to use all 3 Monitors ? The Laptop Monito, the Monitor an my USB-C Connector AND the DisplayLink Port.
Actual i can use Laptop Monitor and USB-C Port Monitor (boot with an Kernel not equal 4.4) 
and if i use Kernel 4.4 with the changes of @odicha i only can work with the DisplayLink.

Thank you !

rbarakjsonar, 2020-06-24:
Thanks for the answer, Michael.
Seems that if, at all, I'm to make DisplayLink work on my CentOS 7, I'd need to build a more advanced kernel (4.4) than the default CentOS 7 kernel (3.10), which, as my machines are used to develop software, is not a possibility.

Thus, for those of us who're stuck with the kernel libraries that come with the CentOS installation - until DisplayLink driver will work with the Linux 3.10 kernel that comes with CentOS 7 - DisplayLink will not be working on CentOS 7.

sajididol, 2023-11-08:
> @ssaavedra : At this moment, my understanding is that **none** of the builds work with a **stock** CentOS7 distro (not even with the latest version, CentOS 7.4). For anyone that is able to manually [...]

Hi,
I was trying to install the mentioned version in Centos 7 kernel 6.3.3-1.el7.elrepo.x86_64. but we were stuck in installing EVDI. can you guide us which version of evdi need to be used for the version.

closed 2024-05-16 by elguero