displaylink-rpm/displaylink-rpm #15: Package evdi separately

rhofour, 2017-01-07:
Since EVDI is supposed to be a reusable kernel module it seems like it would make sense to package it separately. Would you accept a pull request to do this?

ssaavedra, 2017-01-09:
I thought about that. From a theoretical point of view I totally agree with
you, and if we were to ever include any of these packages in Fedora, evdi
could stand a chance of being included.

Although, it seems that in reality no such other use has yet been found, so
I think the argument is still weak.

The only obvious caveat I see is that you should make displaylink depend on
a specific evdi version and that could mean a more cumbersome update when
both packages are updated. That is much easier to do when you have a proper
repo, because dnf will pull the requirements and erase the old version
automatically, but when you are installing from downloaded rpm you have to
provide the paths for both packages.

Not a problem per se, anyway, but it's easier to tell the user to just
install one package (also, friendlier interface with packagekit-based
installs).

However, I'm quite open to the idea, I actually entertained that thought
sometime too, but I failed to see any obvious gain at this point. Of
course, a very interesting point could be that there was some other use of
evdi that should be supported :-)

This would also gain strength if we could get it included in a repo, let's
say RPMfusion, evdi going to free and displaylink going to non-free. If
that is of interest (and I think it could be) then splitting the package
seems the right thing to do.

hadess, 2017-01-09:
If evdi did indeed find another use, it would probably rather easily get included upstream, and then we wouldn't need to ship it as a kmod at all. In the meanwhile, nothing uses it, so the point is moot.

Feel free to ping again if you find another user for the kernel module.

closed 2017-01-09 by hadess

ssaavedra, 2017-01-09:
Related: https://github.com/DisplayLink/evdi/issues/25

rhofour, 2017-01-10:
I was actually just starting work on something that would use evdi which is what prompted me to point out it should probably be split. I can re-commented once I actually have something working.

I was also thinking about trying to get this into RPMFusion as @ssaavedra suggested. I've never maintained a package before, but I'd be willing to put some work in to make this easier to install since it's something I use.

hadess, 2017-01-10:
You're really better off trying to get the module upstream, this would make it easier for everyone.

ssaavedra, 2017-01-10:
Yes, once/if you actually have something open source that uses evdi, it
should be feasible to have it included in the kernel tree.

And DisplayLinkManager would belong to rpmfusion's non-free repo anyway, so
if there is a use case to release evdi independently, then, I agree with
hadess, that its proper place would be upstream.