displaylink-rpm/displaylink-rpm #18: fedora 25 wayland - Failed to start DisplayLink Manager Service.

KASTINpl, 2017-01-19:
```
$ uname -a
Linux 4.9.3-200.fc25.x86_64 #1 GNU/Linux
```

RPM "fedora-25-displaylink-1.2.64-1.x86_64.rpm" and "fedora-25-displaylink-1.2.64-1.src.rpm" downloaded from 
https://github.com/displaylink-rpm/displaylink-rpm/releases
* "fedora-25-displaylink-1.2.64-1.x86_64.rpm" cant be installed becauseof "10" return code, without any extra info
* "fedora-25-displaylink-1.2.64-1.src.rpm" installed successfully

```
$ systemctl status displaylink.service 
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: activating (auto-restart) (Result: exit-code) since czw 2017-01-19 20:49:10 CET; 1s ago
  Process: 5409 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)

sty 19 20:49:10 kastin-laptop systemd[1]: Failed to start DisplayLink Manager Service.
sty 19 20:49:10 kastin-laptop systemd[1]: displaylink.service: Unit entered failed state.
sty 19 20:49:10 kastin-laptop systemd[1]: displaylink.service: Failed with result 'exit-code'.
```

```
$ systemctl restart displaylink.service 
Job for displaylink.service failed because the control process exited with error code.
See "systemctl status displaylink.service" and "journalctl -xe" for details.
```

```
$ journalctl -xe
sty 19 20:51:01 kastin-laptop modprobe[5574]: modprobe: FATAL: Module evdi not found in directory /lib/modules/4.9.3-200.fc25.x86_64
sty 19 20:51:01 kastin-laptop systemd[1]: displaylink.service: Control process exited, code=exited status=1
sty 19 20:51:01 kastin-laptop systemd[1]: Failed to start DisplayLink Manager Service.
```

general fix for "Module evdi not found in directory" error is 
```
$ sudo /sbin/dkms install evdi/1.2.64

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j4 KERNELRELEASE=4.9.3-200.fc25.x86_64 all INCLUDEDIR=/lib/modules/4.9.3-200.fc25.x86_64/build/include KVERSION=4.9.3-200.fc25.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.9.3-200.fc25.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.2.64/build/make.log for more information.
```

after dig a bit deeper (cat  /var/lib/dkms/evdi/1.2.64/build/make.log)
```
echo >&2;                                                       \
echo >&2 "  ERROR: Kernel configuration is invalid.";           \
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";      \
echo >&2 ;
```

has someone DP driver installed on F25 successfully?

zetaomegagon, 2017-01-20:
Hey, going on this Stackoverflow thread:

https://stackoverflow.com/questions/23184181/error-while-running-make-install-include-generated-autoconf-h-or-include-confi

I figured that what you found wasn't the actual error. So I went and found an evdi driver source:

https://github.com/DisplayLink/evdi/issues/69

I did what the person having the error did, except I git clone'd the repo. Now the DisplayLink service is running without errors. I still don't have it working tho. If I do, I'll check back.

KASTINpl, 2017-01-20:
@zetaomegagon thanks a lot :) works fine now (except 1 error...)

```
$ wget https://github.com/DisplayLink/evdi/archive/793a65a8f45bc0ca890d8fa8507c0c6c4a5860f8.zip
$ cd evdi-*
$ make
$ cp -r ./module/ /usr/src/evdi-devel
$ dkms add evdi/devel
$ dkms build evdi/devel
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j4 KERNELRELEASE=4.9.3-200.fc25.x86_64 all INCLUDEDIR=/lib/modules/4.9.3-200.fc25.x86_64/build/include KVERSION=4.9.3-200.fc25.x86_64 DKMS_BUILD=1...
cleaning build area...(bad exit status: 2)

DKMS: build completed.

$ dkms install evdi/devel
$ modprobe evdi
$ systemctl restart displaylink && systemctl status displaylink
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: active (running) since pią 2017-01-20 17:04:04 CET; 5s ago
  Process: 6908 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 6911 (DisplayLinkMana)
    Tasks: 11 (limit: 4915)
   CGroup: /system.slice/displaylink.service
           └─6911 /usr/libexec/displaylink/DisplayLinkManager
```

so waiting for stable evdi version :)

zetaomegagon, 2017-01-29:
Do your displays work?

--Elijah Beale

Sent from a smallish computer ;)

KASTINpl, 2017-01-29:
unfortunatelly seems wayland is not supported yet by displaylink (or fedora and/or kernel modules dont spport DL). I used Xorg session to test and external display was detected but black-screen only, on wayland session no monitors detected. 

for now the only way to run my monitor via usb is to use ubuntu/linux mint - works fine then

zetaomegagon, 2017-01-29:
What kernel versions are you running on Ubuntu and Mint? I've only got
Fedora 25 @ 4.9.3-200 64bit machines right now.

Trying to confirm that displaylink <version> will only work (fully...I
guess) with corresponding evdi <version>.

--Elijah Beale

Sent from a smallish computer ;)

zetaomegagon, 2017-01-29:
nvm...this is what I was looking for-- second to last post:

https://github.com/DisplayLink/evdi/issues/74

Nice that all the DisplayLink/evdi info is all on *one* place... :/

KASTINpl, 2017-01-30:
just to be clear:
works on ubuntu/linux mint with:

`/var/lib/dkms/evdi/1.1.62`
```
~ $ uname -a
Linux 4.4.0-47-generic #68-Ubuntu x86_64 GNU/Linux
```

doesnt work on Fedora 25 xorg and wayland:
Linux 4.9.3-200.fc25.x86_64 #1 GNU/Linux + latest evdi build

zetaomegagon, 2017-01-30:
Yep. We have to wait for DisplayLink to release a new version of
displaylink that will work with the latest evdi for it to work on Fedora 25
and latest kernel w/ xorg.

No clue about Wayland, but I think they may be focused on Mir first...

--Elijah Beale

Sent from a smallish computer ;)

dilom, 2017-02-08:
I can confirm for me on F25 with kernel 4.9.7 but I had to manually type in 

`xrandr --addmode DVI-I-1 1920x1080 `

to get my monitor to work.  And I have to do it every single time.

zetaomegagon, 2017-02-08:
@dilom

Same here, but I cannot rotate my displays or set positioning.

--Elijah Beale

Sent from a smallish computer ;)

davidcastellani, 2017-02-12:
@zetaomegagon if xrandr works now to get a monitor to display, then you should be able to use xrandr to also set positioning and rotation, right?

zetaomegagon, 2017-02-12:
Should be able to, yes. However I can't get it to work for me-- using
xrandr or otherwise.

--Elijah Beale

Sent from a smallish computer ;)

dilom, 2017-03-01:
With the latest version under F25 wayland
`xrandr --addmode DVI-I-1 1920x1080`
no longer brings my monitor up.
Still not detected when plugged in either. Does anybody know if there is a way to make it work?

hadess, 2017-03-01:
There's no support for Wayland in the DisplayLink drivers, nothing specific to the package. I don't see how an xrandr call (which is a legacy call) can do anything.

See https://github.com/DisplayLink/evdi/issues/58

closed 2017-03-01 by hadess