displaylink-rpm/displaylink-rpm #22: What are the steps after install?

reypm, 2017-03-03:
I am running Fedora 25 and I've found this, I have installed the latest package by running `sudo dnf install <package>.rpm` my device is a TU3-DS2 and I have connected one monitor to the dock but I can't see anything on the monitor, do I need anything else rather than install the driver?

bowerth, 2017-03-03:
Hi - I'm using a Lenovo LT1421 and the following did the job for me. You
will need to figure out the optimum screen resolution for your display
first, probably best to check the vendor website.

create new mode
:   `$ cvt 1368 768 60`

add xrandr mode
:   `$ xrandr --newmode "1368x768_60.00"   85.25  1368 1440 1576 1784  768
771 781 798 -hsync +vsync`

select xrandr mode
:   `$ xrandr --addmode DVI-I-1-1 1368x768_60.00`

Apparently, `xrandr` is a legacy command that shouldn't be necessary but
for me it's the only way so far.

In case you aren't lucky, try switching from wayland to xorg when you log
in to gnome

<https://github.com/displaylink-rpm/displaylink-rpm/issues/22>, or mute the
thread

mmi-rperez, 2017-03-03:
@bowerth something is wrong with me, check details below:

```
$ loginctl show-session 2 -p Type
Type=x11

$ cvt 1920 1080 60
# 1920x1080 59.96 Hz (CVT 2.07M9) hsync: 67.16 kHz; pclk: 173.00 MHz
Modeline "1920x1080_60.00"  173.00  1920 2048 2248 2576  1080 1083 1088 1120 -hsync +vsync

$ xrandr --newmode "1920x1080_60.00"   85.25  1368 1440 1576 1784  768
xrandr: failed to parse '1920x1080_60.00' as a mode specification
Try 'xrandr --help' for more information.
```

Any ideas?

bowerth, 2017-03-04:
maybe try:

`xrandr --newmode "1920x1080_60.00"  173.00  1920 2048 2248 2576  1080
1083 1088 1120 -hsync +vsync`

i.e.

`xrandr --newmode [command returned from cvt w/o "Modeline "]`

mmi-rperez, 2017-03-04:
@bowerth first part worked fine, the last step didn't see below:

```
$ xrandr --newmode "1920x1080_60.00"  173.00  1920 2048 2248 2576  1080 1083 1088 1120 -hsync +vsync

$ xrandr --addmode DVI-I-1-1 1920x1080_60.00
xrandr: cannot find output "DVI-I-1-1"
```

how do I find the proper DVI output? **The TU3-DS2** is connected through USB port

reypm, 2017-03-04:
I was trying with the wrong RPM but now trying with the Fedora 25 one I got the following:
```
 rpm -ivh fedora-25-displaylink-1.3.43-2.x86_64.rpm
Preparing...                          ################################# [100%]
Updating / installing...
   1:displaylink-1.3.43-2             ################################# [100%]
warning: %post(displaylink-1.3.43-2.x86_64) scriptlet failed, exit status 1
```

Any help?

reypm, 2017-03-04:
Here is an extra information:

```
/var/log/boot.log
[FAILED] Failed to start DisplayLink Manager Service.
See 'systemctl status displaylink.service' for details.

$ systemctl status displaylink.service
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: activating (auto-restart) (Result: exit-code) since Fri 2017-03-03 21:31:17 EST; 1s ago
  Process: 5065 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)

Mar 03 21:31:17 localhost.mmi systemd[1]: Failed to start DisplayLink Manager Service.
Mar 03 21:31:17 localhost.mmi systemd[1]: displaylink.service: Unit entered failed state.
Mar 03 21:31:17 localhost.mmi systemd[1]: displaylink.service: Failed with result 'exit-code'.

[FAILED] Failed to start Builds and install new kernel modules through DKMS.
See 'systemctl status dkms.service' for details.
$ systemctl status dkms.service
● dkms.service - Builds and install new kernel modules through DKMS
   Loaded: loaded (/usr/lib/systemd/system/dkms.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2017-03-03 21:29:18 EST; 2min 21s ago
     Docs: man:dkms(8)
  Process: 1647 ExecStart=/bin/sh -c dkms autoinstall --verbose --kernelver $(uname -r) (code=exited, status=4)
 Main PID: 1647 (code=exited, status=4)

Mar 03 21:29:12 localhost.mmi systemd[1]: Starting Builds and install new kernel modules through DKMS...
Mar 03 21:29:18 localhost.mmi sh[1647]: Error! Could not locate dkms.conf file.
Mar 03 21:29:18 localhost.mmi sh[1647]: File:  does not exist.
Mar 03 21:29:18 localhost.mmi systemd[1]: dkms.service: Main process exited, code=exited, status=4/NOPERMISSION
Mar 03 21:29:18 localhost.mmi systemd[1]: Failed to start Builds and install new kernel modules through DKMS.
Mar 03 21:29:18 localhost.mmi systemd[1]: dkms.service: Unit entered failed state.
Mar 03 21:29:18 localhost.mmi systemd[1]: dkms.service: Failed with result 'exit-code'.
```

I am running Fedora 25 4.9.13-200.fc25.x86_64, any help?

bowerth, 2017-03-04:
it seems you have an issue with evdi - the command `modprobe evdi` should not return any error message.

reypm, 2017-03-05:
@bowerth here you have a full trace of what I did with the same result. First of all I have made some research and this links comes up:

- [Fedora support for DisplayLink adapters][1]
- http://mattwilmott.com/2016/11/03/how-to-install-the-displaylink-driver-fedora-2425/comment-page-1/
- https://github.com/displaylink-rpm/displaylink-rpm (this is your repository)
- https://github.com/DisplayLink/evdi/issues/64 (this is EVDI repository)

I am using the latest Updated F25 Live isos from: http://tinyurl.com/Live-respins2

- Check the kernel before update|upgrade:
```
$ uname -r
4.9.12-200.fc25.x86_64
```
- Update and reboot, restart using the latest kernel:
```
$ sudo dnf update -y && reboot
```
- Check the kernel after update|upgrade:
```
$ uname -r
4.9.13-200.fc25.x86_64
```
- Install needed libraries (as per this link: http://mattwilmott.com/2016/11/03/how-to-install-the-displaylink-driver-fedora-2425/comment-page-1/)
```
$ sudo dnf install kernel-headers dkms kernel-devel
...
Package kernel-headers-4.9.13-200.fc25.x86_64 is already installed, skipping.
...
Installing:
 dkms        	noarch	2.3-1.20161202gitde1dca9.fc25   	updates 	81 k
 kernel-devel	x86_64	4.9.13-200.fc25                 	updates 	11 M

Transaction Summary
================================================================================
Install  2 Packages
…
Installed:
  dkms.noarch 2.3-1.20161202gitde1dca9.fc25        	kernel-devel.x86_64 4.9.13-200.fc25
```
- Check SELinux status (I have disabled previously)
```
$ getenforce
Disabled
```
- Installing your RPM package:
```
# wget https://github.com/displaylink-rpm/displaylink-rpm/releases/download/v1.3.52/fedora-25-displaylink-1.3.43-2.x86_64.rpm
# ls -la
total 3180
drwxr-xr-x.  2 rperez rperez	4096 Mar  5 15:32 .
drwx------. 16 rperez rperez	4096 Mar  5 15:32 ..
-rw-rw-r--   1 rperez rperez 3244390 Feb 21 09:33 fedora-25-displaylink-1.3.43-2.x86_64.rpm
```
- Before installing, verify that there’s nothing hidden on the rpm.Basic information (Step 3 [here][1])
```
# rpm -qip fedora-25-displaylink-1.3.43-2.x86_64.rpm
Name    	: displaylink
Version 	: 1.3.43
Release 	: 2
Architecture: x86_64
Install Date: (not installed)
Group   	: User Interface/X Hardware Support
Size    	: 8950221
License 	: GPL v2.0, LGPL v2.1 and Proprietary
Signature   : (none)
Source RPM  : displaylink-1.3.43-2.src.rpm
Build Date  : Tue 21 Feb 2017 09:33:38 AM EST
Build Host  : 23cc5818b01c
Relocations : (not relocatable)
Summary 	: DisplayLink VGA/HDMI driver for DL-6xxx, DL-5xxx, DL-41xx and DL-3xxx adapters
Description :
This adds support for HDMI/VGA adapters built upon the DisplayLink DL-6xxx,
DL-5xxx, DL-41xx and DL-3xxx series of chipsets. This includes numerous
docking stations, USB monitors, and USB adapters.
```
- Files that the rpm deploys:
```
# rpm -qlp fedora-25-displaylink-1.3.43-2.x86_64.rpm
/etc/udev/rules.d/99-displaylink.rules
/usr/lib/systemd/system-sleep/displaylink.sh
/usr/lib/systemd/system/displaylink.service
/usr/libexec/displaylink
/usr/libexec/displaylink/DisplayLinkManager
/usr/libexec/displaylink/ella-dock-release.spkg
/usr/libexec/displaylink/firefly-monitor-release.spkg
/usr/libexec/displaylink/libevdi.so
/usr/share/doc/displaylink
/usr/share/doc/displaylink/LICENSE
/usr/src/evdi-1.3.43
/usr/src/evdi-1.3.43/Kconfig
/usr/src/evdi-1.3.43/LICENSE
/usr/src/evdi-1.3.43/Makefile
/usr/src/evdi-1.3.43/dkms.conf
/usr/src/evdi-1.3.43/evdi_connector.c
/usr/src/evdi-1.3.43/evdi_cursor.c
/usr/src/evdi-1.3.43/evdi_cursor.h
/usr/src/evdi-1.3.43/evdi_debug.c
/usr/src/evdi-1.3.43/evdi_debug.h
/usr/src/evdi-1.3.43/evdi_drm.h
/usr/src/evdi-1.3.43/evdi_drv.c
/usr/src/evdi-1.3.43/evdi_drv.h
/usr/src/evdi-1.3.43/evdi_encoder.c
/usr/src/evdi-1.3.43/evdi_fb.c
/usr/src/evdi-1.3.43/evdi_gem.c
/usr/src/evdi-1.3.43/evdi_ioc32.c
/usr/src/evdi-1.3.43/evdi_main.c
/usr/src/evdi-1.3.43/evdi_modeset.c
/usr/src/evdi-1.3.43/evdi_painter.c
/usr/src/evdi-1.3.43/evdi_stats.c
/var/log/displaylink
```
- Scripts that the rpm runs:
```
# rpm -qp --scripts fedora-25-displaylink-1.3.43-2.x86_64.rpm
postinstall scriptlet (using /bin/sh):
/usr/bin/systemctl daemon-reload
/usr/bin/systemctl -q is-enabled dkms.service || /usr/bin/systemctl enable dkms.service
for kernel in $(rpm -q kernel --queryformat '%{VERSION}-%{RELEASE}.%{ARCH}\n') ;do
    /sbin/dkms install evdi/1.3.43 -k $kernel >> /var/log/displaylink/displaylink.log 2>&1
done
preuninstall scriptlet (using /bin/sh):
if [ $1 -eq 0 ] ;then
    /usr/bin/systemctl -q is-active displaylink.service && /usr/bin/systemctl stop displaylink.service
    /sbin/dkms remove evdi/1.3.43 --all >> /var/log/displaylink/displaylink.log
fi
postuninstall scriptlet (using /bin/sh):
/usr/bin/systemctl daemon-reload
```
- Triggers?:
```
# rpm -qp --triggers fedora-25-displaylink-1.3.43-2.x86_64.rpm
```
- All good? ok, install the rpm:
```
# rpm -ivh fedora-25-displaylink-1.3.43-2.x86_64.rpm
Preparing...                      	################################# [100%]
Updating / installing...
   1:displaylink-1.3.43-2         	################################# [100%]
warning: %post(displaylink-1.3.43-2.x86_64) scriptlet failed, exit status 1
```

For some reason your RPM fails to install and I don't know what is missing here:
- Verify that the module has been compiled correctly:
```
# ll /lib/modules/$(uname -r)/extra/evdi.ko
ls: cannot access '/lib/modules/4.9.13-200.fc25.x86_64/extra/evdi.ko': No such file or directory
```
- Check service status
```
# systemctl status displaylink.service
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: inactive (dead)
```

Can I get any help with this?

 [1]: http://nothen.com.ar/en/support-for-displaylink-adapters-on-linux/

bowerth, 2017-03-05:
Looks like you spent an interesting weekend ;)
judging from your second last error on `evdi.ko`, it seems you still have no evdi module
Maybe check https://github.com/DisplayLink/evdi/ ?

reypm, 2017-03-13:
@bowerth sadly I still needing your help. I have tried several ways around this topic "yes" interested weekend trying to get this working but so far I get nothing :-( 

So latest I have tried is install Nvidia through this [Bumblebee Fedora](https://fedoraproject.org/wiki/Bumblebee) project and having that I could compile the EVDI from the Ubuntu package but what I get is a black screen on the second monitor. Any ideas?

bowerth, 2017-03-13:
at some point I tried compiling evdi but without success - seems this wasn't required in my case. I'm putting the instructions here anyway - it seems you enjoy experimenting!

```
$ wget https://github.com/DisplayLink/evdi/archive/793a65a8f45bc0ca890d8fa8507c0c6c4a5860f8.zip
$ unzip 793a65a8f45bc0ca890d8fa8507c0c6c4a5860f8.zip
$ cd evdi-*
$ make
$ cp -r ./module/ /usr/src/evdi-devel
$ dkms add evdi/devel
$ dkms build evdi/devel
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j4 KERNELRELEASE=4.9.3-200.fc25.x86_64 all INCLUDEDIR=/lib/modules/4.9.3-200.fc25.x86_64/build/include KVERSION=4.9.3-200.fc25.x86_64 DKMS_BUILD=1...
cleaning build area...(bad exit status: 2)

DKMS: build completed.

$ dkms install evdi/devel
$ modprobe evdi
$ systemctl restart displaylink && systemctl status displaylink
```

reypm, 2017-03-13:
@bowerth 

> it seems you enjoy experimenting!

Kind off is just I refuse to go back to Ubuntu just for this, I did love Fedora since I have been working with CentOS for a long time.

Well last time I broke the system completely so I have installed once again from scratch is like the 15 time I do this.

I am running `4.9.13-201.fc25.x86_64` kernel. Having the steps above should I ....

 - compile first the EVDI module using the same exact steps as you leave me and then install the RPM from this repo?
 - install the RPM and then build & enable the EVDI modules?
 - try again and compile from Ubuntu ones?

bowerth, 2017-03-13:
Hm I am afraid I don't understand enough about the internals of evdi and displaylink to answer your question - as I said, it was not required in ny case, I could install evdi without having to compile first. My experience with displaylink has been frustrating but with the latest kernel, it worked after manually setting the display resolution. Maybe someone with more knowledge will be able to help you out, good luck!

reypm, 2017-03-13:
:'( ok, just one latest question:

> ... with the latest kernel, it worked after manually setting the display resolution.

Can you add here what do you mean with manually setting the display resolution? Any steps?

bowerth, 2017-03-13:
See my first reply on 3 March about `xrandr`

reypm, 2017-03-13:
@bowerth I give up :-( there is no way I can make this to work thx for the help hopefully some day this will work as should be with an easy setup

hadess, 2017-03-13:
If the driver is installed correctly, you just need to change the display configuration in your desktop environment's Settings. In your case, the driver doesn't install properly.

What's the output of:
`cd /usr/src/evdi-1.3.43 ; make`
?

reypm, 2017-03-13:
@hadess I have reinstalled everything and didn't compile the EVDI driver, can the command be run without this or I need to compile once again the driver? The RPM from this repo doesn't work at all

hadess, 2017-03-13:
> @hadess I have reinstalled everything and didn't compile the EVDI driver, can the command be run without this or I need to compile once again the driver? The RPM from this repo doesn't work at all

The RPM is supposed to compile the driver automatically. It's failing which is why nothing works. We're trying to see which it fails.

reypm, 2017-03-13:
@hadess should I wait until you have something?

reypm, 2017-03-15:
@hadess any news?

ssaavedra, 2017-03-15:
We'd need to know the output of what happens if you run this command: 

`cd /usr/src/evdi-1.3.43 ; make`

This is for trying to manually compile the driver. The RPM tries to do this automatically but fails in your system. We don't know why, and since we can't reproduce the bug we need to know more from your setup. And since the rpm is supposed to be user-friendly it does not output the cause of the error, but it's not necessarily the RPM script's fault, it may be evdi not being compiled correctly.

reypm, 2017-03-15:
@ssaavedra Ok I can try to provide you with such information but I am not clear in what to do so ...

 - should I try to compile the EVDI module from Ubuntu package as described [here](http://mattwilmott.com/2016/11/03/how-to-install-the-displaylink-driver-fedora-2425/comment-page-1/) and then provide you with the info you required?

I have a fresh installation without any of the previous described modifications and if I run the command I got the following:
```
cd /usr/src/evdi-1.3.43 ; make
bash: cd: /usr/src/evdi-1.3.43: No such file or directory
make: *** No targets specified and no makefile found.  Stop
```

So you tell me what to do and I'll do

ssaavedra, 2017-03-15:
I think that first you should try to install the RPM from this repo, then, after the warning (the package should not be uninstalled just because of that, right @hadess?), try to perform the above command.

reypm, 2017-03-15:
@ssaavedra ok here you go:

- Install the RPM from this repo:
```
$ sudo dnf install -y fedora-25-displaylink-1.3.43-2.x86_64.rpm
[sudo] password for rperez: 
Last metadata expiration check: 2:47:27 ago on Wed Mar 15 13:07:49 2017.
Dependencies resolved.
===================================================================================================================================================================================================================
 Package                                             Arch                                           Version                                             Repository                                            Size
===================================================================================================================================================================================================================
Installing:
 displaylink                                         x86_64                                         1.3.43-2                                            @commandline                                         3.1 M

Transaction Summary
===================================================================================================================================================================================================================
Install  1 Package

Total size: 3.1 M
Installed size: 8.5 M
Downloading Packages:
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Installing  : displaylink-1.3.43-2.x86_64                                                                                                                                                                    1/1 
warning: %post(displaylink-1.3.43-2.x86_64) scriptlet failed, exit status 1
Non-fatal POSTIN scriptlet failure in rpm package displaylink
Non-fatal POSTIN scriptlet failure in rpm package displaylink
  Verifying   : displaylink-1.3.43-2.x86_64                                                                                                                                                                    1/1 

Installed:
  displaylink.x86_64 1.3.43-2                                                                                                                                                                                      

Complete!
```
 - Output from the command required:
```
$ cd /usr/src/evdi-1.3.43 ; make
make -C /lib/modules/4.9.13-201.fc25.x86_64/build M=$PWD
make[1]: Entering directory '/usr/src/kernels/4.9.13-201.fc25.x86_64'
mkdir: cannot create directory ‘/usr/src/evdi-1.3.43/.tmp_versions’: Permission denied
  LD      /usr/src/evdi-1.3.43/built-in.o
ar: /usr/src/evdi-1.3.43/built-in.o: Permission denied
scripts/Makefile.build:446: recipe for target '/usr/src/evdi-1.3.43/built-in.o' failed
make[2]: *** [/usr/src/evdi-1.3.43/built-in.o] Error 1
Makefile:1494: recipe for target '_module_/usr/src/evdi-1.3.43' failed
make[1]: *** [_module_/usr/src/evdi-1.3.43] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.9.13-201.fc25.x86_64'
Makefile:57: recipe for target 'module' failed
make: *** [module] Error 2
```

ssaavedra, 2017-03-15:
Are you running the later command as root? If not, run `sudo make` instead of `make`, because it seems you don't have permissions to edit that folder.

reypm, 2017-03-15:
@ssaavedra yup I just read the error and saw it needs to be run as root so I have updated the comment ...

ssaavedra, 2017-03-15:
It seems to build. Try to run `dnf reinstall` on the package, and see if the scriptlets no longer fail. Anyway, it seems you have now build the evdi driver; you should be able to run as root

`modprobe evdi` and `systemctl restart displaylink.service` to get displaylink working (because the rpm is otherwise correctly installed, maybe this is some issue with dkms then).

ssaavedra, 2017-03-15:
Sorry about the updated comment, I had the page already loaded and needed to reload to notice the change :blush:

reypm, 2017-03-15:
@ssaavedra nope, still not working:
```
$ sudo dnf reinstall -y fedora-25-displaylink-1.3.43-2.x86_64.rpm
Last metadata expiration check: 2:54:27 ago on Wed Mar 15 13:07:49 2017.
Dependencies resolved.
===================================================================================================================================================================================================================
 Package                                             Arch                                           Version                                             Repository                                            Size
===================================================================================================================================================================================================================
Reinstalling:
 displaylink                                         x86_64                                         1.3.43-2                                            @commandline                                         3.1 M

Transaction Summary
===================================================================================================================================================================================================================

Total size: 3.1 M
Downloading Packages:
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Reinstalling: displaylink-1.3.43-2.x86_64                                                                                                                                                                    1/2 
warning: %post(displaylink-1.3.43-2.x86_64) scriptlet failed, exit status 1
Non-fatal POSTIN scriptlet failure in rpm package displaylink
Non-fatal POSTIN scriptlet failure in rpm package displaylink
  Erasing     : displaylink-1.3.43-2.x86_64                                                                                                                                                                    2/2 
  Verifying   : displaylink-1.3.43-2.x86_64                                                                                                                                                                    1/2 
  Verifying   : displaylink-1.3.43-2.x86_64                                                                                                                                                                    2/2 

Reinstalled:
  displaylink.x86_64 1.3.43-2                                                                                                                                                                                      

Complete!
$ sudo modprobe evdi
modprobe: FATAL: Module evdi not found in directory /lib/modules/4.9.13-201.fc25.x86_64
```

ssaavedra, 2017-03-15:
This is weird. What happens if you run `cd /usr/src/evdi-1.3.43 && sudo make && sudo modprobe evdi` ?

reypm, 2017-03-15:
@ssaavedra 
```
$ cd /usr/src/evdi-1.3.43 && sudo make && sudo modprobe evdi
make -C /lib/modules/4.9.13-201.fc25.x86_64/build M=$PWD
make[1]: Entering directory '/usr/src/kernels/4.9.13-201.fc25.x86_64'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/kernels/4.9.13-201.fc25.x86_64'
modprobe: FATAL: Module evdi not found in directory /lib/modules/4.9.13-201.fc25.x86_64
```

ssaavedra, 2017-03-15:
Not sure what can be happening, but evdi is not getting moved to your kernel tree's modules directory.

Can you at least insert the kmod manually like this?
`cd /usr/src/evdi-1.3.43 && sudo insmod ./evdi.ko`

ssaavedra, 2017-03-15:
If you can do that, then you have evdi in your kernel now and you *should* now be able to do `systemctl restart displaylink.service` and make it work, it's just a problem of the module not getting built and moved to the appropriate tree in the kernel module space. I'm not sure how to solve that, I'd like somebody else's input on that.

reypm, 2017-03-15:
@ssaavedra  ok here is the result:
```
# cd /usr/libexec/displaylink
# /usr/libexec/displaylink/DisplayLinkManager
[libevdi] ioctl: drop_master error=-1
[libevdi] ioctl: drop_master error=-1
[libevdi] ioctl: drop_master error=-1
Aborted (core dumped)

# dkms install evdi/1.3.43

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j8 KERNELRELEASE=4.9.13-201.fc25.x86_64 all INCLUDEDIR=/lib/modules/4.9.13-201.fc25.x86_64/build/include KVERSION=4.9.13-201.fc25.x86_64 DKMS_BUILD=1...
cleaning build area...(bad exit status: 2)

DKMS: build completed.

evdi.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.9.13-201.fc25.x86_64/extra/
Adding any weak-modules

depmod.....

DKMS: install completed.

# systemctl restart displaylink.service
```
Restart the service makes the screen flash several times but I am still not seeing anything on the extra monitor, this one is connected through HDMI (in the Dock station) => to DVI (in the monitor).

The other problem is now I have random black flashes so what to do in this case?

ssaavedra, 2017-03-15:
It seems you have installed the evdi driver and that the DisplayLinkManager is now running.

I don't understand why isn't it working for you. Try other outputs (VGA vs HDMI if your dock has both), or try restarting the machine. If that doesn't work out, you can always `sudo systemctl stop displaylink.service && sudo rmmod evdi` and that should bring the black flashes to a halt.

The flashes are probably the video subsystem trying to bring the other screen online, maybe you can see relevant information on your journalctl or in the output of your `sudo dmesg -kwL` (the debug kernel messages).

reypm, 2017-03-15:
@ssaavedra I have tried the same RPM at home using the same Fedora version and the same kernel and everything went fine. What's different? The video card support: at home I have an nVidia GTX 970, at work I have a laptop (Ideapad 700) and I believe it has two graphics card enabled: on board (an Intel) and dedicated (an nVidia), could be this the problem?

I was researching and there is no way (or at least I couldn't find it) to leave only the nVidia working and disabled the on board one. On the contrary I can do the opposite  meaning I can disable completely the nVidia and use only the on board.

Is there any command I can run to provide you with more information so you can check if that's the problem? I am not using any specific that needs the nVidia video card so I can disable in case is needed to make displaylink driver to work.

reypm, 2017-03-17:
@ssaavedra @hadess @bowerth by disabling the nVidia graphic card I make DisplayLink driver to _work_. Now I am having issues with the third screen (the closer to the right where the account window is) since I am not able to see anything that is draw there. Can any give me some help to fix this one? We're almost there :)

[image]

ssaavedra, 2017-06-13:
Hi, I'm gonna close this issue and keep track of the newer issue at #23 if that's all right.

closed 2017-06-13 by ssaavedra