displaylink-rpm/displaylink-rpm #23: DisplayLink Display doesn't refresh

mustacheride, 2017-04-03:
I'm having an issue with the Disaplylink devices not updating the screen when things change. For instance, on the laptop screen I am typing this message now, however on the screens connected to the DisplayLink device, it shows the original image of the screen the last time I restarted the DisplaylinkManager. 

Device Info: 
Dell XPS 13 (9360)
Dell D3100 DisplayLink Docking Station
Fedora 25

`$ uname -r
4.10.6-200.fc25.x86_64`

`$ sudo modprobe evdi
$`

`$ sudo systemctl status displaylink
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: active (running) since Mon 2017-04-03 12:03:10 CDT; 6min ago
  Process: 12753 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 12755 (DisplayLinkMana)
    Tasks: 43 (limit: 4915)
   CGroup: /system.slice/displaylink.service
           └─12755 /usr/libexec/displaylink/DisplayLinkManager

Apr 03 12:03:10  systemd[1]: Starting DisplayLink Manager Service...
Apr 03 12:03:10  systemd[1]: Started DisplayLink Manager Service.`

Everything was installed using the git source and running make inside the directory. I did have to perform the following steps before I could get the displaylink manager to run: 

`$ cd /usr/src/evdi-1.3.43/ && sudo make`
`$ sudo insmod ./evdi.ko `
`$ sudo dkms install evdi/1.3.43`

I can see the mouse movement on the DL attached screens, however the browser and workspace switcher does not update to match what is currently on the laptop screen. 

An example is attached showing the disparity between laptop screen and DL attached screen. 
[image]

I'm guessing this is a conflict or something with the laptop's display driver and the DL driver? I'm not really sure whats going on here to cause this issue.

reypm, 2017-04-04:
@ssaavedra @hadess this seems to be related with the issue I am having as in [here](https://github.com/displaylink-rpm/displaylink-rpm/issues/22#issuecomment-287349068), any advice or ideaS?

Geodomus, 2017-04-10:
Also having this issue currently. May have something to do with a newer kernel version? (Blind guess).

Geodomus, 2017-04-10:
just tested it with evdi 1.4. No changes, same bug

ssaavedra, 2017-06-13:
Has this been addressed on the latest versions?

choonming, 2017-07-07:
@ssaavedra I'm using the lastest RPM as of today. I can see my secondary display connected but the displays are not refreshing. I can see my mouse cursor hovering about not when I drag a window over there, nothing happens.

ssaavedra, 2017-07-10:
Version 1.3.54 of the daemon has been released. The builds are now available thanks to #27 
Please, try again and report back.

bowerth, 2017-07-11:
dark screen... only mouse cursor - depending on display setup, I get a wallpaper but not possible to move any window to DL screen (Lenovo LT1421, using evdi 1.4.1 and latest #27 )

ssaavedra, 2017-10-17:
This seems to have been solved in the latest releases. Closing.

closed 2017-10-17 by ssaavedra