displaylink-rpm/displaylink-rpm #24: RHEL7 with 3.10 kernel steps

LinuxETC, 2017-04-20:
I am working on doing the "hand compile" method for a RHEL7 laptop (work related system) which has a 3.10 kernel (yes, I know...dated).  Here is what I have done so far for steps:

```
wget displaylink-rpm-1.3.52.tar.gz
tar -zxvf displaylink-rpm-1.3.52.tar.gz
cd displaylink-rpm-1.3.52
sudo yum install -y gcc rpm-build make wget yum-utils libdrm-devel
sudo make
sudo rpm -ivh displaylink-1.3.43-2.src.rpm
```
However, when I do a "lsmod|grep evdi" (sans quotes) I am not noticing the "evdi" module noted as I see on my Ubuntu 16.04 laptop (personal system).  Did I miss a step or two here?

Thanks in advance.

ssaavedra, 2017-04-27:
You should not install the src.rpm but instead the rpm for your architecture. Installing a src.rpm will only put the specfiles and related content into the appropriate rpmbuild folders.

ssaavedra, 2017-04-27:
If you're using RHEL7, would it work with our centos7 binary release?

LinuxETC, 2017-04-27:
It did not like that the RHEL7 kernel was 3.10.x unfortunately upon trying the .rpm versus the .src.rpm.

ssaavedra, 2017-04-27:
Didn't it work either with your own built rpm?
As I understand, you ran `make` on this repo, which should have compiled the rpm besides creating the .src.rpm.

LinuxETC, 2017-04-27:
It is missing the "evdi" module it seems when doing the .src.rpm method as noted initially.

ssaavedra, 2017-06-13:
When you install a src.rpm file you do not actually build anything. You are just installing the sources, but not the actual module you need. You must build it *separately*.

The src.rpm can just be used to build the actual rpm. Instead of that, you can go all the way to building the binary rpm from this repo by typing `make` from the root of this folder.

If the resulting RPM does not like the fact that you are using a newer kernel you may need to downgrade evdi/displaylink versions until a point in the past where they support such kernel version.

But once again, you need to build the binary rpm; installing the src.rpm does not cause rpm to automagically build the sources for you.

You can take a look at these resources to better understand SRPMS: https://wiki.centos.org/HowTos/RebuildSRPM and https://docs.fedoraproject.org/ro/Fedora_Draft_Documentation/0.1/html/RPM_Guide/ch11s03.html

hadess, 2017-10-10:
The evdi driver plain does not work on RHEL7's 3.10 kernel. We'd need an older version of the evdi driver to have this working.

closed 2017-10-10 by hadess