displaylink-rpm/displaylink-rpm #25: Fedora 25 on 4.11.3-200.fc25.x86_64 fails to install or build

digimer, 2017-06-08:
When I try to install the source RPM, I get:
````
  0 root@pulsar:~/rpmbuild/SPECS# rpmbuild -ba displaylink.spec 
error: File /root/rpmbuild/SOURCES/v%{_version}.tar.gz: No such file or directory
````
When I try to install the precompiled RPM;
````
  0 root@pulsar:~# rpm -Uvh fedora-25-displaylink-1.3.43-2.x86_64.rpm 
Preparing...                          ################################# [100%]
Updating / installing...
   1:displaylink-1.3.43-2             ################################# [100%]
warning: %post(displaylink-1.3.43-2.x86_64) scriptlet failed, exit status 10
````
This was on a freshly rebooted fedora 25 to make sure I was running the latest kernel.

digimer, 2017-06-08:
To add, just to be safe, I ````setenforce 0```` to make sure it wasn't an SELinux issue.
````
  0 root@pulsar:~# sestatus 
SELinux status:                 enabled
SELinuxfs mount:                /sys/fs/selinux
SELinux root directory:         /etc/selinux
Loaded policy name:             targeted
Current mode:                   permissive
Mode from config file:          enforcing
Policy MLS status:              enabled
Policy deny_unknown status:     allowed
Max kernel policy version:      30
````
cheers

alhalama, 2017-06-08:
The last release here was on Feb 21st and evdi had a commit to add support for kernels newer than 4.10 on February 28th.  The version of evdi used needs to be increased to use this on the newer kernel.

digimer, 2017-06-09:
Thanks for the reply. I'm fairly new to this... Are their plans to release a new / testing RPM that pulls those changes in by chance?

digimer, 2017-06-09:
OK, I had more success thanks to you hint. I added the following to the top of the .spec file;

````
%global debug_package %{nil}
+%global _version 1.4.1
+%global _daemon_version 1.3.52
+%global _release 1.3.52

Name:           displaylink
````
And I manually downloaded the 1.4.1 source for the ````evdi```` package and put it into ````rpmbuild/SOURCES```` and I was able to build and install the RPM. This is better than I've gotten so far!

When I plug the monitor in (an ASUS MB169B+), Fedora sees it and for a very brief moment, I can see something on the screen, then it freezes. If I open the display menu in gnome's settings, it shows both monitors and I can move the position around and when I apply changes, it briefly shows the screen again, then freezes. I can move the mouse and windows into the area where the monitor is (off the main screen), but I can't see anything. Once I unplug the monitor, it returns to normal operation.

My laptop is a Thinkpad P70 with the 4k screen and I have disabled HiDPI, in case that matters. Here is what I see in ````journalctl````;

````
Jun 08 22:01:01 pulsar.alteeve.ca run-parts[20466]: (/etc/cron.hourly) finished 0anacron
Jun 08 22:02:23 pulsar.alteeve.ca kernel: usb 2-1: new SuperSpeed USB device number 11 using xhci_hcd
Jun 08 22:02:23 pulsar.alteeve.ca kernel: usb 2-1: New USB device found, idVendor=17e9, idProduct=ff0b
Jun 08 22:02:23 pulsar.alteeve.ca kernel: usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 08 22:02:23 pulsar.alteeve.ca kernel: usb 2-1: Product: ASUSTEK MB169B+
Jun 08 22:02:23 pulsar.alteeve.ca kernel: usb 2-1: Manufacturer: DisplayLink
Jun 08 22:02:23 pulsar.alteeve.ca kernel: usb 2-1: SerialNumber: G3LMTF127356
Jun 08 22:02:23 pulsar.alteeve.ca mtp-probe[20709]: checking bus 2, device 11: "/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1"
Jun 08 22:02:23 pulsar.alteeve.ca mtp-probe[20709]: bus: 2, device: 11 was not an MTP device
Jun 08 22:02:23 pulsar.alteeve.ca fwupd[10410]: interface found, but not interface data
Jun 08 22:02:23 pulsar.alteeve.ca systemd[1]: Starting DisplayLink Manager Service...
Jun 08 22:02:23 pulsar.alteeve.ca systemd[1]: Started DisplayLink Manager Service.
Jun 08 22:02:23 pulsar.alteeve.ca audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=displaylink comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jun 08 22:02:26 pulsar.alteeve.ca kernel: nouveau 0000:01:00.0: DRM: resuming object tree...
Jun 08 22:02:26 pulsar.alteeve.ca kernel: nouveau 0000:01:00.0: priv: HUB0: 614900 00800000 (18408200)
Jun 08 22:02:27 pulsar.alteeve.ca kernel: nouveau 0000:01:00.0: DRM: resuming fence...
Jun 08 22:02:27 pulsar.alteeve.ca kernel: nouveau 0000:01:00.0: DRM: resuming display...
Jun 08 22:02:27 pulsar.alteeve.ca kernel: nouveau 0000:01:00.0: DRM: resuming console...
Jun 08 22:02:27 pulsar.alteeve.ca kernel: evdi: [W] evdi_painter_disconnect:471 (dev=-1) An unknown connection to ffff994bae925600 tries to close us
Jun 08 22:02:27 pulsar.alteeve.ca kernel: evdi: [W] evdi_painter_disconnect:472  - ignoring
Jun 08 22:02:27 pulsar.alteeve.ca kernel: evdi: [D] evdi_painter_connect:450 (dev=2) Connected with ffff994bae925600
Jun 08 22:02:27 pulsar.alteeve.ca kernel: evdi: [D] evdi_detect:75 (dev=2) Painter is connected
Jun 08 22:02:27 pulsar.alteeve.ca kernel: evdi: [D] evdi_painter_crtc_state_notify:359 (dev=2) Notifying crtc state: 3
Jun 08 22:02:27 pulsar.alteeve.ca kernel: evdi: [D] evdi_detect:75 (dev=2) Painter is connected
Jun 08 22:02:27 pulsar.alteeve.ca kernel: evdi: [D] evdi_painter_get_edid_copy:192 (dev=2) 00 ff ff
Jun 08 22:02:27 pulsar.alteeve.ca /usr/libexec/gdm-x-session[9862]: (II) modeset(0): EDID vendor "MEI", prod id 38562
Jun 08 22:02:27 pulsar.alteeve.ca /usr/libexec/gdm-x-session[9862]: (II) modeset(0):     EDID quirk: Use maximum size instead of detailed timing sizes.
Jun 08 22:02:27 pulsar.alteeve.ca /usr/libexec/gdm-x-session[9862]: (II) modeset(0): Using hsync ranges from config file
Jun 08 22:02:27 pulsar.alteeve.ca /usr/libexec/gdm-x-session[9862]: (II) modeset(0): Using vrefresh ranges from config file
Jun 08 22:02:27 pulsar.alteeve.ca /usr/libexec/gdm-x-session[9862]: (II) modeset(0): Printing DDC gathered Modelines:
Jun 08 22:02:27 pulsar.alteeve.ca /usr/libexec/gdm-x-session[9862]: (II) modeset(0): Modeline "3840x2160"x0.0  536.99  3840 4002 4010 4026  2160 2168 2176 2223 +hsync +vsync (133.4 kHz eP)
Jun 08 22:02:27 pulsar.alteeve.ca /usr/libexec/gdm-x-session[9862]: (II) modeset(0): Allocate new frame buffer 5760x2160 stride
Jun 08 22:02:27 pulsar.alteeve.ca kernel: evdi: [D] evdi_painter_mode_changed_notify:383 (dev=2) Notifying mode changed: 1920x1080@60; bpp 32; 
Jun 08 22:02:27 pulsar.alteeve.ca kernel: evdi: [D] evdi_painter_mode_changed_notify:384 pixel format 875713112
Jun 08 22:02:27 pulsar.alteeve.ca kernel: evdi: [D] evdi_painter_crtc_state_notify:359 (dev=2) Notifying crtc state: 0
Jun 08 22:02:27 pulsar.alteeve.ca kernel: evdi: [D] evdi_painter_dpms_notify:346 (dev=2) Notifying dpms mode: 0
Jun 08 22:02:27 pulsar.alteeve.ca /usr/libexec/gdm-x-session[9862]: (II) modeset(0): EDID vendor "MEI", prod id 38562
Jun 08 22:02:27 pulsar.alteeve.ca /usr/libexec/gdm-x-session[9862]: (II) modeset(0):     EDID quirk: Use maximum size instead of detailed timing sizes.
Jun 08 22:02:27 pulsar.alteeve.ca /usr/libexec/gdm-x-session[9862]: (II) modeset(0): Using hsync ranges from config file
[... cut ...]
Jun 08 22:02:33 pulsar.alteeve.ca kernel: nouveau 0000:01:00.0: DRM: suspending display...
Jun 08 22:02:33 pulsar.alteeve.ca kernel: nouveau 0000:01:00.0: DRM: evicting buffers...
Jun 08 22:02:33 pulsar.alteeve.ca kernel: nouveau 0000:01:00.0: DRM: waiting for kernel channels to go idle...
Jun 08 22:02:33 pulsar.alteeve.ca kernel: nouveau 0000:01:00.0: DRM: suspending fence...
Jun 08 22:02:33 pulsar.alteeve.ca kernel: nouveau 0000:01:00.0: DRM: suspending object tree...
````
Should I be reporting this here or on the evdi repo?

Thanks!

digimer, 2017-06-09:
In case it helps, here are the src.rpm and x86_64 RPMs from my build;

https://www.alteeve.com/an-repo/files/displaylink-1.4.1-1.3.52.src.rpm
https://www.alteeve.com/an-repo/files/displaylink-1.4.1-1.3.52.x86_64.rpm

ment0s, 2017-06-10:
Is your p70 with discrete graphics ?

digimer, 2017-06-10:
It's using the hybrid mode. You can't disable the nvidia anymore.
````
  0 root@pulsar:~# lspci |grep VGA
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 530 (rev 06)
01:00.0 VGA compatible controller: NVIDIA Corporation GM107GLM [Quadro M600M] (rev a2)
````

ment0s, 2017-06-17:
You can give it a try on solely NVidia, I am running 4.10.17-100.fc24.x86_64 and EVDI just broke for me. I got around it by Switching solely to NVIDIA.

therevoman, 2017-06-22:
in order to get it working on my Lenovo T460s I had to add the following xorg files:

$ lspci |grep VGA
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 520 (rev 07)

/etc/X11/xorg.conf.d/11-displaylink.conf 
Section "Monitor"
Identifier "DVI-I-1-1"
Modeline "1920x1080@49" 138.55 1920 2032 2232 2544 1080 1081 1084 1111 -HSync +Vsync
Option "PreferredMode" "1920x1080@49"
EndSection

and also this file:
/etc/X11/xorg.conf.d/20-intel.conf 
Section "Device"
Identifier "intel"
Driver "modesetting"
Option "kmsdev" "/dev/dri/card0"
Option "PageFlip" "off"
Option "SWCursor" "on"
Option "ShadowFB" "true"
EndSection

Section "Device"
Identifier "USB3"
BusID "USB"
Driver "modesetting"
Option "kmsdev" "/dev/dri/card1"
Option "PageFlip" "off"
Option "SWCursor" "on"
Option "ShadowFB" "true"
EndSection

ssaavedra, 2017-10-17:
Is this still an issue with the latest released version?

ssaavedra, 2018-01-19:
Fedora 25 has been EOL for more than a month already. Closing.

closed 2018-01-19 by ssaavedra