displaylink-rpm/displaylink-rpm #32: Fedora 26 support is currently broken

chlongv, 2017-08-16:
First, with the latest F26 4.12 kernel, the module does not build anymore.

A fix was already added in the upstream DisplayLink/evdi repo: https://github.com/DisplayLink/evdi/commit/6c689d69457820512f25204d83f0880824a932f3

But even when applying this patch, I have other errors with evdi that prevent it from working (it was fine yesterday but is broken with latest package upgrades).

Here is a log:

Aug 16 09:57:53 CH900635.keymile.net kernel: evdi: [D] add_store:195 Increasing device count to 1
Aug 16 09:57:53 CH900635.keymile.net kernel: evdi: [D] evdi_crtc_init:310 drm_crtc_init: 0
Aug 16 09:57:53 CH900635.keymile.net kernel: evdi: [W] evdi_painter_crtc_state_notify:362 Painter does not exist!
Aug 16 09:57:53 CH900635.keymile.net kernel: evdi: [D] evdi_detect:78 Painter is disconnected
Aug 16 09:57:53 CH900635.keymile.net kernel: [drm] Cannot find any crtc or sizes - going 1024x768
Aug 16 09:57:53 CH900635.keymile.net kernel: evdi evdi.0: fb1: evdidrmfb frame buffer device
Aug 16 09:57:53 CH900635.keymile.net kernel: [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
Aug 16 09:57:53 CH900635.keymile.net kernel: [drm] No driver support for vblank timestamp query.
Aug 16 09:57:53 CH900635.keymile.net kernel: [drm] evdi: evdi_stats_init
Aug 16 09:57:53 CH900635.keymile.net kernel: [drm] Initialized evdi 1.4.1 20170417 for evdi.0 on minor 1
Aug 16 09:57:53 CH900635.keymile.net kernel: evdi: [D] evdi_detect:78 Painter is disconnected
Aug 16 09:57:53 CH900635.keymile.net kernel: evdi: [W] evdi_painter_disconnect:471 (dev=0) An unknown connection to ffff9d8ac4105200 tries to close us
Aug 16 09:57:53 CH900635.keymile.net kernel: evdi: [W] evdi_painter_disconnect:472  - ignoring

chlongv, 2017-08-16:
When manually enabling the displaylink.service (systemctl enable displaylink.service) it now works with the F26 4.12 kernel as well.

The current RPM fails to run the 2 post install scriplets that:
1. build the kernel module (this is fixed withe commit referenced in the previous message)
2. enable/start the displaylink.service

The RPM should be fixed so these 2 post install scriplets run successfuly with current fedora 26.

tartansandal, 2017-08-17:
Just made a PR (https://github.com/displaylink-rpm/displaylink-rpm/pull/33) that should help with this.

tduis, 2017-08-17:
Thank you guys, was struggeling to get this working. Did the patching by hand (/usr/src/evdi-1.4.1/<files>) on FC26 with kernel (4.12.5-300.fc26.x86_64 #1 SMP Mon Aug 7 15:27:25 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux)  And IT WORKS - great - THANK YOU

linuxcrash, 2017-08-23:
I have the same issue.
Is there an ETA when the RPM will be updated for general availability? 

Thanks

bowerth, 2017-08-24:
same here, could get it to work with following sequence:
1) build and install https://github.com/tartansandal/displaylink-rpm (master)
2) build and install evdi-1.4.1 kernel module from https://github.com/DisplayLink/evdi/ (devel)

linuxcrash, 2017-08-28:
@bowerth  Somehow I did not get this to work. Would you be able to detail the process a bit more? Seems I'm not familiar enough with build and install tasks.

tartansandal, 2017-09-05:
The aforementioned patch https://github.com/displaylink-rpm/displaylink-rpm/pull/33 has been merged with master. If you checkout https://github.com/displaylink-rpm/displaylink-rpm and do

   make clean
   make rawhide

it should do all the magic for you.

dilom, 2017-09-05:
I can verify that the magic works in F26!

DevNiall, 2017-09-05:
@tartansandal Thanks for the effort dude! My F26 Dell XPS is usable again.

RandieM, 2017-09-05:
Hello everyone,

First of all, I wanted to say a big thanks for this package. Without it, it was almost impossible for me to use my second screen over a DisplayLink adapter.

Unfortunately, since upgrading to the latest kernel (or the one before last; my current version is: Linux fedora 4.12.9-300.fc26.x86_64 #1 SMP Fri Aug 25 13:09:43 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux), the driver no longer worked.

I followed the process described above by @tartansandal and I got the second screen to be detected, but all I can see is my background and the cursor. If I move a window to the secondary screen, it does not show at all. Is there anything I can do to fix this?

Thanks!

bowerth, 2017-09-05:
@linuxcrash the described procedure was only a temporary work-around until the patch [#33](https://github.com/displaylink-rpm/displaylink-rpm/pull/33) could be merged with master

RandieM, 2017-09-05:
Thanks for your answer @linuxcrash. I am not sure what you mean though. It is the master branch, after the merge of #33, that I cloned and used.

tartansandal, 2017-09-05:
Morning People!

5:08am in Melbourne at the moment and I'm just starting on my first coffee, so apologies for any incoherence.

@DevNiall you are welcome. Also thanks to @ssaavedra for some very valuable feedback and to all the others who have contributed to this in the past ;-)

@RandieM Sounds like you are doing things correctly, but there is a chance that past installs have confused the system. Master now has both #33 and #34 and merged with master and now has some neat new _rawhide_ support. (I got my PR numbers mixed up in my last comment).  If you do `rpm -q displaylink` you should get something something like:

    displaylink-1.4.1-5.rawhide.20170904.x86_64

If you get something like

    displaylink-1.5.0-5.x86_64

you should remove that and build and reinstall the 'official' _rawhide_ one.

From your description, the module is loading, but it looks like something I've seen before with some of the xorg hacks.  Check the contents of `/etc/X11/xorg.conf.d/`. There should see `20-displaylink.conf` which has the tweaks needed to get the 'modesetting' driver to work with displaylink. If you've added other hack files, like the 'intel' ones described elsewhere, they may be stepping on each others toes. If this is the case you will need to either, remove the 'intel' hack file or merge what works for you into the 'displaylink' one. That file is now marked as a configuration file, so future upgrades shouldl not touch your modifications.

Hope this helps.

(Now for my second cup)

tartansandal, 2017-09-06:
A further update. This is now failing for me for kernel 4.12.9-300.fc26.x86_64, but works for 4.12.8-300.fc26.x86_64. This may be a case where we have to wait for a fix from the evdi team, noting that they already have an issue open for kernel 4.13. If I get a spare moment, might ask for some info.

RandieM, 2017-09-07:
@tartansandal thanks a lot for your detailed answer, which, at first, sounded very promising, because I indeed had an intel hack file in that directory and I had totally forgotten about it. It is impressive that you thought about this! But first things first.

The displaylink version that I have installed is displaylink-1.4.1-5.rawhide.20170906.x86_64, which I cloned from the latest commit of the project's master branch. As I said, I did have an intel config file in /etc/X11/xorg.conf.d/, which I tried to a) remove completely, and b) merge the differences with the displaylink config. I even re-installed the displaylink package after having made the intel-related changes. Unfortunately, none of these helped.

Since you mention that you also experience problems with same kernel version, I guess I should wait and hope this gets fixed soon. In any case, if you have any news or need me to test something, please let me know. And, once again, thanks for everyone's time and help.

linuxcrash, 2017-09-10:
Thank you all guys. With all the above descriptions I managed to download and install the version before PR#33 was added to master and it worked for a while. 
Now I'm stuck again with the new kernel update described by @RandieM above which is surely not your fault :-)

Thanks again for you continuous work and effort to keep this running and up to date!

linuxcrash, 2017-09-10:
Gentlemen,
I just downloaded the current master release and did again a make rawhide and reinstall of the RPM and now it even works with my two external 27" displays where it only with one before!

Seems like the EVDI team already has a patch for the new kernel in their dev branch?!

Finally I really can use ny monitors as planned until the next kernel update is killing it again :-)

tartansandal, 2017-09-10:
Glad things are working for you @linuxcrash :-). I had a similar experience this morning, with a similar duel monitor setup, although mine required an extra reboot and a couple of replugs of the usb3 cable (which triggers a displaylink service restart). Unfortunately, the devel branch of edvi has not been updated since 12 Jun so I don't think we can attribute the fix there. However, there have be a couple of OS updates over the past few days which may have something to do with it. Have seen suspicious things in the logs but have not been able to pin down exactly what is going on. One unusual entry suggested that Gnome could not figure out the size of one of the virtual displays. Maybe the update to gnome-settings fixed this?

linuxcrash, 2017-09-11:
I actually did not check the devel branch but was rather assuming they had updated some stuff. so yes it could be the Gnome update but I did not verify this either.

itblaked, 2017-09-12:
Hi guys, I can't get anything to show on my 2nd monitor when using a Targus dock with displaylink-1.4.1-5.rawhide.20170912.x86_64. 
This is my first time trying to get the display working via usb, decided i'm sick of the hdmi on my laptop and wanted to get the display via usb working and landed here. I'm running an up to date version of F26.

I had followed the displaylink article (http://support.displaylink.com/knowledgebase/articles/679060) to port the drivers/install and thought the install script worked should work fine as is, which it appeared to and brought me to the same result I've got from this RPM, which is the display shows as detected is lsusb but nothing appears on it and it doesn't show in the displays app.

I then found this repo and uninstalled the displaylink sourced drivers using:
`sudo ./displaylink-installer.sh uninstall`

After reading the comments in this issue, I performed the following:
 ```git clone git@github.com:displaylink-rpm/displaylink-rpm.git
cd displaylink-rpm/
make clean
make rawhide
sudo dnf install rpm-build
make clean
make rawhide
sudo dnf install libdrm-devel
make rawhide
sudo dnf install ./displaylink-1.4.1-5.rawhide.20170912.x86_64.rpm

systemctl status displaylink
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset
   Active: active (running) since Tue 2017-09-12 10:49:43 AEST; 51s ago
  Process: 5440 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 5441 (DisplayLinkMana)
    Tasks: 24 (limit: 4915)
   Memory: 14.2M
      CPU: 25.658s
   CGroup: /system.slice/displaylink.service
           └─5441 /usr/libexec/displaylink/DisplayLinkManager

systemd[1]: Starting DisplayLink Manager Service...
systemd[1]: Started DisplayLink Manager Service.
```
Rebooted, disconnected and reconnected USB cable a couple times, same result.

Hoping you might have some advice?

Thanks!

tartansandal, 2017-09-12:
Just did a system update this morning including and upgrade kernel to 4.12.11-300.fc26.x86_64. DKMS did its thing and the first external monitor came up straight away (without having to rebuild the displaylink rpm). However, I did have to replug the usb3 port once for it to register the second external monitor. All seems to be working well.

tartansandal, 2017-09-13:
@itblaked from the above it appears that the displaylink manager service is starting ok and the EVDI module has been loaded. Just want to check a few assumptions before we dig any deeper (apologies if they seem silly):

* You are logging in using Xorg rather than Wayland?
* You are plugging into a usb 3 port (not a usb 2)
* other ports on the Targus are working like extra usb ports and sound (not familiar with that dock so just guessing here.)

itblaked, 2017-09-13:
@tartansandal I'm using Wayland, is there a issue with Wayland? Using usb3 and other Targus based services are coming through fine including eth, camera, k+m.

tartansandal, 2017-09-13:
@itblaked the EVDI driver only works for Xorg at the moment, although Wayland support is being worked on (see https://github.com/DisplayLink/evdi/issues/58). 

So you can try logging out, clicking on the gear icon, and selecting an Xorg session.

Yes, you miss out on some extra Wayland goodness, but for me, two external monitors was more important.

itblaked, 2017-09-13:
Ok cool, I'll test it with xorg tomorrow when I'm back at work and let you know, thankyou!

tartansandal, 2017-09-13:
Posting a trick here in case others find it useful.

I had a bit of trouble getting my 2nd external display to work yesterday after a reboot. Re-plugging did not help. On a hunch I tried logging out and logging back in again -- effectively restarting the gnome-session, gnome-shell, and X. Happily this helped to restore the 2nd external display, and supports the hunch that there are some OS level interactions that may be causing issues.

itblaked, 2017-09-13:
Started my laptop, logged into Xorg and the two monitors hanging off the Targus dock came to life straight away, encountered no problems at all. Very happy, thanks @tartansandal for the tipoff about Wayland

itblaked, 2017-09-14:
@tartansandal Thanks for your last tip, it helped me. I undocked for a meeting, returned and couldn't detect any services at all via dock USB3 at all, bounced displaylink svc, still no good. Didn't investigate as other stuff to do. Restarted laptop and got 1 out of 2 monitors at logon. Disconnected, reconnected USB3, didn't help. Logged out and back in and got both monitors working.

bivainis, 2017-09-15:
Compiled rawhide from master branch.

It was a pain to learn how to properly do it, but I somehow managed. This post helped with compiling from src.rpm to regular rpm:  https://blog.packagecloud.io/eng/2015/04/20/working-with-source-rpms/

I encountered many problems along the way, especially got stuck at
`error: File /root/rpmbuild/SOURCES/v%{_version}.tar.gz: No such file or directory` 
as mentioned in https://github.com/displaylink-rpm/displaylink-rpm/issues/25#issue-234589727

Solved this one and some others by setting global vars in the spec file

```
%global _version 1.4.1
%global _daemon_version 1.3.54
%global _release 1.3.52
```

Had to set `%global _daemon_version` to `1.3.54` instead of `1.3.52`
Had a problem with display arrangement (displaylink screen was on the bottom of my hdmi screen), but it went away today.

It works now!
I can even wake my laptop without it getting stuck, unlike previous versions.
I can boot my laptop with usb plugged in, and it not getting stuck, unlike previous versions.
I can shutdown without it getting stuck, unlike previous versions.

Happy!

Remaining issue: cursor rapidly blinks and sometimes isn't visible on primary hdmi monitor, but I can live with that.

itblaked, 2017-10-05:
Hey, heads up I did a system update this morning which bumped the kernel up to 4.13.4-200.fc26.x86_64 and displaylink won't start now due to missing evdi module for new kernel. 

Not sure if it relates to this [https://github.com/DisplayLink/evdi/issues/99].

systemd[1]: Starting DisplayLink Manager Service...
-- Subject: Unit displaylink.service has begun start-up
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit displaylink.service has begun starting up.
modprobe[5292]: modprobe: **FATAL: Module evdi not found in directory /lib/modules/4.13.4-200.fc26.x86_64**
systemd[1]: displaylink.service: Control process exited, code=exited status=1
systemd[1]: Failed to start DisplayLink Manager Service.
-- Subject: Unit displaylink.service has failed
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit displaylink.service has failed.
-- 
-- The result is failed.
systemd[1]: displaylink.service: Unit entered failed state.
systemd[1]: displaylink.service: Failed with result 'exit-code'.
`

kurtis318, 2017-10-06:
I have the same problem as @itblaked .  My kernels is 4.13.4-200.fc26.x86_64 and modprobe called in the displaylink.service is failing.  Sounds like something changed somewhere in the kernel source.

tartansandal, 2017-10-07:
Tried a rebuild against 1.5.0 and kernel-4.13.4-200.fc26.x86_64 this morning and got a core dump when trying to load the module:
```
Oct 07 04:55:44 fortitude kernel: evdi: [D] evdi_painter_crtc_state_notify:481 (dev=1) Notifying crtc state: 3
Oct 07 04:55:44 fortitude kernel: BUG: unable to handle kernel NULL pointer dereference at           (null)
Oct 07 04:55:44 fortitude kernel: IP: __list_add_valid+0x10/0x70
Oct 07 04:55:44 fortitude kernel: PGD 270c2c067 
Oct 07 04:55:44 fortitude kernel: P4D 270c2c067 
Oct 07 04:55:44 fortitude kernel: PUD 270c2d067 
Oct 07 04:55:44 fortitude kernel: PMD 0 
Oct 07 04:55:44 fortitude kernel: 
Oct 07 04:55:44 fortitude kernel: Oops: 0000 [#1] SMP
Oct 07 04:55:44 fortitude kernel: Modules linked in: evdi(O) uas usb_storage rfcomm fuse ccm xt_CHECKSUM ipt_MASQUERADE nf_nat_masquerade_ipv4 tun nf_conntrack_netbios_ns nf_conntrack_broadcast xt_CT ip6t_rpfilter
Oct 07 04:55:44 fortitude kernel:  x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel dell_laptop kvm dell_smbios dcdbas snd_hda_codec_realtek snd_hda_codec_generic ath10k_pci ath10k_core snd_soc_core irqbyp
Oct 07 04:55:44 fortitude kernel:  tpm_tis_core intel_hid rfkill pinctrl_sunrisepoint intel_lpss_acpi pinctrl_intel intel_lpss tpm sparse_keymap int340x_thermal_zone acpi_als kfifo_buf industrialio int3400_thermal
Oct 07 04:55:44 fortitude kernel: CPU: 1 PID: 17 Comm: kworker/1:0 Tainted: G           O    4.13.4-200.fc26.x86_64 #1
Oct 07 04:55:44 fortitude kernel: Hardware name: Dell Inc. XPS 13 9360/0115N5, BIOS 2.2.1 08/18/2017
Oct 07 04:55:44 fortitude kernel: Workqueue: events drm_mode_rmfb_work_fn [drm]
Oct 07 04:55:44 fortitude kernel: task: ffff9aaf346226c0 task.stack: ffffbaf700d34000
Oct 07 04:55:44 fortitude kernel: RIP: 0010:__list_add_valid+0x10/0x70
Oct 07 04:55:44 fortitude kernel: RSP: 0018:ffffbaf700d37be8 EFLAGS: 00010246
Oct 07 04:55:44 fortitude kernel: RAX: 0000000000000000 RBX: ffff9aaec1d57a00 RCX: 0000000c80000003
Oct 07 04:55:44 fortitude kernel: RDX: ffff9aaec1d57af0 RSI: 0000000000000000 RDI: ffff9aaec9521088
Oct 07 04:55:44 fortitude kernel: RBP: ffffbaf700d37be8 R08: ffff9aaec1d57af0 R09: ffff9aaf35803680
Oct 07 04:55:44 fortitude kernel: R10: 00000000000002bf R11: 0000000000000001 R12: ffff9aaec9521088
Oct 07 04:55:44 fortitude kernel: R13: 0000000000000000 R14: ffff9aaec1d57af0 R15: ffff9aaef1807000
Oct 07 04:55:44 fortitude kernel: FS:  0000000000000000(0000) GS:ffff9aaf3e480000(0000) knlGS:0000000000000000
Oct 07 04:55:44 fortitude kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Oct 07 04:55:44 fortitude kernel: CR2: 0000000000000000 CR3: 000000023a476000 CR4: 00000000003406e0
Oct 07 04:55:44 fortitude kernel: Call Trace:
Oct 07 04:55:44 fortitude kernel:  evdi_painter_send_event+0x32/0x70 [evdi]
Oct 07 04:55:44 fortitude kernel:  evdi_painter_crtc_state_notify+0x8e/0x100 [evdi]
Oct 07 04:55:44 fortitude kernel:  evdi_crtc_disable+0x27/0x50 [evdi]
Oct 07 04:55:44 fortitude kernel:  __drm_helper_disable_unused_functions+0x83/0xe0 [drm_kms_helper]
Oct 07 04:55:44 fortitude kernel:  drm_crtc_helper_set_config+0x141/0x980 [drm_kms_helper]
Oct 07 04:55:44 fortitude kernel:  ? __update_load_avg_se.isra.28+0x155/0x160
Oct 07 04:55:44 fortitude kernel:  ? __update_load_avg_se.isra.28+0x155/0x160
Oct 07 04:55:44 fortitude kernel:  __drm_mode_set_config_internal+0x65/0x110 [drm]
Oct 07 04:55:44 fortitude kernel:  drm_mode_set_config_internal+0x25/0x40 [drm]
Oct 07 04:55:44 fortitude kernel:  drm_crtc_force_disable+0x4b/0x70 [drm]
Oct 07 04:55:44 fortitude kernel:  drm_framebuffer_remove+0x2f7/0x3b0 [drm]
Oct 07 04:55:44 fortitude kernel:  ? __switch_to+0x1fc/0x4a0
Oct 07 04:55:44 fortitude kernel:  drm_mode_rmfb_work_fn+0x55/0x70 [drm]
Oct 07 04:55:44 fortitude kernel:  process_one_work+0x193/0x3c0
Oct 07 04:55:44 fortitude kernel:  worker_thread+0x4a/0x3a0
Oct 07 04:55:44 fortitude kernel:  kthread+0x125/0x140
Oct 07 04:55:44 fortitude kernel:  ? process_one_work+0x3c0/0x3c0
Oct 07 04:55:44 fortitude kernel:  ? kthread_park+0x60/0x60
Oct 07 04:55:44 fortitude kernel:  ret_from_fork+0x25/0x30
Oct 07 04:55:44 fortitude kernel: Code: c7 00 00 00 00 00 c7 40 08 00 00 00 00 e9 29 ff ff ff 90 90 90 90 90 90 90 90 90 55 48 8b 42 08 49 89 d0 48 89 e5 48 39 f0 75 19 <48> 8b 10 4c 39 c2 75 25 48 39 c7 74 34 48 
Oct 07 04:55:44 fortitude kernel: RIP: __list_add_valid+0x10/0x70 RSP: ffffbaf700d37be8
Oct 07 04:55:44 fortitude kernel: CR2: 0000000000000000
Oct 07 04:55:45 fortitude kernel: ---[ end trace 6ca27ba6b885f44d ]---
```

cacarr-pdxweb, 2017-10-08:
As of October 6, it was indicated that a refreshed release compatible with the newer kernels should be released within a[ matter of days.
](http://displaylink.org/forum/showthread.php?t=65577)

alhalama, 2017-10-11:
It appears that displaylink 1.4 was released today and this is needed to work with evdi 1.5.  I submitted a pull request #41 that updates the evdi version as well as the downloaded displaylink version.  

Note that I do have an issue with the resolution on one of my two external monitors after the upgrade though I am able to use both monitors on kernel 4.13.4-200.fc26.x86_64.

ssaavedra, 2017-10-17:
RPMs have been released now. Please open if PR #41 does not solve this.

closed 2017-10-17 by ssaavedra

RandieM, 2017-10-24:
Fantastic! It now works for me with kernel version 4.13.5-200.fc26.x86_64. Many thanks to everyone.

ShamimIslam, 2018-03-05:
Broken again. Rawhide compilation for displaylink exits with failed status. Evdi compile shows the following:

CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes" make -C module 
make[1]: Entering directory '/root/evdi/module'
make -C /lib/modules/4.15.6-200.fc26.x86_64/build M=$PWD
make[2]: Entering directory '/usr/src/kernels/4.15.6-200.fc26.x86_64'
  CC [M]  /root/evdi/module/evdi_connector.o
/root/evdi/module/evdi_connector.c: In function ‘evdi_best_single_encoder’:
/root/evdi/module/evdi_connector.c:91:6: error: passing argument 2 of ‘drm_encoder_find’ makes pointer from integer without a cast [-Werror=int-conversion]
      enc_id);
      ^~~~~~
In file included from ./include/drm/drm_modeset_helper_vtables.h:33:0,
                 from ./include/drm/drm_crtc_helper.h:43,
                 from /root/evdi/module/evdi_connector.c:16:
./include/drm/drm_encoder.h:217:35: note: expected ‘struct drm_file *’ but argument is of type ‘int’
 static inline struct drm_encoder *drm_encoder_find(struct drm_device *dev,
                                   ^~~~~~~~~~~~~~~~
/root/evdi/module/evdi_connector.c:87:9: error: too few arguments to function ‘drm_encoder_find’
  return drm_encoder_find(connector->dev,
         ^~~~~~~~~~~~~~~~
In file included from ./include/drm/drm_modeset_helper_vtables.h:33:0,
                 from ./include/drm/drm_crtc_helper.h:43,
                 from /root/evdi/module/evdi_connector.c:16:
./include/drm/drm_encoder.h:217:35: note: declared here
 static inline struct drm_encoder *drm_encoder_find(struct drm_device *dev,
                                   ^~~~~~~~~~~~~~~~
/root/evdi/module/evdi_connector.c:92:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: all warnings being treated as errors
make[3]: *** [scripts/Makefile.build:317: /root/evdi/module/evdi_connector.o] Error 1
make[2]: *** [Makefile:1513: _module_/root/evdi/module] Error 2
make[2]: Leaving directory '/usr/src/kernels/4.15.6-200.fc26.x86_64'
make[1]: *** [Makefile:57: module] Error 2
make[1]: Leaving directory '/root/evdi/module'
make: *** [Makefile:8: all] Error 2

ssaavedra, 2018-03-06:
@ShamimIslam duplicate of https://github.com/displaylink-rpm/displaylink-rpm/issues/55

Please, don't comment on *old* issues to raise *new* issues.