displaylink-rpm/displaylink-rpm #35: Any interest in adding support for Secure Boot via dkms module signing?

tartansandal, 2017-09-05:
Its pretty easy to get your system to support Secure Boot with dkms modules and the MOK keyring. All you need to do is:

1. Ensure Secure boot is turned on
1. Create an appropriate key pair
1. Add the new key to the MOK keyring
1. Reboot to install the key
1. Set up dkms to resign the evdi modules whenever they are rebuilt 

The first 4 steps can be easily covered by a bit of documentation (with official references).

The last step requires a configuration file, module signing script, and a bit of advice on how to do this all without defeating the whole purpose of Secure Boot.

I've sketch out a proposed implementation of the last step at 

https://github.com/tartansandal/displaylink-rpm/tree/support-for-module-signing

If there is any genuine interest in this, I can do some work on documenting the first 4 steps :-)

And, yes, there is a project to get evdi mainlined into the kernel, so it can be signed by official keys, but I think that may take a fair while.  It may require someone to code an open source reference implementation of the DisplayLink userspace service for this to get enough traction.

ssaavedra, 2017-09-16:
This is indeed very interesting, but I'm thinking this infrastructure should probably either be part of the distributions themselves or be a project in its own right, to glue not just this module, but all modules involving dkms. I feel it may be a bit wrong to edit dkms configuration files from a dkms package, imagine all dkms packages wanted to do this (in slightly different ways, just to have file conflicts).

I think it is very interesting to have this configuration, but maybe as a different package (and just query here whether that one is installed, and acknowledge that we are compliant with that). This issue is in fact being discussed elsewhere, and it may be a good idea to establish a prototype with this module, but at least it should be a different repo. If you'd like, I can create such a repo in this organization and we can collaborate on that (or maybe create a different organization for such an standalone purpose).

Of course, getting evdi mainlined is interesting, but that does not compete with proper key signing support from dkms, and the benefits we could provide by externalizing that from the displaylink-rpm package is that a more larger part of the community could benefit from that.

ssaavedra, 2017-09-16:
I'm very interested in what you think about this, and maybe I'm wrong and this would first fit here, but I can't help thinking we're putting more responsabilities on this single package (maybe create a sidecar package with the dkms signing features should be in place, if we take that path).

ssaavedra, 2017-09-16:
Also, if you want to help maintaining this effort, I think @hadess and I would be quite ok with it.

tartansandal, 2017-09-17:
Thanks for the feedback @ssaavedra. After some thought and experimentation, perhaps this is not such a great idea, or rather, there are a some issues that may be just too difficult to overcome. Also, I agree that this should not be so intimately coupled with the `displaylink-rpm` project.

My understanding is that the dkms configuration is only picked up dkms when the evid module is recompiled because the conf file name is evid.conf. Distribution packages maintaining different dkms modules can dump configuration files in /etc/dkms/, they just have to name them after their module so they are picked up, and there should be no file conflicts. Unless I am wrong about the above, it seems that this is the appropriate place for a distribution to place configuration tweaks to an upstream dkms module. (Here I'm thinking of `displaylink-rpm` as producing the distribution RPMs for the upstream `edvi` project).

However, as you have mention, this is all problematic: we would not want every DKMS package managing signing of modules in different ways, perhaps with different keys. So it would be better if there was a `dkms-signing` package/project that pulled all this together in a consistent fashion, and perhaps had pointers to `displaylink-rpm` as an example of how it might be used. You mentioned that others are discussing this? Pointers to any discussions you are aware of would be most welcome :-)  

My major issue with automating the signing is handling the various ways in which it can fail, and how to report that back to the user.  There is a big difference between updating the kernel from Gnome Software and from updating it via `dnf` on the command line. Gnome Software will put you into a reboot and install cycle where there is no opportunity for user interaction.  How do we feedback to the user that the signing failed? How do we let them know how to fix it? Can we setup some system configuration to ensure, on kernel upgrade, that a usb drive is mounted and to specify where the keys are located. Perhaps a post-installation service that could be set to prompt for module resigning (via a dbus message) after booting into a new kernel?

Currently I just have a script that I have to remember to run after a kernel upgrade. The fact that my external monitors stop working is sufficient to prompt me to do this.   Maybe a package with a simple script and some documentation would be sufficient? Other packages could require or suggest it.

tartansandal, 2017-09-17:
Also, happy to help out with maintaining this project. Looks like I'll be using this D3100 for at least another year :-)

hadess, 2017-09-17:
All the signing support needs to be in DKMS itself, not in this project. There's no way we can extend this project in a way that would support all distributions, there's a much better chance of this being supported if DKMS itself knows how to sign modules.

The only changes to this project would be to verify whether modules are signed, and throw a very clear error about it, pointing to the DKMS signing instructions if module insertion failed.

tartansandal, 2017-09-18:
@hadess yes, I definitely agree that signing needs to be detached from this project. I'm also not sure DKMS can handle it all by itself. I've just created a separate (and currently empty) project to track ideas on this (https://github.com/tartansandal/dkms-signing-tool).

Not sure how we would detect that module needs to be signed. We could check if Secure Boot is enabled, but the RPM may be being built on a machine distinct from the target. I pretty sure 'something else' needs to monitor dkms builds and prompt the user to sign the modules, especially since this may need to happen after a reboot.

tartansandal, 2017-09-24:
Thanks for all you thoughts on this. Definitely seems to be a much larger project than I had originally envisaged. I'm glad I asked before digging too deep into a PR :-). I think it may be best to leave it there.

closed 2017-09-24 by tartansandal