displaylink-rpm/displaylink-rpm #38: CentOS7 with DisplayLink

p1lgr1m, 2017-09-21:
I'm hoping that this helps someone else out. I spent way too much time on figuring this out this yesterday.

Here's how I got my DisplayLink dock going with CentOS 7 : https://github.com/displaylink-rpm/displaylink-rpm/issues/13#issuecomment-331164103

ssaavedra, 2017-09-22:
Maybe we should just remove the builds that are not available for CentOS7 so that users go to the latest "working" release?

Or even document that on the README, and close both these issues.

tartansandal, 2017-09-24:
I just spun up an instance of CentOS7 on AWS to get a handle on the compilation problem.  That system is fully updated and running kernel 3.10.  The minimum supported version for edvi is 3.16.  Yikes.  And CentOS8 is going to be a long wait.  We could a `legacy` target to the Makefile to checkout and build against the 1.2 tag like we do for `rawhide.

ssaavedra, 2017-09-25:
If anything, it should probably be a `centos7` target, or we will soon have
a naming problem in the Makefile.

Sounds ok for local compilation, does that evdi version work with the
latest userland binary, then? That's the real issue. Otherwise we should
just instruct CentOS users to download a previous version.

tartansandal, 2017-09-26:
Agree, it should be a `centos7` target if anything :-) 

Have played around with this a bit to get a handle on the issue. As @hadess pointed out in #35   userspace binary requires the newer libstdc++.   Now bundling libstdc++ is possible:  there is some RPM voodoo that allows you to exclude arbitrary 'provides'  so you can get around the rpmdb conflict problem that @ssaavedra noted. Since our package would providing the libstdc++ that it requires, we can safely ignore dependency resolution for that piece of the puzzle.

However, I'm not sure how to getting the userspace service and the driver to always use our bundled libstdc++. Is LD_LIBRARY_PATH enough? Will that work with DKMS?  

This is sounding like a lot of work. Can we justify that if the old version works reliably. Perhaps we are better off documenting that fact and removing the non-working builds from GitHub releases.

ssaavedra, 2017-09-26:
It is definitely a good idea to remove the non-working builds from GH Releases.

hadess, 2017-10-10:
See #13

closed 2017-10-10 by hadess