displaylink-rpm/displaylink-rpm #39: trouble following installation procedure on Fedora 26

danquantum, 2017-09-28:
sorry guys if this is a noob issue rather than a real one but I am unable to install the displaylink driver for Fedora 26 I currently run. I don't seem to be able to understand how this driver should be installed. I am trying to make an Asus MB169B+ display run with my Lenovo i510

1. I first downloaded the fedora-26-displaylink-1.4.1-4.x86_64.rpm and installed it using the Software Install and checked with sudo systemctl status displaylink.services and it didn't work, When checked with journal -xe I saw there was an error modprobe: FATAL: Module evdi not found in directory /lib/modules/4.11.11-300.fc26.x86_64

2. I then uninstalled the rpm, made a git repository in my file system and executed the following commands:

git init
git remote add github https://github.com/displaylink-rpm/displaylink-rpm
git pull github master
sudo make

It downloaded and compiled stuff but not service was installed, configured etc. I found the same rpm package inside the x86_64 folder which if I install it gives the same result as in the first case.

Please help me understand where I'm doing things wrong. Thank you for your patience

bivainis, 2017-09-29:
The only way I could make it work on kernel v4.12.11 was to compile the rawhide version from source, then installing it. You can check my comment here. Those are not the most detailed instructions but I applied most of the tricks that other people posted in their issues here.
https://github.com/displaylink-rpm/displaylink-rpm/issues/32#issuecomment-329731698

danquantum, 2017-09-29:
thanks for the reply and I will try your suggestion.  My fedora is not the latest, is only 4.11.11-300, but I think the rpm does not create all the necessary links. I tried to install directly with the driver for Ubuntu and it says it doesn't have the matching headers for the kernel, yet I checked and the correct kernel-headers package is installed. Damn.

danquantum, 2017-10-03:
I have tried this method and so far the result is the same. After I pulled the code with git, then built with make rawhide and installed the source rpm, I went to ~/rpmbuild/specs and added the global variables

%global _version 1.4.1
%global _daemon_version 1.3.54
%global _release 1.3.52

After this the rpmbuild -ba displaylink.spec executed with no problems. Still, when tried to install the rpm from ~/rpmbuild/RPMS/x86_64, I encounter the same error as before:

sudo rpm -ivh displaylink-1.4.1-1.3.52.x86_64.rpm
Preparing...                          ################################# [100%]
Updating / installing...
   1:displaylink-1.4.1-1.3.52         ################################# [100%]
Job for displaylink.service failed because the control process exited with error code.
See "systemctl  status displaylink.service" and "journalctl  -xe" for details.
warning: %post(displaylink-1.4.1-1.3.52.x86_64) scriptlet failed, exit status 1

When checked with journalctl -xe I see the following problem:

-- Unit displaylink.service has begun starting up.
Oct 03 12:48:20 penguin.nabla modprobe[6657]: modprobe: FATAL: Module evdi not found in directory /lib/modules/4.11.11-300.fc26.x86_64
Oct 03 12:48:20 penguin.nabla systemd[1]: displaylink.service: Control process exited, code=exited status=1
Oct 03 12:48:20 penguin.nabla systemd[1]: Failed to start DisplayLink Manager Service.
-- Subject: Unit displaylink.service has failed

I believe there should be a symbolic link in the /lib/modules/4.11.11-300/fc26/x86_64 to the evdi library binary found in /usr/libexec, but I am not sure and wouldn't want to break my system just by trying. Maybe someone can enlighten me, please? Thank you

tartansandal, 2017-10-03:
Why are you installing the src rpm?  That is unnecessary. Just install the
normal rpm from the appropriate arch directory.

danquantum, 2017-10-03:
Because I first tried to install the normal rpm from the appropriate arch directory with the exact same result. That's why I tried doing it from source rpm as @bivainis suggested but it seems it makes no difference. I am pretty much lost at this point.

tartansandal, 2017-10-03:
Ah okay. Appologies for being terse. Writing on phone. Yes, it would make
no difference.

Installing the x86_64 rpm is the way to go. Make sure you install it using
'dnf'.

Also, just need to check: Secure boot is off? Logging in with Xorg and not
Wayland?

Cheers,

Kal

tartansandal, 2017-10-03:
Also, rawhide rpm should have rawhide in its filename. Uninstall previous
versions first.

danquantum, 2017-10-03:
You said something new I didn't try, to install it using dnf. Maybe this is causing problems. 

I was on X but it was freezing the laptop when powering off and other issues so I switched back to Wayland until I am able to install the displaylink, then I can attempt to run it on X again and see if it works properly and what is causing the freezes (probably the nouveau driver).

I did uninstall each version after each attempt.

danquantum, 2017-10-03:
I tried with dnf, same result. modprobe error still in the journal, module evdi not found in directory /lib/modules/4.11.11-300.fc26.x86_64. Below is the transaction log from the installation, it gives a 'warning: %post(displaylink-1.4.1-5.rawhide.20171003.x86_64) scriptlet failed, exit status 1' at the end.

Last metadata expiration check: 0:08:34 ago on Tue 03 Oct 2017 02:03:46 PM EEST.
Dependencies resolved.
===================================================================================================================================================================================================================
 Package                                         Arch                                       Version                                                         Repository                                        Size
===================================================================================================================================================================================================================
Installing:
 displaylink                                     x86_64                                     1.4.1-5.rawhide.20171003                                        @commandline                                     3.1 M

Transaction Summary
===================================================================================================================================================================================================================
Install  1 Package

Total size: 3.1 M
Installed size: 8.5 M
Is this ok [y/N]: y
Downloading Packages:
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                                                                                                                           1/1 
  Installing       : displaylink-1.4.1-5.rawhide.20171003.x86_64                                                                                                                                               1/1 
  Running scriptlet: displaylink-1.4.1-5.rawhide.20171003.x86_64                                                                                                                                               1/1 
Job for displaylink.service failed because the control process exited with error code.
See "systemctl  status displaylink.service" and "journalctl  -xe" for details.
warning: %post(displaylink-1.4.1-5.rawhide.20171003.x86_64) scriptlet failed, exit status 1
Non-fatal POSTIN scriptlet failure in rpm package displaylink
Non-fatal POSTIN scriptlet failure in rpm package displaylink
  Verifying        : displaylink-1.4.1-5.rawhide.20171003.x86_64                                                                                                                                               1/1 

Installed:
  displaylink.x86_64 1.4.1-5.rawhide.20171003

danquantum, 2017-10-03:
ok, I managed to make it work. In my case it was an incompatibility with the kernel-devel version. Previously I have checked the kernel-headers version and was correct, but I did not check for kernel-devel and finally solved the problem with a system update. 

For any whom is interested in a short guide, here are the steps to make it work for my system. You can post them where you find appropriate if you think the short guide is useful for other noobs like me:

1. do a system update from the command line (or from Software application do a system update - worked for me, but I had previously issues with this)
sudo dnf update

2. create a directory somewhere you like, let's say in home:
$cd ~
$mkdir displaylink-rpm
$cd displaylink-rpm

3. pull the source code from the git repository

$git init
$git remote add github https://github.com/displaylink-rpm/displaylink-rpm
$git pull github master

4. you should have the rpm-build package installed. To be sure do:
$sudo dnf install rpm-build

5. build the rawhide version. This will create displaylink-1.4.1-5.rawhide.201710xx.src.rpm and two folders i386 and x85_64 containing the binary rpms for the corresponding archtiectures
$make rawhide

6. Go and install the rpm for your architecture
$cd x86_64
$sudo dnf install displaylink-1.4.1-5.rawhide.201710xx.x86_64.rpm

Normally this will check for dependencies (dkms, kernel-devel), install everything, including the displaylink.service

To check if displaylink is installed do $sudo systemctl status displaylink.service.  If there are errors, check with $journalctl -xe. In my case was a mismatch between versions of different packages, now is solved. 

To use de display you have to log in with Gnome on Xorg. It is not necessary to change default session settings, you can choose this from the login password screen.

Cheers

closed 2017-10-03 by danquantum