displaylink-rpm/displaylink-rpm #40: Building and Installing displaylink in RHEL7.3 fails with dependecy-issue Kernel > 4.7

knumskull, 2017-10-09:
Hi, I was trying to install the generated RPM from source on RHEL-7.3. But it is failing due to missing dependecies. Is there anything I can 

~~~
# cat /etc/redhat-release
Red Hat Enterprise Linux Workstation release 7.3 (Maipo)

# uname -a
Linux rhel73.example.com 3.10.0-514.21.2.el7.x86_64 #1 SMP Sun May 28 17:08:21 EDT 2017 x86_64 x86_64 x86_64 GNU/Linux
~~~

~~~
# sudo yum localinstall x86_64/displaylink-1.4.1-5.x86_64.rpm                                                                                                                                                                                                                                                                                                                                                                                                                 
Loaded plugins: changelog, fs-snapshot, priorities, product-id, refresh-packagekit, rhnplugin, rpm-warm-cache, search-disabled-repos, subscription-manager, verify, versionlock
There was an error communicating with RHN.
RHN channel support will be disabled.
Error communicating with server. The message was:
Unable to connect to the host and port specified
Repository google-chrome is listed more than once in the configuration
Repository google-talkplugin is listed more than once in the configuration
Examining x86_64/displaylink-1.4.1-5.x86_64.rpm: displaylink-1.4.1-5.x86_64
Marking x86_64/displaylink-1.4.1-5.x86_64.rpm to be installed
Resolving Dependencies
--> Running transaction check
---> Package displaylink.x86_64 0:1.4.1-5 will be installed
--> Processing Dependency: kernel > 4.7 for package: displaylink-1.4.1-5.x86_64
--> Processing Dependency: kernel-devel > 4.7 for package: displaylink-1.4.1-5.x86_64
--> Processing Dependency: dkms for package: displaylink-1.4.1-5.x86_64
--> Processing Dependency: libstdc++.so.6(CXXABI_1.3.8)(64bit) for package: displaylink-1.4.1-5.x86_64
--> Processing Dependency: libstdc++.so.6(GLIBCXX_3.4.20)(64bit) for package: displaylink-1.4.1-5.x86_64
--> Processing Dependency: libstdc++.so.6(GLIBCXX_3.4.21)(64bit) for package: displaylink-1.4.1-5.x86_64
--> Finished Dependency Resolution
Error: Package: displaylink-1.4.1-5.x86_64 (/displaylink-1.4.1-5.x86_64)
           Requires: libstdc++.so.6(GLIBCXX_3.4.20)(64bit)
Error: Package: displaylink-1.4.1-5.x86_64 (/displaylink-1.4.1-5.x86_64)
           Requires: dkms
Error: Package: displaylink-1.4.1-5.x86_64 (/displaylink-1.4.1-5.x86_64)
           Requires: libstdc++.so.6(GLIBCXX_3.4.21)(64bit)
Error: Package: displaylink-1.4.1-5.x86_64 (/displaylink-1.4.1-5.x86_64)
           Requires: libstdc++.so.6(CXXABI_1.3.8)(64bit)
Error: Package: displaylink-1.4.1-5.x86_64 (/displaylink-1.4.1-5.x86_64)
           Requires: kernel-devel > 4.7
           Installed: kernel-devel-3.10.0-514.16.1.el7.x86_64 (@production-rhel-x86_64-workstation-7.2)
               kernel-devel = 3.10.0-514.16.1.el7
           Installed: kernel-devel-3.10.0-514.21.1.el7.x86_64 (@production-rhel-x86_64-workstation-7.2)
               kernel-devel = 3.10.0-514.21.1.el7
           Installed: kernel-devel-3.10.0-514.21.2.el7.x86_64 (@production-rhel-x86_64-workstation-7.2)
               kernel-devel = 3.10.0-514.21.2.el7
Error: Package: displaylink-1.4.1-5.x86_64 (/displaylink-1.4.1-5.x86_64)
           Requires: kernel > 4.7
           Installed: kernel-3.10.0-514.16.1.el7.x86_64 (@production-rhel-x86_64-workstation-7.2)
               kernel = 3.10.0-514.16.1.el7
           Installed: kernel-3.10.0-514.21.1.el7.x86_64 (@production-rhel-x86_64-workstation-7.2)
               kernel = 3.10.0-514.21.1.el7
           Installed: kernel-3.10.0-514.21.2.el7.x86_64 (@production-rhel-x86_64-workstation-7.2)
               kernel = 3.10.0-514.21.2.el7
 You could try using --skip-broken to work around the problem
~~~

knumskull, 2017-10-09:
I attached the logfile of `make` for building the rpm package

[make.log](https://github.com/displaylink-rpm/displaylink-rpm/files/1368241/make.log)

knumskull, 2017-10-09:
~~~
# sudo rpm -ivh x86_64/displaylink-1.4.1-5.x86_64.rpm
error: Failed dependencies:
        dkms is needed by displaylink-1.4.1-5.x86_64
        kernel > 4.7 is needed by displaylink-1.4.1-5.x86_64
        kernel-devel > 4.7 is needed by displaylink-1.4.1-5.x86_64
        libstdc++.so.6(CXXABI_1.3.8)(64bit) is needed by displaylink-1.4.1-5.x86_64
        libstdc++.so.6(GLIBCXX_3.4.20)(64bit) is needed by displaylink-1.4.1-5.x86_64
        libstdc++.so.6(GLIBCXX_3.4.21)(64bit) is needed by displaylink-1.4.1-5.x86_64
~~~

hadess, 2017-10-10:
See #13

closed 2017-10-10 by hadess