displaylink-rpm/displaylink-rpm #43: Monitors Not working on D3100 Dock DisplayLink

RicardoAD, 2017-10-19:
Hi Guys,
Hope you can help me I just bought a D3100 dock and two 27'' monitors, but I cant make them to work on my Fedora 26, Can you help me with this.

I have Fedora 4.13.6-200.fc26.x86_64

**And displaylink**
#dnf info displaylink*
++++++++++++++++++++++++++++
Last metadata expiration check: 8 days, 2:46:58 ago on Wed 11 Oct 2017 09:11:02 AM BST.
Installed Packages
Name         : displaylink
Version      : 1.5.0
Release      : 1
Arch         : x86_64
Size         : 8.7 M
Source       : displaylink-1.5.0-1.src.rpm
Repo         : @System
From repo    : @commandline
Summary      : DisplayLink VGA/HDMI driver for DL-6xxx, DL-5xxx, DL-41xx and DL-3xxx adapters
License      : GPL v2.0, LGPL v2.1 and Proprietary
Description  : This adds support for HDMI/VGA adapters built upon the DisplayLink DL-6xxx,
             : DL-5xxx, DL-41xx and DL-3xxx series of chipsets. This includes numerous
             : docking stations, USB monitors, and USB adapters.
+++++++++++++++++++++

**This is what I get when querying the status of displaylink** 

$ systemctl status displaylink.service
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: active (running) since Thu 2017-10-19 11:47:25 BST; 15min ago
  Process: 9601 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 9602 (DisplayLinkMana)
    Tasks: 34 (limit: 4915)
   CGroup: /system.slice/displaylink.service
           └─9602 /usr/libexec/displaylink/DisplayLinkManager

Oct 19 11:47:25 macos.mask systemd[1]: Starting DisplayLink Manager Service...
Oct 19 11:47:25 macos.mask systemd[1]: Started DisplayLink Manager Service.

$ loginctl show-session 2 -p Type
Type=wayland

$ sudo dnf install -y edora-26-displaylink-1.5.0-1.x86_64.rpm
[sudo] password for mask: 
Last metadata expiration check: 1:28:29 ago on Thu 19 Oct 2017 10:13:22 AM BST.
Can not load RPM file: edora-26-displaylink-1.5.0-1.x86_64.rpm.
Could not open: edora-26-displaylink-1.5.0-1.x86_64.rpm
[mask@macos Mapas]$ sudo dnf install -y fedora-26-displaylink-1.5.0-1.x86_64.rpm
Last metadata expiration check: 1:28:43 ago on Thu 19 Oct 2017 10:13:22 AM BST.
Dependencies resolved.
======================================================================================================================
 Package                Arch              Version                                    Repository                  Size
======================================================================================================================
Installing:
 displaylink            x86_64            1.5.0-1                                    @commandline               3.1 M
Installing dependencies:
 dkms                   noarch            2.4.0-1.20170926git959bd74.fc26            updates-testing             81 k

Transaction Summary
======================================================================================================================
Install  2 Packages

Total size: 3.2 M
Total download size: 81 k
Installed size: 8.9 M
Downloading Packages:
dkms-2.4.0-1.20170926git959bd74.fc26.noarch.rpm                                        72 kB/s |  81 kB     00:01    
----------------------------------------------------------------------------------------------------------------------
Total                                                                                  35 kB/s |  81 kB     00:02     
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                              1/1 
  Installing       : dkms-2.4.0-1.20170926git959bd74.fc26.noarch                                                  1/2 
  Running scriptlet: dkms-2.4.0-1.20170926git959bd74.fc26.noarch                                                  1/2 
  Installing       : displaylink-1.5.0-1.x86_64                                                                   2/2 
  Running scriptlet: displaylink-1.5.0-1.x86_64                                                                   2/2 
  Running scriptlet: displaylink-1.5.0-1.x86_6   2/2 
  Verifying        : displaylink-1.5.0-1.x86_6   1/2 
  Verifying        : dkms-2.4.0-1.20170926git9   2/2 

Installed:
  displaylink.x86_64 1.5.0-1                         
  dkms.noarch 2.4.0-1.20170926git959bd74.fc26 

$ sudo  make
make -C /lib/modules/4.13.6-200.fc26.x86_64/build M=$PWD
make[1]: Entering directory '/usr/src/kernels/4.13.6-200.fc26.x86_64'
  AR      /usr/src/evdi-1.5.0/built-in.o
  CC [M]  /usr/src/evdi-1.5.0/evdi_drv.o
  CC [M]  /usr/src/evdi-1.5.0/evdi_modeset.o
  CC [M]  /usr/src/evdi-1.5.0/evdi_connector.o
  CC [M]  /usr/src/evdi-1.5.0/evdi_encoder.o
  CC [M]  /usr/src/evdi-1.5.0/evdi_main.o
  CC [M]  /usr/src/evdi-1.5.0/evdi_fb.o
  CC [M]  /usr/src/evdi-1.5.0/evdi_gem.o
  CC [M]  /usr/src/evdi-1.5.0/evdi_stats.o
  CC [M]  /usr/src/evdi-1.5.0/evdi_painter.o
  CC [M]  /usr/src/evdi-1.5.0/evdi_debug.o
  CC [M]  /usr/src/evdi-1.5.0/evdi_cursor.o
  CC [M]  /usr/src/evdi-1.5.0/evdi_ioc32.o
  LD [M]  /usr/src/evdi-1.5.0/evdi.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /usr/src/evdi-1.5.0/evdi.mod.o
  LD [M]  /usr/src/evdi-1.5.0/evdi.ko
make[1]: Leaving directory '/usr/src/kernels/4.13.6-200.fc26.x86_64'

tartansandal, 2017-10-19:
On 20 October 2017 at 02:32, RicardoAD <notifications@github.com> wrote:
>
> Hi Guys,
> Hope you can help me I just bought a D3100 dock and two 27'' monitors, but I cant make them to work on my Fedora 26, Can you help me with this.
>
> I have Fedora 4.13.6-200.fc26.x86_64
....
>
> $ loginctl show-session 2 -p Type
>
> Type=wayland

That's your problem. Displaylink drivers currently only work with Xorg.

Try logging out, click on the 'cog' icon to select an Xorg session,
and log back in.

RicardoAD, 2017-10-20:
Hello @tartansandal ,

Kal you were right about Xorg thanks, I realized that with my current kernel version kernel-4.13.8-200.fc26.x86_64 displaylink doesn't work and with kernel-4.13.8-200.fc26.x86_64 it is extremely slow
is there any trick to improve performance on the secondary monitor. I just connected one HDMI directly to the laptop and the performance is totally different to the one connected to the D3100 dock.

Thanks

tartansandal, 2017-10-21:
Glad that you've got it working. I'm experiencing similar slow-down with
the new kernel, and others are reporting difficulties with other non-Fedora
systems. Unfortunately these issues cannot be reproduced on Ubuntu systems,
so its quite hard to get the right people to see and fix them at the moment.

closed 2018-01-19 by ssaavedra