displaylink-rpm/displaylink-rpm #45: displaylink.service cant enable

hjaltioj, 2017-11-04:
[root@fedora-x1]# systemctl enable displaylink.service
The unit files have no installation config (WantedBy, RequiredBy, Also, Alias
settings in the [Install] section, and DefaultInstance for template units).
This means they are not meant to be enabled using systemctl.
Possible reasons for having this kind of units are:
1) A unit may be statically enabled by being symlinked from another unit's
   .wants/ or .requires/ directory.
2) A unit's purpose may be to act as a helper for some other unit which has
   a requirement dependency on it.
3) A unit may be started when needed via activation (socket, path, timer,
   D-Bus, udev, scripted systemctl call, ...).
4) In case of template units, the unit is meant to be enabled with some
   instance name specified.

Can anyone help me with this?

dilom, 2017-11-13:
Since my monitors don't show up with the latest kernel (4.13.11-200) update on F26 I tried starting the service manually and got the same message. 
Before the update one of the accounts on the same computer the monitor worked fine and on the other account it didn't work. The monitor used to show up on both accounts but not refresh on one of them.
Since the latest update the monitor no longer shows up at all anywhere.

dilom, 2017-11-13:
Ok, after reinstalling the rpm everything works again. But it seems I have to reinstall the rpm after each kernel update. Any ideas?

alhalama, 2017-12-11:
Instead of reinstalling the rpm, you could probably just use:

`dkms install evdi/1.5.0`

If that works, maybe there is an issue with your dkms configuration on your machine.  Maybe checking the status will give you more details:

`dkms status`

I know that I had an issue where a config file couldn't be found and that caused the module not to build automatically when I upgraded the kernel before.

ssaavedra, 2018-01-19:
Responding to OP, you **NEED NOT enable** the displaylink.service, as it will be automatically **started** and stopped whenever udev detects DisplayLink devices connected/disconnected.

Duplicate of https://github.com/displaylink-rpm/displaylink-rpm/issues/50#issuecomment-358955611 

@dilom if you are having a different issue, please open a different thread for easier tracking.

closed 2018-01-19 by ssaavedra