displaylink-rpm/displaylink-rpm #46: after install displaylink.service complain that modprobe evdi  status=1/FAILURE)

guitarjhk, 2017-11-21:
Used the "displaylink-rpm" - zip download - unpack - cd displaylink-rpm-master - run ./ci/fedora.sh no error - excep for the:
 Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2017-11-21 14:48:59 CET; 3s ago
  Process: 24912 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)

if looking at the "https://github.com/displaylink-rpm/displaylink-rpm" it say:
The package includes the Open Source evdi library. 

any suggestions 

thanks in advance 

John

ffgiff, 2017-12-29:
Duplicate of https://github.com/displaylink-rpm/displaylink-rpm/issues/48?

ssaavedra, 2018-01-19:
Duplicate, we'll keep track on #48 which is more active.

closed 2018-01-19 by ssaavedra