displaylink-rpm/displaylink-rpm #47: Fedora 27 RPM

mikerussellnz, 2017-11-22:
Now that Fedora 27 is released it would be nice to have a RPM for it. 
I am too scared to update my machine as displaylink in general is touchy.

alhalama, 2017-12-16:
@ssaavedra I see that you already added Fedora 27, is the reason there is no new release that the RELEASE variable wasn't updated with this change?

@mikerussellnz I have been running Fedora 27 for a few weeks now without issue.

LanceKroepke, 2017-12-16:
@alhalama I'm also running Fedora 27 without any major problems.

Howeever I do experience major mouse lag on external monitors

alhalama, 2017-12-18:
@LanceKroepke I haven't had issues with mouse lag and I am not sure what to suggest.

mikerussellnz, 2017-12-19:
I downgraded to 1.4.1-4 as I had very bad mouse lag on 1.5.   Downgrading to 1.4.1-4 fixed the issue for me.  This was on fedora 26.

LanceKroepke, 2017-12-19:
@mikerussellnz I remember installing F26 again + trying to downgrade myself but running into _package not found_ errors.

how did you manage to do so?

alhalama, 2017-12-19:
@LanceKroepke maybe issue [61 for evdi](https://github.com/DisplayLink/evdi/issues/61) is related.

LanceKroepke, 2017-12-20:
@alhalama don't think so since it happens for me whether the lid is closed or opened. Doesn't matter.

Ive also tried all their workarounds except the USB C 3.1 Dock. Which my Laptop sadly does not support!

@mikerussellnz Would you be so kind and give me some advice on how to successfully downgrade to 1.4.1?

ssaavedra, 2018-01-19:
Folks, there's a new version already released. Hopefully it would fix those lag problems. I'm closing this, but open another issue if you still need 1.4.1 for F27, that's easy if needed.

@alhalama sorry, I wasn't very active this last month around here. Yes, it was probably that.

closed 2018-01-19 by ssaavedra