displaylink-rpm/displaylink-rpm #48: Fedora 27 - displaylink.service: Failed with result 'exit-code'.

baulie, 2017-11-30:
I'm getting the error below, any suggestions? 

systemctl status displaylink.service
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2017-11-30 12:14:22 CET; 1s ago
  Process: 18447 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)

Nov 30 12:14:22 localhost.localdomain systemd[1]: displaylink.service: Failed with result 'exit-code'.

LanceKroepke, 2017-11-30:
what is the output of /sbin/modprobe evdi in terminal on its own?

baulie, 2017-11-30:
@LanceKroepke could you please let me know what you need exactly?
Process: 2866 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)

LanceKroepke, 2017-11-30:
@baulie well.. the output of /sbin/modprobe evdi if you type it into terminal..

can't get any more specific than this, sry.

baulie, 2017-11-30:
modprobe: FATAL: Module evdi not found in directory /lib/modules/4.13.9-300.fc27.x86_64

LanceKroepke, 2017-11-30:
whats the output of **uname -a** ?

did you update your fedora during/after the displaylink-rpm installation process?
did you restart after displaylink-rpm installation process?

what kernel-headers do you have currently installed?

mzwagerman, 2017-12-04:
I had the same error after upgrading to F27, for me the issue was dkms was confused about an old evdi install located here: /var/lib/dkms/evdi after removing that, this worked great.

bliemli, 2017-12-08:
Same problem here, though output of `modprobe evdi` is:
`modprobe: ERROR: could not insert 'evdi': Required key not available`. It seems this is because UEFI Secure Boot is enabled and the evdi kmod wasn't signed by a valid key?

LanceKroepke, 2017-12-08:
@bliemli thats another error message you've got.

@baulie, mzwagerman and me had some problems with kernel headers and dkms.

ffgiff, 2017-12-29:
'Me too'. Removing /var/lib/dkms/evdi didn't seem to help, i.e. the rpm install still complained, but manually running `dkms build` and `dkms install` got the module built and everything works normally now.

pir8don, 2018-01-12:
'me too' but I don't know the dkms commands to build and install the evdi module. kernel version is  4.13.9-300.fc27.x86_64. I tried removing and installing the displaylink fed26 rpm after removing /var/lib/dkms/evdi but no success. Any help appreciated!

LanceKroepke, 2018-01-13:
@pir8don 
- create /usr/sr/evdi-1.5.0 if it doesn't exist already (1.5.0 is the latest release)
- download latest evdi-release https://github.com/DisplayLink/evdi/releases 
- unzip it
- cd into unzipped evdi-folder
- _sudo make_
- cd into modules-folder
- _sudo mv * /usr/src/evdi-1.5.0/_
- _sudo dkms add -m evdi -v 1.5.0_
- _sudo dkms build -m evdi -v 1.5.0_
- _sudo dkms install -m evdi -v 1.5.0_

maybe you need a reboot after that. my displaylink.service usually just restars after those steps and i'm able to use my monitor even via GUI.

however i'm still experiencing major mouse lag & performance issues. you might consider living with it :P

pir8don, 2018-01-16:
Thanks heaps Lance.

ssaavedra, 2018-01-19:
Please take a look on whether the new tag solves the issue.

@bliemli about the key signing, if you're using Secure Boot you have to do the same you would do with any kernel module that does not come with your distro. In brief, enroll a MOK (Machine Owner's Key) in your trusted chain of certificates (you can do that with mokutil and shim). After you have done so, you'll need to sign the .ko with /usr/src/kernels/$(uname -r)/scripts/sign-file sha256 /path/to/your/mok.key /path/to/your/mok.crt /path/to/your/module.ko

khughitt, 2018-01-31:
Running into a similar issue when trying to build on CentOS 7 + kernel 4.15, however, the Invocation to make results in a missing target error:
```
$ sudo make
CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes" make -C module 
make[1]: Entering directory `/root/evdi-1.5.0/module'
make -C /lib/modules/4.15.0-1.el7.elrepo.x86_64/build M=$PWD
make[2]: Entering directory `/usr/src/kernels/4.15.0-1.el7.elrepo.x86_64'
make[3]: *** No rule to make target `/root/evdi-1.5.0/module/evdi_drv.o', needed by `/root/evdi-1.5.0/module/evdi.o'.  Stop.
make[2]: *** [_module_/root/evdi-1.5.0/module] Error 2
make[2]: Leaving directory `/usr/src/kernels/4.15.0-1.el7.elrepo.x86_64'
make[1]: *** [module] Error 2
make[1]: Leaving directory `/root/evdi-1.5.0/module'
make: *** [all] Error 2
```

Any thoughts? Any suggestions would be greatly appreciated!

bliemli, 2018-02-08:
@ssaavedra Thanks for the pointer. I realized this earlier but it nevertheless didn't work until I installed the current release v4.1.9. So, working now 👍

Yoyo310, 2018-02-08:
@bliemli I am facing exactly the same pb as you, but I installed the current release v4.1.9, and it still does not work:
[modprobe: ERROR: could not insert 'evdi': Operation not permitted]
Should I follow @ssaavedra advice too ? May you give some more details for this step for "less advance" people ?
Thank you very much !

ssaavedra, 2018-02-12:
@Yoyo310 You issue, "Operation not permitted" is not the same as "Required key not available". Yours may happen for several reasons, but the most obvious one is that you must be root to run modprobe. E.g., `sudo modprobe evdi`. Although all the steps must be done as root, so you can `sudo -i` and then after logging in, (with default bash settings your prompt should turn from `$` to `#`) you can run the rest of the commands.

ssaavedra, 2018-02-12:
@FranciscoSilveira you mean that the latest release is not working without these manual steps?

evdi/1.5.0 should be already bundled in this release.

Also, @LanceKroepke the `add` and `build` steps for dkms should not be necessary when running `install`, as `install` should run the previous two steps if needed (according to its man page), and that's why only the `install` step is performed in the `%post` section of the specfile.

If the `%post` script is failing, please check the install logs, as it should contain the `dkms install` output, which may hopefully point to the underlying cause for the problem.

Does the /usr/src/evdi-1.5.0 exist after installing the latest version on your machine? (It should, and it should also disappear after uninstalling the rpm, as well as being pointed out as owned by the package when running `rpm -qf /usr/src/evdi-1.5.0`)

If that doesn't happen, then something is wrong with the packaging.

Now, if the evdi-1.5.0 folder exists, but after installing you get an error like this:

```
Job for displaylink.service failed because the control process exited with error code.
See "systemctl  status displaylink.service" and "journalctl  -xe" for details.
warning: %post(displaylink-1.5.0-2.x86_64) scriptlet failed, exit status 1
Non-fatal POSTIN scriptlet failure in rpm package displaylink
```

Then, you may possibly have some missing kernel-headers or kernel-devel packages for all your installed kernel packages.

Check that you have as many kernel as kernel-headers and kernel-devel packages.

That is: `rpm -q kernel kernel-headers kernel-devel`. You should find a matching version for each package. If you have some kernel versions without a kernel-devel, the package will fail to run the installation script. If that happens, you can still reboot your machine and, if you have a kernel-headers for the kernel you want to boot in, dkms should still compile the evdi module on boot. If you don't want to reboot, `dkms install evdi/1.5.0 --kernel $(uname -r)` should do the trick.

Maybe the %postin script should just be that and avoid pre-installing evdi for all installed kernels.

Yoyo310, 2018-02-12:
@ssaavedra Thank you for your answer, the error is more explicit then I get `modprobe: ERROR: could not insert 'evdi': Required key not available`.  So should I enroll a MOK.. ? May you give more detais about that ? Thank you again.

ssaavedra, 2018-02-13:
@Yoyo310 this is a different issue, but here is some info. These should give you some pointers to look for further information.

- https://askubuntu.com/questions/755238/why-disabling-secure-boot-is-enforced-policy-when-installing-3rd-party-modules/797442#797442
- https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/7/html/system_administrators_guide/sect-signing-kernel-modules-for-secure-boot
- https://elrepo.org/tiki/SecureBootKey

FranciscoSilveira, 2018-02-17:
@ssaavedra That is correct, the latest release with evdi 1.5.0 did not run, if I recall correctly it said the evdi module was not found. ```modprobe evdi``` confirmed this. Compiling the package solved it. If you'd like I can uninstall the package and retry to give you some deeper info on this.

Yoyo310, 2018-02-20:
Thanks @RandieM, but I am running on kernel 4.13.9-300.fc27.x86_64, and displaylink.service works. My problem is that even if I disable xwayland, so using xorg, `xrandr`  does not recognize any of the outputs of the docking station I have plugged (USB3 or USB-C). Although `lsusb -d 17e9:` does see the USB connexion. Any help appreciated :)

RandieM, 2018-02-21:
@Yoyo310 I just referred to the solution proposed here by @LanceKroepke in another, possibly unrelated issue. I did not mean to suggest that the other issue had anything to do with the problem that you experience or that it can provide you with any solution to it. Hope you get it to work soon.

ssaavedra, 2018-02-21:
The original issue here should be solved by https://github.com/displaylink-rpm/displaylink-rpm/commit/08e62e217b3083a609d9dd2019ebe4882103fc83 as only building for your current kernel should be possible if you match the `Requires: kernel-devel`.

@Yoyo310 please open a *different* issue with a descriptive title (and description) for your problem, since cluttering a single issue with many problems leads to more complicated troubleshooting and communication. This way I can close this issue for its original problem.

If the original problem persists after installing v4.1.9-1 please open a new issue.

closed 2018-02-21 by ssaavedra

ssaavedra, 2018-03-12:
@Les-Linux does your problem bear any resemblance to `displaylink.service: Failed with result exit code'`?
I don't think so. Your issue is the same as https://github.com/displaylink-rpm/displaylink-rpm/issues/55 and even if you wish with all our heart that your issue will go away sooner it won't change the fact that the same rationale applies. Instead of commenting on an unrelated closed issue, you can add yourself to the watchers list of #55 (click the "Subscribe" button there on the right) and when there's a fix you'll get to know it.

Les-Linux, 2018-03-12:
@ssaavedra I didn't even notice that it was closed :). Will connect with #55. 

Thanks for the response.

nerydlg, 2018-05-10:
Hi I'm facing this problem 
the output of the modprobe evdi as root is: 
```# modprobe evdi
modprobe: ERROR: could not insert 'evdi': Operation not permitted
```
but it doesn't give any other information about why it isn't permitted, in the folder 
`/lib/modules/4.16.7-300.fc28.x86_64/extra`
I have a file evdi.ko.xz 

my kernel is: `4.16.7-300.fc28.x86_64`

when I try to start the displaylink service it throws me that error.

nerydlg, 2018-05-10:
@jkramarz  thanks for your answer, I already signed it with my own key, but now I'm getting this:
`modprobe: ERROR: could not insert 'evdi': Invalid argument`

any idea about how to fix it

Joyvalley, 2019-05-21:
> @pir8don
> 
>     * create /usr/src/evdi-1.5.0 if it doesn't exist already (1.5.0 is the latest release)
> 
>     * download latest evdi-release https://github.com/DisplayLink/evdi/releases
> 
>     * unzip it
> 
>     * cd into unzipped evdi-folder
> 
>     * _sudo make_
> 
>     * cd into (created) modules-folder. it contains the needed dkms.conf and other useful files
> 
>     * _sudo mv * /usr/src/evdi-1.5.0/_
> 
>     * _sudo dkms add -m evdi -v 1.5.0_
> 
>     * _sudo dkms build -m evdi -v 1.5.0_
> 
>     * _sudo dkms install -m evdi -v 1.5.0_
> 
> 
> maybe you need a reboot after that. my displaylink.service usually just restars after those steps and i'm able to use my monitor even via GUI.
> 
> however i'm still experiencing major mouse lag & performance issues. you might consider just living with it :P

I had the same Problem and could fix it with running the above commands with the latest release 1.6.1 thanks alot

brunotacca, 2019-10-07:
> @pir8don
>
>     * create /usr/src/evdi-1.5.0 if it doesn't exist already (1.5.0 is the latest release)
>
>     * download latest evdi-release https://github.com/DisplayLink/evdi/releases
>
>     * unzip [...]

I had a related problem and managed to solve by following these with the latest release 1.6.2.
I am using Ubuntu 18.04, my problem was, I bought a compatible DisplayLink adapter, followed the instructions found here (support.displaylink.com/knowledgebase/articles/684649) and installed successfully.
It worked, but after restart it stopped working giving the modprobe error after typing journalctl -xe.

Thanks.