displaylink-rpm/displaylink-rpm #49: why I installed it successfully? but my virtualbox fedora26 can't boot to screen?

ricardo0624, 2017-12-15:
before I install it,
I install kernel-headers-${uname -r} kernel-devel-${uname -r}
the I installed it successfully and without any mistake
type systemctl status displaylink
it show Running
but the virtual machine can't boot to screnn to login ,Is it normal ? should I ingore this?
I'm try to install to my pc, I'm afraid of this happen to my real pc

env is fedora26 kernel 4.11.8-300.fc26.x86_64 26

ricardo0624, 2017-12-15:
it show 
'''
Started command Scheduler.
Started Job spooling tools
Starting Hold until boot process finished up...
'''

then it is stop to boot to screen and freeze init this scense

ricardo0624, 2017-12-15:
I delete the virtualbox machine and create another one , like before I install this rpm
but this time ,
I use grub2-mkconfig -o /boot/efi/EFI/fedora/grub.cfg to generate grub.cfg
then the virtualbox machine can boot to screen ,and everything is all right
why?

ricardo0624, 2017-12-15:
Thanks fof your great jobs!
When I install it to my real PC, it's all right.
but another question,It can't start when I login to the system,I need to use the command.
systemctl start displaylink.service, and login out and then login again ,so it can works!
how to set ? let it auto start when I login to my system?
I have tried systemctl enabled diskplaylink.service ,but it can't, it said diskplaylink doesn't support it

ssaavedra, 2018-01-19:
The command should be `systemctl enable displaylink.service`. Hope it works.

closed 2018-01-19 by ssaavedra