displaylink-rpm/displaylink-rpm #50: systemctl enable displaylink.service

ilippert, 2018-01-08:
Hi,

thanks to this community I got displaylink to work on 4.14.11-300.fc27.x86_64.

Now the following gets an interesting reaction...
```
sudo systemctl enable displaylink.service
[sudo] password for ilippert: 
The unit files have no installation config (WantedBy, RequiredBy, Also, Alias
settings in the [Install] section, and DefaultInstance for template units).
This means they are not meant to be enabled using systemctl.
Possible reasons for having this kind of units are:
1) A unit may be statically enabled by being symlinked from another unit's
   .wants/ or .requires/ directory.
2) A unit's purpose may be to act as a helper for some other unit which has
   a requirement dependency on it.
3) A unit may be started when needed via activation (socket, path, timer,
   D-Bus, udev, scripted systemctl call, ...).
4) In case of template units, the unit is meant to be enabled with some
   instance name specified.
```

Any ideas?

ssaavedra, 2018-01-19:
Yes, it seems the service is missing a section like this:

```
[Install]
WantedBy=default.target
```

However, I recall this having worked before...

ssaavedra, 2018-01-19:
Actually, you NEED NOT enable the displaylink.service like that.

As you can see in 99-displaylink.rules for udev, whenever it is detected that your system has a DisplayLink device connected, it will automatically require the displaylink.service to be **started** (not *enabled*), and whenever it is removed, it will be stopped, so there is no intervention needed.

If for some reason you need to restart the displaylink userland binary, use **systemctl start** and **stop** commands, **not** enable.

closed 2018-01-19 by ssaavedra