displaylink-rpm/displaylink-rpm #52: issue with display ports on docking

DaveSys911, 2018-02-06:
hi running the following kernel and distro - 4.14.16-300.fc27.x86_64
have a dell xps 13,dell d6000 docking
installed the following rpm
fedora-26-displaylink-1.5.0-1.x86_64.rpm

for some reason the driver only works with hdmi ports on the docking but not the display ports

kindly assist

rgarcia89, 2018-02-06:
I do have the same issue wiht my Dell 7480 and the D6000 docking. None of my display port monitors is being recognized on the docking. 

I installed fedora-27-displaylink-1.5.0-2.x86_64.rpm on fedora version 4.14.16-300.fc27.x86_64.
Would be awesome if we could get this fixed.
thx

ssaavedra, 2018-02-07:
Do you get the DisplayPort device at xrandr (without parameters)?

rgarcia89, 2018-02-08:
@ssaavedra I was able to fix it by compiling and installing evdi separately. However, now from time to time it seems like the DisplayLinkManager looses the connection to the monitors. To fix this I have to re-plugin the USB C cable. After that everything is fine.

`displaylink.x86_64                                         1.5.0-2                                         @@commandline`
`/var/lib/dkms/evdi/kernel-4.14.16-300.fc27.x86_64-x86_64
/var/lib/dkms/evdi/1.5.0/4.14.16-300.fc27.x86_64
`
I also had sure that evdi is really compiled for the kernel that I am using:
`[raulgs@raulgs-xm1 ~]$ ll /lib/modules/$(uname -r)/extra/evdi.ko
-rw-r--r--. 1 root root 121896  6. Feb 18:31 /lib/modules/4.14.16-300.fc27.x86_64/extra/evdi.ko`

Is there something I can provide you to check why my screens are loosing the connection from time to time?

DaveSys911, 2018-02-08:
@rgarcia89 when you say "installed" evdi separately ,you mean without using dkms?

rgarcia89, 2018-02-08:
@DaveSys911 correct - I complied it. Before you are able to do so you have to install 

kernel-devel
libdrm-devel

after that you can run the make clean and make command

ssaavedra, 2018-02-08:
You may probably find more information in the output of `dmesg`.

rgarcia89, 2018-02-08:
See below the log of the dmesg after the last issue

> [47205.340697] usb 4-1.1: Disable of device-initiated U1 failed.
> [47205.347714] usb 4-1.1: Disable of device-initiated U2 failed.
> [47205.347802] cdc_ncm 4-1.1:1.5 enp59s0u1u1i5: unregister 'cdc_ncm' usb-0000:3b:00.0-1.1, CDC NCM
> [47205.385719] usb 4-1.1: Set SEL for device-initiated U1 failed.
> [47205.392875] usb 4-1.1: Set SEL for device-initiated U2 failed.
> [47205.691160] usb 4-1.1: reset SuperSpeed USB device number 3 using xhci_hcd
> [47205.717526] usb 4-1.1: Warning! Unlikely big volume range (=767), cval->res is probably wrong.
> [47205.717535] usb 4-1.1: [4] FU [Mic Capture Volume] ch = 2, val = -4592/7680/16
> [47205.720369] usb 4-1.1: Warning! Unlikely big volume range (=672), cval->res is probably wrong.
> [47205.720378] usb 4-1.1: [7] FU [Dell USB Audio Playback Volume] ch = 6, val = -10752/0/16
> [47205.746537] cdc_ncm 4-1.1:1.5: MAC-Address: 9c:eb:xxxxxx
> [47205.746547] cdc_ncm 4-1.1:1.5: setting rx_max = 16384
> [47205.746651] cdc_ncm 4-1.1:1.5: setting tx_max = 16384
> [47205.747520] cdc_ncm 4-1.1:1.5 usb0: register 'cdc_ncm' at usb-0000:3b:00.0-1.1, CDC NCM, 9c:eb:xxxxxx
> [47205.748559] usb 4-1.1: usbfs: process 7657 (DeviceWindowDis) did not claim interface 0 before use
> [47205.798000] cdc_ncm 4-1.1:1.5 enp59s0u1u1i5: renamed from usb0
> [47205.844318] IPv6: ADDRCONF(NETDEV_UP): enp59s0u1u1i5: link is not ready
> [47205.845897] IPv6: ADDRCONF(NETDEV_UP): enp59s0u1u1i5: link is not ready
> [47205.876060] cdc_ncm 4-1.1:1.5 enp59s0u1u1i5: network connection: disconnected
> [47207.812115] cdc_ncm 4-1.1:1.5 enp59s0u1u1i5: 100 mbit/s downlink 100 mbit/s uplink
> [47207.844152] cdc_ncm 4-1.1:1.5 enp59s0u1u1i5: network connection: connected
> [47207.844178] IPv6: ADDRCONF(NETDEV_CHANGE): enp59s0u1u1i5: link becomes ready

rgarcia89, 2018-02-09:
Still not been able to resolve it...
any idea?

Below I have attached a few more logs:

**Failed DisplayLink Service shows inactive**
> [raulgs@raulgs-xm1 ~]$ sudo systemctl status displaylink.service 
> [sudo] password for raulgs: 
> displaylink.service - DisplayLink Manager Service
> Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor p
> Active: inactive (dead)
> 
> Feb 09 08:26:10 raulgs-xm1 DisplayLinkManager[1266]: [libevdi] ioctl: drop_maste
> Feb 09 08:26:10 raulgs-xm1 DisplayLinkManager[1266]: [libevdi] ioctl: drop_maste
> Feb 09 08:26:10 raulgs-xm1 DisplayLinkManager[1266]: [libevdi] ioctl: drop_maste
> Feb 09 08:26:10 raulgs-xm1 DisplayLinkManager[1266]: [libevdi] ioctl: drop_maste
> Feb 09 08:26:10 raulgs-xm1 systemd[1]: Stopped DisplayLink Manager Service.
> Feb 09 08:26:10 raulgs-xm1 systemd[1]: Starting DisplayLink Manager Service...
> Feb 09 08:26:10 raulgs-xm1 systemd[1]: Started DisplayLink Manager Service.
> Feb 09 09:03:04 raulgs-xm1 systemd[1]: Stopping DisplayLink Manager Service...
> Feb 09 09:03:07 raulgs-xm1 DisplayLinkManager[6210]: [libevdi] ioctl: drop_maste
> Feb 09 09:03:07 raulgs-xm1 systemd[1]: Stopped DisplayLink Manager Service.

dmesg
> [ 4139.871595] usb 4-1.1: Disable of device-initiated U1 failed.
> [ 4139.878601] usb 4-1.1: Disable of device-initiated U2 failed.
> [ 4139.878702] cdc_ncm 4-1.1:1.5 enp59s0u1u1i5: unregister 'cdc_ncm' usb-0000:3b:00.0-1.1, CDC NCM
> [ 4139.914598] usb 4-1.1: Set SEL for device-initiated U1 failed.
> [ 4139.922127] usb 4-1.1: Set SEL for device-initiated U2 failed.
> [ 4140.216681] usb 4-1.1: reset SuperSpeed USB device number 3 using xhci_hcd
> [ 4140.241269] usb 4-1.1: Warning! Unlikely big volume range (=767), cval->res is probably wrong.
> [ 4140.241275] usb 4-1.1: [4] FU [Mic Capture Volume] ch = 2, val = -4592/7680/16
> [ 4140.242114] usb 4-1.1: Warning! Unlikely big volume range (=672), cval->res is probably wrong.
> [ 4140.242118] usb 4-1.1: [7] FU [Dell USB Audio Playback Volume] ch = 6, val = -10752/0/16
> [ 4140.268780] cdc_ncm 4-1.1:1.5: MAC-Address: 9c:eb:xxxxxx
> [ 4140.268787] cdc_ncm 4-1.1:1.5: setting rx_max = 16384
> [ 4140.268870] cdc_ncm 4-1.1:1.5: setting tx_max = 16384
> [ 4140.269364] cdc_ncm 4-1.1:1.5 usb0: register 'cdc_ncm' at usb-0000:3b:00.0-1.1, CDC NCM, 9c:eb:xxxxxx
> [ 4140.275340] usb 4-1.1: usbfs: process 6343 (DeviceWindowDis) did not claim interface 0 before use
> [ 4140.290266] cdc_ncm 4-1.1:1.5 enp59s0u1u1i5: renamed from usb0
> [ 4140.327714] IPv6: ADDRCONF(NETDEV_UP): enp59s0u1u1i5: link is not ready
> [ 4140.328047] IPv6: ADDRCONF(NETDEV_UP): enp59s0u1u1i5: link is not ready
> [ 4140.352345] usb 4-1.1: USB disconnect, device number 3
> [ 4140.355857] cdc_ncm 4-1.1:1.5 enp59s0u1u1i5: unregister 'cdc_ncm' usb-0000:3b:00.0-1.1, CDC NCM
> [ 4140.377134] evdi: [D] evdi_painter_disconnect:611 (dev=1) Disconnected from ffff9e7a3f072400
> [ 4140.377137] evdi: [D] evdi_detect:78 Painter is disconnected
> [ 4140.382887] evdi: [D] evdi_painter_disconnect:611 (dev=2) Disconnected from ffff9e7a3f073000
> [ 4140.382890] evdi: [D] evdi_detect:78 Painter is disconnected
> [ 4140.396848] evdi: [D] evdi_detect:78 Painter is disconnected
> [ 4140.397129] evdi: [D] evdi_detect:78 Painter is disconnected
> [ 4140.699251] evdi: [D] evdi_painter_crtc_state_notify:481 (dev=-1) Notifying crtc state: 3
> [ 4140.699255] evdi: [W] evdi_painter_send_crtc_state:387 Painter is not connected!
> [ 4140.705173] evdi: [D] evdi_painter_crtc_state_notify:481 (dev=-1) Notifying crtc state: 3
> [ 4140.705179] evdi: [W] evdi_painter_send_crtc_state:387 Painter is not connected!
> [ 4141.722543] evdi: [D] evdi_detect:78 Painter is disconnected
> [ 4141.722598] evdi: [D] evdi_detect:78 Painter is disconnected
> [ 4142.268132] usb 4-1.1: new SuperSpeed USB device number 5 using xhci_hcd
> [ 4142.280697] usb 4-1.1: New USB device found, idVendor=17e9, idProduct=6006
> [ 4142.280699] usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 4142.280700] usb 4-1.1: Product: Dell Universal Dock D6000
> [ 4142.280701] usb 4-1.1: Manufacturer: DisplayLink
> [ 4142.280702] usb 4-1.1: SerialNumber: 1707194358
> [ 4142.293621] usb 4-1.1: Warning! Unlikely big volume range (=767), cval->res is probably wrong.
> [ 4142.293623] usb 4-1.1: [4] FU [Mic Capture Volume] ch = 2, val = -4592/7680/16
> [ 4142.294102] usb 4-1.1: Warning! Unlikely big volume range (=672), cval->res is probably wrong.
> [ 4142.294103] usb 4-1.1: [7] FU [Dell USB Audio Playback Volume] ch = 6, val = -10752/0/16
> [ 4142.318430] cdc_ncm 4-1.1:1.5: MAC-Address: 9c:eb:xxxxxx
> [ 4142.318433] cdc_ncm 4-1.1:1.5: setting rx_max = 16384
> [ 4142.318531] cdc_ncm 4-1.1:1.5: setting tx_max = 16384
> [ 4142.318730] cdc_ncm 4-1.1:1.5 usb0: register 'cdc_ncm' at usb-0000:3b:00.0-1.1, CDC NCM, 9c:eb:xxxxxx
[... 29 lines cut ...]
> [ 4336.027065] usb 4-1.1: reset SuperSpeed USB device number 3 using xhci_hcd
> [ 4336.054078] usb 4-1.1: Warning! Unlikely big volume range (=767), cval->res is probably wrong.
> [ 4336.054087] usb 4-1.1: [4] FU [Mic Capture Volume] ch = 2, val = -4592/7680/16
> [ 4336.057091] usb 4-1.1: Warning! Unlikely big volume range (=672), cval->res is probably wrong.
> [ 4336.057100] usb 4-1.1: [7] FU [Dell USB Audio Playback Volume] ch = 6, val = -10752/0/16
> [ 4336.084470] cdc_ncm 4-1.1:1.5: MAC-Address: 9c:eb:xxxxxxx
> [ 4336.084478] cdc_ncm 4-1.1:1.5: setting rx_max = 16384
[... cut ...]
> [ 4338.184995] IPv6: ADDRCONF(NETDEV_CHANGE): enp59s0u1u1i5: link becomes ready

**xrandr always looks the same**
> [raulgs@raulgs-xm1 ~]$ xrandr --listproviders 
> Providers: number : 3
> Provider 0: id: 0x97 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 2 name:modesetting
> Provider 1: id: 0x6f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
> Provider 2: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
>

rgarcia89, 2018-02-09:
> [11485.190424] usb 4-1.1: Disable of device-initiated U1 failed.
> [11485.197788] usb 4-1.1: Disable of device-initiated U2 failed.
> [11485.197855] cdc_ncm 4-1.1:1.5 enp59s0u1u1i5: unregister 'cdc_ncm' usb-0000:3b:00.0-1.1, CDC NCM
> [11485.232423] usb 4-1.1: Set SEL for device-initiated U1 failed.
> [11485.239700] usb 4-1.1: Set SEL for device-initiated U2 failed.
> [11485.530725] usb 4-1.1: reset SuperSpeed USB device number 3 using xhci_hcd
> [11485.556889] usb 4-1.1: Warning! Unlikely big volume range (=767), cval->res is probably wrong.
> [11485.556896] usb 4-1.1: [4] FU [Mic Capture Volume] ch = 2, val = -4592/7680/16
> [11485.559315] usb 4-1.1: Warning! Unlikely big volume range (=672), cval->res is probably wrong.
> [11485.559321] usb 4-1.1: [7] FU [Dell USB Audio Playback Volume] ch = 6, val = -10752/0/16
> [11485.586621] cdc_ncm 4-1.1:1.5: MAC-Address: 9c:eb:xxxxxx
> [11485.586627] cdc_ncm 4-1.1:1.5: setting rx_max = 16384
> [11485.586855] cdc_ncm 4-1.1:1.5: setting tx_max = 16384
> [11485.587465] cdc_ncm 4-1.1:1.5 usb0: register 'cdc_ncm' at usb-0000:3b:00.0-1.1, CDC NCM, 9c:eb:xxxxxx
> [11485.596870] usb 4-1.1: usbfs: process 15772 (DeviceWindowDis) did not claim interface 0 before use
> [11485.626143] cdc_ncm 4-1.1:1.5 enp59s0u1u1i5: renamed from usb0
> [11485.670760] IPv6: ADDRCONF(NETDEV_UP): enp59s0u1u1i5: link is not ready
> [11485.671056] IPv6: ADDRCONF(NETDEV_UP): enp59s0u1u1i5: link is not ready
> [11485.688787] cdc_ncm 4-1.1:1.5 enp59s0u1u1i5: network connection: disconnected
> [11487.560866] cdc_ncm 4-1.1:1.5 enp59s0u1u1i5: 100 mbit/s downlink 100 mbit/s uplink
> [11487.592857] cdc_ncm 4-1.1:1.5 enp59s0u1u1i5: network connection: connected
> [11487.592991] IPv6: ADDRCONF(NETDEV_CHANGE): enp59s0u1u1i5: link becomes ready

rgarcia89, 2018-02-12:
I have figured out that the issue only happens when I have two monitors connected with the docking station. With only one, everything is working fine :-/

ssaavedra, 2018-02-12:
On previous versions of the DisplayLink driver using more than one monitor was not functional and a known issue. Now the known issues listed at https://support.displaylink.com/knowledgebase/articles/641668 seem not to include that case anymore, but maybe you'd have a better chance on the DisplayLink forums. This does not seem related to the packaging (and we are not associated to DisplayLink nor can change their upstream code). We could choose to package a different version of evdi if that would solve the issue.

rgarcia89, 2018-02-12:
I would be available to try a different evdi version if you could provide me one.
Thanks

ssaavedra, 2018-02-12:
You can try to compile against a different evdi version by downloading this repo and changing VERSION in the Makefile, or you can try to compile against the latest devel branch (or other branch by changing EVDI_DEVEL_BRANCH) by compiling with `make rawhide`.

If that doesn't work this may be a problem in the closed-source userspace binary. I never had multiple displays connected to a DisplayLink adapter, but maybe others here can comment.

ssaavedra, 2018-02-20:
@DaveSys911 have you managed to make this work?

@rgarcia89 how about you? You commented on a newer issue, which makes me wonder your issue involving this case was solved.

rgarcia89, 2018-02-20:
@ssavedra you won't believe how I managed to get the other issue which I reported working. I have changed the connector manufacturer. Changed from Dell to Lenovo after that it worked. 

This instead is a new issue. I had to boot the old kernel to be able to work with the connector. As the service is not starting with the newest kernel :(

ssaavedra, 2018-02-21:
@rgarcia89 so **this** issue here is considered resolved for you?

rgarcia89, 2018-02-21:
Correct

jabby, 2018-02-22:
@rgarcia89 which connector do you use now?

rgarcia89, 2018-02-22:
@jabby I am currently using this connector
https://www.i-tec-europe.eu/?t=3&v=510

jabby, 2018-02-22:
@rgarcia89 thanks.

ssaavedra, 2018-06-18:
I'm closing this since OP never got back.

closed 2018-06-18 by ssaavedra