displaylink-rpm/displaylink-rpm #54: Fedora 26 - Failed to enable unit: Unit file displaylink.service does not exist.

Jntshumaker, 2018-02-13:
Hello,

     When attempting to install and configure the displaylink RPM I am hitting an issue while trying to enable the displaylink service.  After reading through several posts I am a bit lost but may also be missing something very simple.

I have done the following:

- Downloaded the zip from the main github page

ssaavedra, 2018-02-20:
Source RPMs are not the final binaries. If you want the binary you can just grab the latest release at [the releases page](https://github.com/displaylink-rpm/displaylink-rpm/releases).

Please use the i386 or the x86_64 version according to your architecture, but bear in mind that `.src.rpm` files are the sources needed to build the binary RPMs. You were trying to install the sources of a package (which should probably create a ~/rpmbuild, at your user or root) instead of the binary.

If you have an idea to improve the wording of the README file to make this more clear, please feel free to comment back.

Jntshumaker, 2018-02-20:
@ssaavedra Thank you, I followed the instructions you provided however encountered an error which I will open another post if needed but will do my own searching to begin with.

As far as the readme, nowhere does it point to the release page instead it (at least to me) reads as if I need to create the rpm from the source.

closed 2018-02-20 by Jntshumaker