displaylink-rpm/displaylink-rpm #55: Driver no longer works in Fedora with Kernel 4.15.3

RandieM, 2018-02-20:
Dear everyone,

I am having trouble with the displaylink driver, since the last kernel update (`Linux fedora 4.15.3-300.fc27.x86_64 #1 SMP Tue Feb 13 17:02:01 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`).
Basically, I can no longer start the displaylink service, therefore my second screen does not work.

The version of the driver that I am using is `fedora-27-displaylink-1.5.0-2.x86_64.rpm` and since kernel 4.13 I have had to follow the instructions of @LanceKroepke in issue https://github.com/displaylink-rpm/displaylink-rpm/issues/48 on installing the latest evdi module.

Please let me know if I can provide you with more information or if you think this should submitted to DisplayLink/evdi instead.

Many thanks!

diarmaidocualain, 2018-02-20:
+1. I am having the same issue since the kernel update.

rgarcia89, 2018-02-20:
+1 same here. It seems like evdi is not working anymore.

ssaavedra, 2018-02-20:
Can you run `modprobe evdi` (as root)?

Does it work?

If the module does not exist, try running `sudo dkms install -m evdi -v 1.5.0` again on the new kernel and then `modprobe evdi` again.

Please report back if this works or not. If it doesn't, it is probably an upstream error.

ssaavedra, 2018-02-20:
And please avoid commenting just "+1" without adding further information, since it only makes the issue longer without adding real value. I don't want to be disrespectful about that, but if you want to be notified of further comments (and you want to include yourself as a participant in the conversation) you don't need to reply, you can also "react" to a post, or just subscribe yourself on the right in the web interface. Reactions allow us all to see people affected as a number and is more descriptive than counting +1's anyway.

*This message doesn't need a thanks/sorry/other reaction either (in case you wonder :slightly_smiling_face:)*

diarmaidocualain, 2018-02-20:
My apologies ssaavedra, I did not notice that we could add a reaction rather than add a +1. 
Below is my output of the suggested commands: 
`misty•~» uname -a                                                    [14:56:02]
Linux localhost.localdomain 4.15.3-300.fc27.x86_64 #1 SMP Tue Feb 13 17:02:01 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
misty•~» sudo modprobe evdi                                          [14:56:05]
[sudo] password for misty: 
modprobe: FATAL: Module evdi not found in directory /lib/modules/4.15.3-300.fc27.x86_64
misty•~» sudo dkms install -m evdi -v 1.5.0                          [18:45:30]

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j4 KERNELRELEASE=4.15.3-300.fc27.x86_64 all INCLUDEDIR=/lib/modules/4.15.3-300.fc27.x86_64/build/include KVERSION=4.15.3-300.fc27.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.15.3-300.fc27.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.5.0/build/make.log for more information.
misty•~» cat /var/lib/dkms/evdi/1.5.0/build/make.log                                                                                                                                                    [18:48:35]
DKMS make.log for evdi-1.5.0 for kernel 4.15.3-300.fc27.x86_64 (x86_64)
Tue Feb 20 18:46:00 GMT 2018
make KBUILD_VERBOSE=1 SUBDIRS=/var/lib/dkms/evdi/1.5.0/build SRCROOT=/var/lib/dkms/evdi/1.5.0/build CONFIG_MODULE_SIG= -C /lib/modules/4.15.3-300.fc27.x86_64/build modules
make[1]: Entering directory '/usr/src/kernels/4.15.3-300.fc27.x86_64'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
mkdir -p /var/lib/dkms/evdi/1.5.0/build/.tmp_versions ; rm -f /var/lib/dkms/evdi/1.5.0/build/.tmp_versions/*
make -f ./scripts/Makefile.build obj=/var/lib/dkms/evdi/1.5.0/build
  gcc -Wp,-MD,/var/lib/dkms/evdi/1.5.0/build/.evdi_drv.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-redhat-linux/7/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -funit-at-a-time -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_FXSAVEQ=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_CRC32=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_AVX512=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -pipe -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -DRETPOLINE -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-int-in-bool-context -O2 --param=allow-store-data-races=0 -DCC_HAVE_ASM_GOTO -Wframe-larger-than=2048 -fstack-protector-strong -Wno-unused-but-set-variable -Wno-unused-const-variable -fvar-tracking-assignments -g -pg -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_drv"'  -DKBUILD_MODNAME='"evdi"' -c -o /var/lib/dkms/evdi/1.5.0/build/evdi_drv.o /var/lib/dkms/evdi/1.5.0/build/evdi_drv.c
  gcc -Wp,-MD,/var/lib/dkms/evdi/1.5.0/build/.evdi_modeset.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-redhat-linux/7/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -funit-at-a-time -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_FXSAVEQ=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_CRC32=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_AVX512=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -pipe -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -DRETPOLINE -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-int-in-bool-context -O2 --param=allow-store-data-races=0 -DCC_HAVE_ASM_GOTO -Wframe-larger-than=2048 -fstack-protector-strong -Wno-unused-but-set-variable -Wno-unused-const-variable -fvar-tracking-assignments -g -pg -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wno [... cut ...]
  if [ "-pg" = "-pg" ]; then if [ /var/lib/dkms/evdi/1.5.0/build/evdi_modeset.o != "scripts/mod/empty.o" ]; then ./scripts/recordmcount  "/var/lib/dkms/evdi/1.5.0/build/evdi_modeset.o"; fi; fi;
make[1]: *** [Makefile:1512: _module_/var/lib/dkms/evdi/1.5.0/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.15.3-300.fc27.x86_64'
make: *** [Makefile:22: all] Error 2
misty•~»                                                                                                                                                                                                [18:55:18]
`

LanceKroepke, 2018-02-20:
i can verify this error.

the actually interesting lines would be:

```
echo >&2 " ERROR: Kernel configuration is invalid.";	
echo >&2 " include/generated/autoconf.h or include/config/auto.conf are missing.";
echo >&2 " Run 'make oldconfig && make prepare' on kernel src to fix it.";	

```

and

```
In file included from ./include/drm/drm_crtc.h:38:0,
from ./include/drm/drmP.h:69,
from /var/lib/dkms/evdi/1.5.0/build/evdi_connector.c:13:
./include/drm/drm_mode_object.h:117:25: note: expected ‘struct drm_file *’ but argument is of type ‘int’
struct drm_mode_object *drm_mode_object_find(struct drm_device *dev,
^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/evdi/1.5.0/build/evdi_connector.c:90:6: error: too few arguments to function ‘drm_mode_object_find’
drm_mode_object_find(connector->dev, enc_id,
^~~~~~~~~~~~~~~~~~~~
In file included from ./include/drm/drm_crtc.h:38:0,
from ./include/drm/drmP.h:69,
from /var/lib/dkms/evdi/1.5.0/build/evdi_connector.c:13:
./include/drm/drm_mode_object.h:117:25: note: declared here
struct drm_mode_object *drm_mode_object_find(struct drm_device *dev,

```

seems like there are some files missing.
@all old kernel still works fine. you should be able to chose it during bootsequence

ssaavedra, 2018-02-21:
Thanks for the troubleshoot. As the previous two comments read, the problem is related to evdi not being compatible with the newest kernel.

As we can see on the error trace, the signature for `drm_mode_object_find`may have changed on the kernel upstream, and evdi needs to update in order to work.

This is **not** a packaging problem. This is a problem upstream at https://github.com/displaylink/evdi and they need to release a newer version to restore compatibility with the kernel.

I will however keep this issue open because:

1. Once evdi is released upstream, we'll need to update the version here (please, reference this issue if you make the change in a PR)
2. Closed issues aren't shown by default on GitHub's issue tracker, and this needs to be fixed upstream.

rgarcia89, 2018-02-24:
I was able to get it working installing the following EVDI pull and using the latest Kernel 4.15.4 https://github.com/DisplayLink/evdi/pull/121

ssaavedra, 2018-02-24:
Ok, we'll wait for a bit for the merge upstream, but for the moment you can build this package yourselves against that evdi branch with:
`make rawhide EVDI_DEVEL_REPO=https://github.com/mintel/evdi.git`

petermalone, 2018-02-27:
What do we need to do after running make on https://github.com/mintel/evdi.git?

make rawhide doens't work. The Makefile doesn't have install commands either.

rgarcia89, 2018-02-27:
Make sure you have installed libdrm-devel.
1. Compile with sudo make all and sudo make clean without errors.
2. copy "modules" contents into /usr/src/evdi-1.50 (invented "future" version)
3. compile with sudo dkms install -m evdi/1.50 --kernelsourcedir /usr/src/linux-YOURKERNELVERSION
4. reboot
5. sudo modprobe evdi
Confirmed loaded module with lsmod | grep evdi

After that you should be fine

petermalone, 2018-02-28:
@rgarcia89 - those instructions worked, thank you!

@ssaavedra I'm not sure why but after installing the rpm that was built, and rebooting, evdi wasnt' loaded and modprobe couldn't find it.

I followed the instructions in the post above yours and they worked.

Thanks everyone!

pmanu64, 2018-03-01:
Hi,

drm_mode_object_find on kernel-4.15 needs 1 more argument
```
--- evdi_connector.c.old	2017-09-05 08:50:44.000000000 +0200
+++ evdi_connector.c	2018-03-01 13:50:21.647397548 +0100
@@ -85,9 +85,10 @@
 	int enc_id = connector->encoder_ids[0];
 	struct drm_mode_object *obj;
 	struct drm_encoder *encoder;
+	struct drm_file *file_priv;
 
 	obj =
-	    drm_mode_object_find(connector->dev, enc_id,
+	    drm_mode_object_find(connector->dev, file_priv, enc_id,
 				 DRM_MODE_OBJECT_ENCODER);
 	if (!obj)
 		return NULL;
```

ssaavedra, 2018-03-02:
@pmanu64 yes. Please, read the thread, this has already been stated at https://github.com/displaylink-rpm/displaylink-rpm/issues/55#issuecomment-368221234
Evdi code does not reside in this repo, and there is already https://github.com/DisplayLink/evdi/pull/121 with the changes.

Also, I don't see an advantage on using a stack variable if you are not going to use it later; and instead you should probably pass a null pointer instead, as you can see in the referenced PR.

closed 2018-03-27 by ssaavedra

ssaavedra, 2018-03-28:
Problem upstream has been solved (I'm thinking on keeping the tag for historical reference) and PR #58 addresses the changes. It is now tagged and being built by Travis.

commit dfd1e1ae (2018-03-27): Merge pull request #58 from mateusduboli/master

Update Displaylink to 4.2.29

Fixes #55, #57