displaylink-rpm/displaylink-rpm #59: External monitor on DisplayLink (DVI) not working

Jorijn, 2018-04-05:
Hi,

I have a MacBook Pro  with Fedora 27 (4.15.14-300.fc27.x86_64). I tried installing the latest release RPM and tried building the rawhide one. Both result in the same situation where the laptop screen briefly turns black when plugging in the USB cable but the external monitor doesn't turn on. 

It's a Fujitsu DisplayLink 3.0 Hub. To add: USB and Ethernet does work on the device, it's just the DVI monitor that isn't recognized.

Please send help.

jeffmcutter, 2018-04-06:
What does your status look like for `systemctl status displayport`?

If that status is good, an unofficial long shot, but I've been able to jump start mine into action with xrandr in the past.  With monitor unplugged do an xrandr and take not of what displays you have, then plug it in and do it again to figure out which one is your displayport.  After that, I used:

```
xrandr --addmode DVI-I-1-1 1920x1080
```

`DVI-I-1-1` will need to be whatever your monitor is on and the resolution should match your monitor.

Jorijn, 2018-04-09:
Installing Fedora was an experiment to see if I could back away from OS X but unfortunately that experiment failed for a multitude of reasons, where the inability of using my DisplayLink device was one of them. In the end I was unable to log in to my system which did it and triggered the restore of my backup.. 

Thanks for responding though, I'll keep your suggestion in mind when I try again in a year or so :-)

closed 2018-04-09 by Jorijn