displaylink-rpm/displaylink-rpm #60: Fedora 27 kernel update is breaking my displaylink service

jeffmcutter, 2018-04-11:
I was able to get my displaylink monitor working again recently and that's great.  I have been noticing it stops working each time I have a kernel update though.  The logs show:

```
-- Unit displaylink.service has begun starting up.
Apr 11 13:14:52 fedora27.example.com modprobe[3869]: modprobe: FATAL: Module evdi not found in directory /lib/modules/4.15.15-300.fc27.x86_64
Apr 11 13:14:52 fedora27.example.com systemd[1]: displaylink.service: Control process exited, code=exited status=1
Apr 11 13:14:52 fedora27.example.com systemd[1]: Failed to start DisplayLink Manager Service.
```

Removing and reinstalling the rpm fixes the issue.

Is it just me?

alhalama, 2018-04-12:
I'm assuming that dkms is failing to build the module automatically for the new kernel when you restart your machine.  You can use `dkms status` to see if there is an issue with DKMS on your machine.

Note that instead of reinstalling the rpm, you should be able to just use `dkms install evdi/1.5.0.2` to build the evdi module for the current kernel.

jeffmcutter, 2018-04-12:
dkms status showed that dkms was in fact having issues:
```
$ sudo dkms status
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/evdi/1.0.138/source/dkms.conf does not exist.
```
I found that was an old one and removed it, and now it seems a lot happier:
```
$ sudo dkms status
evdi, 1.5.0.2, 4.15.15-300.fc27.x86_64, x86_64: installed
$
```
Thanks!
-Jeff

closed 2018-04-12 by jeffmcutter

Lyalpha, 2018-06-28:
Also having a similar problem each time I upgrade the kernel (F28), but doesn't seem to be any issue with dkms, if there are suggestions for other things to check?

```
# uname -a
Linux astro 4.17.2-200.fc28.x86_64 #1 SMP Mon Jun 18 20:09:31 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

# dkms status
bbswitch, 0.8.0, 4.16.11-100.fc26.x86_64, x86_64: installed
bbswitch, 0.8.0, 4.16.12-300.fc28.x86_64, x86_64: installed
bbswitch, 0.8.0, 4.16.7-100.fc26.x86_64, x86_64: installed (WARNING! Missing some built modules!)
bbswitch, 0.8.0, 4.17.2-200.fc28.x86_64, x86_64: installed
evdi, 1.5.0.2, 4.15.16-200.fc26.x86_64, x86_64: installed (WARNING! Missing some built modules!)
evdi, 1.5.0.2, 4.15.17-200.fc26.x86_64, x86_64: installed (WARNING! Missing some built modules!)
evdi, 1.5.0.2, 4.16.12-300.fc28.x86_64, x86_64: installed
evdi, 1.5.0.2, 4.17.2-200.fc28.x86_64, x86_64: installed

# dkms install evdi/1.5.0.2
Module evdi/1.5.0.2 already installed on kernel 4.17.2-200.fc28.x86_64/x86_64

# modprobe evdi

# systemctl status displaylink.service 
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: active (running) since Thu 2018-06-28 13:28:38 BST; 4min 20s ago
  Process: 14660 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 14661 (DisplayLinkMana)
    Tasks: 31 (limit: 4915)
   Memory: 18.7M
   CGroup: /system.slice/displaylink.service
           └─14661 /usr/libexec/displaylink/DisplayLinkManager
```

Not sure if the `(WARNING! Missing some built modules!)` are anything related, but it was working on those versions at the time, as well as 4.16.12.

Just tried reinstalling the displaylink-rpm but external monitor still not working. I can see it in 'displays' settings but I can't enable it.

ssaavedra, 2018-07-16:
@Lyalpha not sure about your issue, but it's most likely a different one. If you haven't solved it, please open it in a new thread for easier tracking