displaylink-rpm/displaylink-rpm #61: "kernel-mt" in displaylink.spec

pjwelsh, 2018-04-25:
Shouldn't this be "kernel-ml" or better "kernel-lt" in %global in displaylink.spec?

ssaavedra, 2018-04-30:
You are probably right. It came originally from this patch by @hadess https://github.com/displaylink-rpm/displaylink-rpm/issues/13#issuecomment-336748311 and issue #13 is still open, which means it's not resolved.

I hadn't got the time to get back at that myself, but contributions are very much welcome.

Since this project should build against any kernel version compatible with evdi, I'm not sure which one should we target (maybe both?).

brianhussey, 2018-10-25:
I'd rather go with kernel-lt as well. It looks like just a typo.
kernel-lt is currently up to 4.4, kernel-ml is up to 4.19.
My question before trying to help out with this is, with these constraints
%{kernel_pkg_name} > 4.7, %{kernel_pkg_name}-devel > 4.7
Do we need to be using the mainline stable to satisfy those versions or not?

It looks like original supported version of Ubuntu was 14.04 which only got as far as the 4.4 kernel (https://wiki.ubuntu.com/Kernel/Support)

pjwelsh, 2018-10-25:
I'm not enough of a dev guy to know if there is a requirement for >4.7 kernel-*, but IIRC, I thought the original displaylink requirement was only like kernel 3.18+.
Thanks for looking into this.
PJ

pjwelsh, 2019-02-10:
Just to circle back, I believe the original commit simply has a typo. This is especially true based on the comment of the commit  https://github.com/displaylink-rpm/displaylink-rpm/issues/13#issuecomment-336748311:
"On the top of the patch above. Ideally, this would support both the kernel-lt and kernel-mt from elrepo.org. I'll let @ssaavedra take it from here."
There is no kernel-mt! There is a kerenl-ml, however.

closed 2020-06-22 by elguero