displaylink-rpm/displaylink-rpm #62: External monitors are not working.

pbaitule, 2018-05-02:
I am running Fedora 27 on a Dell Latitude 7490. I also have a Dell D6000 dock connected to the laptop using a USB Type-C cable. 

My peripherals, networking and power work fine when connected to the dock but my two 1080p monitors are blank. The monitors are connected via the displayports out from the dock.

I used this release - `fedora-27-displaylink-1.5.0.2-2.x86_64.rpm` to install the displaylink drivers.

Kernel version -
```
$ uname -r
4.16.4-200.fc27.x86_64
```

Displaylink service status -
```
$ systemctl status displaylink.service 
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: active (running) since Wed 2018-05-02 12:13:30 EDT; 5min ago
  Process: 1177 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 1183 (DisplayLinkMana)
    Tasks: 11 (limit: 4915)
   CGroup: /system.slice/displaylink.service
           └─1183 /usr/libexec/displaylink/DisplayLinkManager

May 02 12:13:30  systemd[1]: Starting DisplayLink Manager Service...
May 02 12:13:30  systemd[1]: Started DisplayLink Manager Service.
```

lsusb output -
```
$ lsusb
Bus 002 Device 004: ID 05e3:0620 Genesys Logic, Inc. 
Bus 002 Device 003: ID 17e9:6006 DisplayLink 
Bus 002 Device 002: ID 05e3:0620 Genesys Logic, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0c45:6715 Microdia 
Bus 001 Device 008: ID 0a5c:5832 Broadcom Corp. 
Bus 001 Device 009: ID 0461:4d22 Primax Electronics, Ltd 
Bus 001 Device 010: ID 06c4:c411 Bizlink International Corp. 
Bus 001 Device 006: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 011: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 007: ID 413c:2107 Dell Computer Corp. 
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

evdi module has loaded -
```
$ lsmod | grep evdi
evdi                   61440  0
drm_kms_helper        196608  2 evdi,i915
drm                   454656  17 evdi,i915,drm_kms_helper
```
My display settings only show the laptop's built-in display - 
[image]

Can anyone please help?

groovyman, 2018-05-02:
Salut

i am also running F27 notebook (acer) connected with a D6000 dock, and 
it is running.
At the office 2HD monitors are connected via hdmi and displayport to the 
docking station.
I have no clue, where you got the driver, but i checked out this project

https://github.com/displaylink-rpm/displaylink-rpm

and build the driver calling make.

The only problem i have is this one kde/plasma related:

https://bugzilla.redhat.com/show_bug.cgi?id=1570367

Christian

pbaitule, 2018-05-03:
@alhalama  I am using X11.

pbaitule, 2018-05-03:
If I look at the sound settings, the laptop can "see" the dock but still can't output an image on the displays.
[image]

ACiDGRiM, 2018-05-26:
https://ask.fedoraproject.org/en/question/96134/how-can-i-set-xorg-as-a-default-on-fedora-25/

This was resolved by disabling Wayland on Fedora 28 with kernel 4.16

pbaitule, 2018-05-30:
@ACiDGRiM Nope. Still no luck with Fedora 28 and kernel 4.16.12.

SunUltraSparc, 2018-05-31:
Thank guys, the wayland issues fixed my problem! Way to go!

pbaitule, 2018-05-31:
@SunUltraSparc Just disabling Wayland fixed it for you? Or did you do something else too? Can you share the steps to fix it?

pbaitule, 2018-05-31:
@SunUltraSparc I've followed the same steps. No luck.

This is in the output of `usb-devices` - 

```
T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 6
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev=04.16
S:  Manufacturer=Linux 4.16.12-300.fc28.x86_64 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  5 Spd=5000 MxCh= 4
D:  Ver= 3.10 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=05e3 ProdID=0620 Rev=88.16
S:  Manufacturer=GenesysLogic
S:  Product=USB3.1 Hub
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=02 Lev=02 Prnt=05 Port=00 Cnt=01 Dev#=  7 Spd=5000 MxCh= 0
D:  Ver= 3.10 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1
P:  Vendor=17e9 ProdID=6006 Rev=31.27
S:  Manufacturer=DisplayLink
S:  Product=Dell Universal Dock D6000
S:  SerialNumber=
C:  #Ifs= 7 Cfg#= 1 Atr=e0 MxPwr=8mA
I:  If#=0x0 Alt= 0 #EPs= 6 Cls=ff(vend.) Sub=00 Prot=03 Driver=(none)
I:  If#=0x1 Alt= 0 #EPs= 0 Cls=fe(app. ) Sub=01 Prot=01 Driver=(none)
I:  If#=0x2 Alt= 0 #EPs= 1 Cls=01(audio) Sub=01 Prot=20 Driver=snd-usb-audio
I:  If#=0x3 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=20 Driver=snd-usb-audio
I:  If#=0x4 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=20 Driver=snd-usb-audio
I:  If#=0x5 Alt= 0 #EPs= 1 Cls=02(commc) Sub=0d Prot=00 Driver=cdc_ncm
I:  If#=0x6 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=01 Driver=cdc_ncm

T:  Bus=02 Lev=02 Prnt=05 Port=01 Cnt=02 Dev#=  6 Spd=5000 MxCh= 4
D:  Ver= 3.10 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=05e3 ProdID=0620 Rev=88.17
S:  Manufacturer=GenesysLogic
S:  Product=USB3.1 Hub
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
```

Could `Driver=(none)` be the problem?

SunUltraSparc, 2018-06-01:
Mine is very similar, I did install the Ubuntu drivers
<http://www.displaylink.com/downloads/ubuntu> first. There might be a
library you are missing. Mine is also a portable monitor not a docking
station.

T:  Bus=03 Lev=01 Prnt=01 Port=01 Cnt=01 Dev#=  4 Spd=5000 MxCh= 0
D:  Ver= 3.10 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1
P:  Vendor=17e9 ProdID=ff03 Rev=31.93
S:  Manufacturer=DisplayLink
S:  Product=MB168B
S:  SerialNumber=
C:  #Ifs= 2 Cfg#= 1 Atr=80 MxPwr=504mA
I:  If#=0x0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=03 Driver=usbfs
I:  If#=0x1 Alt= 0 #EPs= 0 Cls=fe(app. ) Sub=01 Prot=01 Driver=(none)

pbaitule, 2018-06-01:
Reinstalled everything again. And it works now!
Still have the issue of external monitors turning off randomly with the laptop lid closed but at least the displays work off of the hub now.

Closing the issue.

closed 2018-06-01 by pbaitule

SunUltraSparc, 2018-06-01:
Congrats!

wottopoteat, 2018-08-11:
I have tried on a few releases of fedora and the best I get is blue and pink bands down the display. I have seen the display work on a windows machine so I know it is good. 

The latest attempt was on a Lenovo T420 with F28 and the packaged x86_64 rpm.

Any suggestions?