displaylink-rpm/displaylink-rpm #63: RPM builds and Installs on Fedora 28 but does not install evdi module nor displaylink manager ?

viktormadarasz, 2018-06-27:
Hi

I have two machines where I m trying to make this work under Fedora 28
lenovo Thinkpad Ultra USB 3 dock with HDMI and Displayport out ( trying to use HDMI Out)

Both machines are running Nvidia propietary drivers.

RPM Builds and installs without error message on both but evdi module and displaylink manager service does not seem to be installed

$modprobe evdi
modprobe: FATAL: Module evdi not found in directory /lib/modules/4.17.2-200.fc28.x86_64

systemctl does not recognize a service as displaylink.service

Viktor

sdjohnson72, 2018-06-29:
Viktor,
I have found that the rpm install is a false success.  I have had to go look at the displaylink.log file in /var/log/displaylink to see what is still required for building of the module and service.  From memory I can recall rpm-build needs to be installed and libdrm-devel.  I am fighting a Fedora 28 install issue of my own on a separate issue.  If you get this working, please let me know.

Thanks
Shane

viktormadarasz, 2018-06-30:
sdjohnson72 

Mine after building the src rpm and installing it does not even create a displaylink log file in /var/logs on Fedora 28 ? strange

[viktormadarasz@fedoraws displaylink-rpm]$ sudo rpm -ivh displaylink-1.5.0.2-2.src.rpm
Updating / installing...
   1:displaylink-1.5.0.2-2            ################################# [100%]

[viktormadarasz@fedoraws log]$ cat /var/log/displaylink
cat: /var/log/displaylink: No such file or directory

viktormadarasz, 2018-06-30:
**it tells me** 

sudo dnf reinstall displaylink-1.5.0.2-2.src.rpm

Failed to synchronize cache for repo 'username-vte291-ng', disabling.
Failed to synchronize cache for repo 'bintray-tvheadend-stable', disabling.
Last metadata expiration check: 0:00:24 ago on Sat 30 Jun 2018 21:34:59 CEST.
Package displaylink-1.5.0.2-2.src not installed, cannot reinstall it
.No match for argument: displaylink-1.5.0.2-2.src.rpm

**if i try to install via dnf it tells me:**

sudo dnf install displaylink-1.5.0.2-2.src.rpm

Failed to synchronize cache for repo 'username-vte291-ng', disabling.
Failed to synchronize cache for repo 'bintray-tvheadend-stable', disabling.
Last metadata expiration check: 0:00:00 ago on Sat 30 Jun 2018 21:37:15 CEST

Error: Will not install a source rpm package (displaylink-1.5.0.2-2.src).

viktormadarasz, 2018-06-30:
> What happens if you run dnf reinstall ./displaylink-1.5.0.2-2.src.rpm from the directory where your rpm is saved to?

it tells me

sudo dnf reinstall displaylink-1.5.0.2-2.src.rpm
Failed to synchronize cache for repo 'username-vte291-ng', disabling.
Failed to synchronize cache for repo 'bintray-tvheadend-stable', disabling.
Last metadata expiration check: 0:00:24 ago on Sat 30 Jun 2018 21:34:59 CEST.
Package displaylink-1.5.0.2-2.src not installed, cannot reinstall it.
No match for argument: displaylink-1.5.0.2-2.src.rpm

viktormadarasz, 2018-06-30:
if i try to install via dnf it tells me:

sudo dnf install displaylink-1.5.0.2-2.src.rpm
Failed to synchronize cache for repo 'username-vte291-ng', disabling.
Failed to synchronize cache for repo 'bintray-tvheadend-stable', disabling.
Last metadata expiration check: 0:00:00 ago on Sat 30 Jun 2018 21:37:15 CEST.
Error: Will not install a source rpm package (displaylink-1.5.0.2-2.src).

sdjohnson72, 2018-07-02:
Viktor,  It appears that file only installs the following:
`rpm -ql ./displaylink-1.5.0.2-2.src.rpm
20-displaylink.conf
99-displaylink.rules
DisplayLink USB Graphics Software for Ubuntu 4.2.29.zip
displaylink-sleep-extractor.sh
displaylink.service
displaylink.spec
v1.5.0.2.tar.gz
`
I use the `displaylink-1.5.0.2-2.x86_64.rpm` file.  This _should_ install everything.

viktormadarasz, 2018-07-02:
Where can I get  displaylink-1.5.0.2-2.x86_64.rpm file?

mine only creates the source rpm file

any chance to share? would work on F28?

sdjohnson72, 2018-07-02:
When I built from source it was in the x86_64 directory from where I ran make from.

ssaavedra, 2018-07-02:
I have just committed a change to add Fedora 28 to the build matrix in Travis, it should get built in the tag that has just been pushed to the repo.

ssaavedra, 2018-07-02:
If the rpm built by Travis solves your problems, then it's a problem with your build environment. If you get to identify which is it, then a small excerpt for inclusion in the README would be very appreciated so that other people can avoid repeating the mistake.

If the rpm built by Travis does not solve your issues, then we've got a problem with displaylink-rpm and Fedora 28 which should need more research :)

viktormadarasz, 2018-07-04:
Ok i got further and installed the normal rpm it created , it launches the external display when i connect to it but the screen resolution is not recognized correctly (nor offered the proper ones) also the main screen does a scrolling motion towards the left till a while before the cursor passes over to the external screen on the left ( resolution set and being correct on laptops main screen but still doing this mentioned effect *laptop main screen  is the left screen*

is there any irc or discord chat ( where perhaps i can link video or picture ) to continue this discussion as the issue itself can be closed as it indeed builds correctly on F28 its just something with my config on my side

THanks 

Viktor

closed 2018-07-04 by viktormadarasz