displaylink-rpm/displaylink-rpm #64: No providers in Fedora 28 after apparent successful install.

sdjohnson72, 2018-06-29:
Hello, 
I have been trying to get this working for a while now but I am stuck.  
Here is my process.  I downloaded the source and extracted it.  I ran make and after I installed all the dependencies, It completed successfully.  I have .rpm files in i386 and the x86_64 directories.  I ran dnf install ./displaylink-1.5.0.2-2.x86_64.rpm.  Then after it said it installed correctly but actually didn't due to 
 more dependencies that weren't required for rpm-build and libdrm that were not required by the rpm, it said it installed correctly and appears to be so.  

`$ systemctl status displaylink
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor >
   Active: active (running) since Fri 2018-06-29 09:46:43 MDT; 40min ago
  Process: 1008 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 1020 (DisplayLinkMana)
    Tasks: 11 (limit: 4620)
   Memory: 12.2M
   CGroup: /system.slice/displaylink.service
           └─1020 /usr/libexec/displaylink/DisplayLinkManager

Jun 29 09:46:43 localhost systemd[1]: Starting DisplayLink Manager Service...
Jun 29 09:46:43 localhost systemd[1]: Started DisplayLink Manager Service.
`
` lsmod | grep evdi
evdi                   61440  0
drm_kms_helper        196608  2 evdi,i915
drm                   458752  6 drm_kms_helper,evdi,i915
[shane@localhost x86_64]$ 
`

But when I have my device plugged in, there is only the original modesetting device.

`[shane@localhost x86_64]$ xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x42 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 1 associated providers: 0 name:modesetting
[shane@localhost x86_64]$ 
`
The device is present but has no drivers associated with it. (usb-devices output)

`T:  Bus=07 Lev=01 Prnt=01 Port=01 Cnt=01 Dev#=  2 Spd=5000 MxCh= 0
D:  Ver= 3.10 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1
P:  Vendor=17e9 ProdID=6000 Rev=30.00
S:  Manufacturer=DisplayLink
S:  Product=USB 3.0 Dual 4K HDMI adapter
S:  SerialNumber=GWCD00160902353
C:  #Ifs= 5 Cfg#= 1 Atr=80 MxPwr=504mA
I:  If#=0x0 Alt= 0 #EPs= 6 Cls=ff(vend.) Sub=00 Prot=03 Driver=(none)
I:  If#=0x1 Alt= 0 #EPs= 0 Cls=fe(app. ) Sub=01 Prot=01 Driver=(none)
I:  If#=0x2 Alt= 0 #EPs= 1 Cls=01(audio) Sub=01 Prot=20 Driver=snd-usb-audio
I:  If#=0x3 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=20 Driver=snd-usb-audio
I:  If#=0x4 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=20 Driver=snd-usb-audio
[shane@localhost x86_64]$ 
`
Here is my /etc/X11/xorg.conf.d/20-displaylink.conf.  The settings are per the recommendations I found on the displaylink site.
`[shane@localhost x86_64]$ cat /etc/X11/xorg.conf.d/20-displaylink.conf 
Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "kmsdev" "/dev/dri/card0"
    Option "PageFlip" "off"
    Option "SWCursor" "on"
    Option "ShadowFB" "true"
EndSection

Section "Device"
	Identifier "DisplayLink"
	Driver "modesetting"
	Option "PageFlip" "false"
EndSection

Section "Device"
    Identifier "USB3"
    BusID "USB"
    Driver "modesetting"
    Option "kmsdev" "/dev/dri/card1"
    Option "PageFlip" "off"
    Option "SWCursor" "on"
    Option "ShadowFB" "true"
EndSection

Section "Device"
    Identifier "USB3"
    BusID "USB"
    Driver "modesetting"
    Option "kmsdev" "/dev/dri/card2"
    Option "PageFlip" "off"
    Option "SWCursor" "on"
    Option "ShadowFB" "true"
EndSection

Section "Device"
    Identifier "USB3"
    BusID "USB"
    Driver "modesetting"
    Option "kmsdev" "/dev/dri/card3"
    Option "PageFlip" "off"
    Option "SWCursor" "on"
    Option "ShadowFB" "true"
EndSection
`
Please let me know what I am missing on this.  I would really like to get this working.

Thank you in advance for your help.

Shane

ssaavedra, 2018-07-02:
Do you think your issue is the a duplicate of #63 or are there any differences that I haven't spotted?
Please, take a look at the latest release and see if the rpm built by Travis for Fedora 28 solves your issue, instead of your build.

sdjohnson72, 2018-07-02:
Not duplicated.  In mine I was able to get the modules and service to install where in that issue they were not getting installed by the src.rpm file.

mrprompt, 2018-07-03:
In Fedora 28 with Xorg, its fine.

sdjohnson72, 2018-07-03:
I just removed the displaylink package from my system and tried to reinstall via the Travis CL rpm and I still don't show any screens or providers available for use.  I am using X11 so I am not sure why my results are different then MrPrompt.  Any help is appreciated on this.

mrprompt, 2018-07-03:
@sdjohnson72 to work for me, on the login window, i choose "Gnome with Xorg", you tried this too?

sdjohnson72, 2018-07-03:
I don't have Gnome, I am using Xfce.  From what I have read, the displaylink driver is supposed to work with Xfce. I don't think gnome vs xfce would make a difference as xrandr, which should drive the screens independent of the windows manager, doesn't show any additional screens, or providers.  Also when I run `usb-devices` the displayport device is show with no drivers.  If I am wrong on this, please let me know.   

Thank you
Shane

mrprompt, 2018-07-03:
I do installed xfce desktop in my computer, the unique difference between this and Gnome is: by default, XFCE disable the monitors from adapter. Go to preferences, displays and try enable this.

sdjohnson72, 2018-07-03:
Thank you for testing that for me.  I do not have the option of enabling any other monitors though.
Can you please run `xrandr` and `xrandr --listproviders` and compare it to my original post to see if you output differs from mine?  Especially the `--listproviders` as that is critical to this working from everything I have read.

Shane

mrprompt, 2018-07-04:
Are you using Xorg or Wayland?

sdjohnson72, 2018-07-05:
I am using X11.  This is my output:
`[shane@localhost x86_64]$ xrandr --listproviders Providers: number : 1 Provider 0: id: 0x42 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 1 `
This is only the built in VGA connection for this system. 
So this leads me to question why yours is working and mine is not.
Can you list the output of `usb-devices` for just the displaylink adapter?  

Thanks
Shane

sdjohnson72, 2018-07-12:
I still can't use my device.  Does anyone have any ideas or suggestions on how to proceed troubleshooting this?  At this time all I can think of is that something is missing?  
Can someone please post their results from a working Fedora 28 system for the following commands? 
`lsmod | grep evdi`
`systemctl status displaylink.service`

Thank you.

Shane

groovyman, 2018-07-12:
Salut Shane,
i am also running F28 on a old Acer notebook, using a intel x11 driver. The notebook is connected with a Dell UBS-Docking station serving 2 additional monitors. I am running KDE as a desktop so i am running X11 too. The only problem i recognized might be a timing issue on the very first desktop-start that may fail and throws you back to the login-dialogue, but the secound time the system is coming up ... dare;-)

Here are some configs:

```
$ lsmod | grep evdi
evdi                   61440  4
drm_kms_helper        196608  3 evdi,i915,nouveau
drm                   458752  18 drm_kms_helper,evdi,i915,ttm,nouveau
```
```
xrandr --listproviders
Providers: number : 4
Provider 0: id: 0xd1 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 3 name:modesetting
Provider 1: id: 0xa1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x71 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x3f cap: 0x5, Source Output, Source Offload crtcs: 0 outputs: 0 associated providers: 1 name:modesetting
```
You may try to look into the kernel log, that might give you a hint, why kms did not come up.

sdjohnson72, 2018-07-12:
Groovyman,
Thank you for posting that.  I think kms is running....
`[root@localhost ~]# lsmod | grep evdi
evdi                   61440  0
drm_kms_helper        196608  2 evdi,i915
drm                   458752  6 drm_kms_helper,evdi,i915
`
I still only have the one provider for the internal Intel video card though...
`[root@localhost ~]# xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x42 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 1 associated providers: 0 name:modesetting
`
[DisplayLinkManager.log](https://github.com/displaylink-rpm/displaylink-rpm/files/2189018/DisplayLinkManager.log)

I have attached the encrypted log file in case that is of any help.  

Thank you
Shane

mfdgroot, 2018-08-16:
+1 for me on fedora 28 using xorg 1.19.6 with a dell d6000 dock
displaylink.service is running, using Gnome on Xorg
evdi and kms_helper show up in lsmod, no error messages in dmesg
xrandr --listproviders shows only the internal provider
it's pretty frustrating there are no hints to go on

ampious, 2018-09-05:
+1 for me as @haydos89 using Fedora 28, kernel 1.4.19-200 using D6000 DisplayLink dock.

Everything compiles. builds RPM and installs
displaylink service running OK
evdi kernel module loaded
xrandr only shows one Provider

Not much else to add.....

groovyman, 2018-09-05:
Saluta

same for me, also had my F28 running with 2 versions of a D6000 and a 
Startech Adapter.
I had some issues but that link helped me to get it running.

Finally, this solution worked better with my SOHO Acer than original 
Dell Hardware with W7 and W10.

Groovie

Morlack, 2018-09-05:
Same for me, running Fedora 28 with a Dell D6000 dock. 

Like the others:
Everything compiles. builds RPM and installs
displaylink service running OK
evdi kernel module loaded
xrandr only shows one Provider

thegeologician, 2018-09-27:
+1 same story.
All mentioned above seems to run, but I see only internal display in settings.
Interestingly, xrandr show two outputs for me:
```
$ xrandr --listproviders
Providers: number : 2
Provider 0: id: 0x76 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 1 name:modesetting
Provider 1: id: 0x3f cap: 0x5, Source Output, Source Offload crtcs: 0 outputs: 0 associated providers: 1 name:modesetting
```

sdjohnson72, 2018-09-27:
I have been wondering if this is the correct place for support?  We have seen multiple reports of the same behavior but with no responses.  Would the displaylink support be a better avenue?  Or are we missing a key piece of data in order to get a response?

Thank you
Shane

ssaavedra, 2018-09-28:
Hi, one of the maintainers here.

For me it's hard to answer these questions without being able to replicate them. For some time I used a DisplayLink adapter every day, but I no longer have access to it, so I can only test packaging-related issues that don't involve actually running tests on hardware. I'm letting this open so that other users and maintainers can try to solve this.

Also, this repo is community-run and we are not associated with DisplayLink Inc., we were just interested in having the driver correctly packaged for our computers :)

sdjohnson72, 2018-10-23:
ssaavedra,
Thank you for the insight.  I appreciate the work done here in providing a RPM package for displaylink products.
Besides what has been posted here, what would you recommend in order to those of us that are having issues to get them resolved?  Would posting our issues on the forum for displaylink driver for linux page help at all?   Or are the differences between the Ubuntu package and Fedora package just too different? 

Thank you again for your help and work on this.

Shane

itzwolf, 2018-10-27:
Looks like Ubuntu updated the drivers on 17 Oct 2018. I haven't tried anything just yet because I am on Fedora 29beta. What would be required to get the new files from Ubuntu integrated and changes for upcoming F29?

mfdgroot, 2018-10-29:
I have made the changes for the update to displaylink 4.4.24, it installs fine.
Sadly, it doesn't solve anything for me and I still get no devices.
@ssaavedra do you want me to post a pull request?
Obviously it's untested...

ssaavedra, 2018-11-05:
#68 addresses some of these concerns, can you try the latest tag?
This will probably not fix the "no providers" bug in xrandr, but it may fix the X11 hangups due to glamorgl.

Also, we now build for F29 ;)

ampious, 2018-11-14:
> +1 for me as @haydos89 using Fedora 28, kernel 1.4.19-200 using D6000 DisplayLink dock.
> 
> Everything compiles. builds RPM and installs
> displaylink service running OK
> evdi kernel module loaded
> xrandr only shows one Provider
> 
> Not much else to add.....

Well things changed after someone else plugged a Windows laptop into the D6000 and checked the firmware was up to date. No firmware was updated because the Windows software reported the firmware was already up to date...

When I plugged in my Dell XPS13 9350 it pops-up the screen selector so I can extend the screen, mirror, use external or laptop, etc. This is KDE Plasma 5.13.5-1 on Fedora 28, BTW.

I can also then see providers using 'xrandr', which I couldn't see before (see output below).

However, when I go into "Displays" I can see the external monitors, which are disabled, and when I try to enable them and "Apply", my laptop screen has a blank black overlay on part of the screen and the external montitors don't show anything, or even wake up from sleep.

So it still doesn't work but it is doing more than it was before!

Diagnostic info:
]$ xrandr --listproviders
Providers: number : 3
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 2 name:modesetting
Provider 1: id: 0x262 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x23b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

]$ lsmod|grep evdi
evdi                   61440  4
drm_kms_helper        196608  2 evdi,i915
drm                   475136  19 drm_kms_helper,evdi,i915

]$ systemctl status displaylink
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: active (running) since Wed 2018-11-14 14:19:14 AEDT; 49min ago
  Process: 13723 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 13724 (DisplayLinkMana)
    Tasks: 69 (limit: 4915)
   Memory: 16.9M
   CGroup: /system.slice/displaylink.service
           └─13724 /usr/libexec/displaylink/DisplayLinkManager

Nov 14 14:19:14 ampslimlap systemd[1]: Starting DisplayLink Manager Service...
Nov 14 14:19:14 ampslimlap systemd[1]: Started DisplayLink Manager Service.

Opps, looks like I mistated my kernel version in my original message...
]$ uname -r
4.18.17-200.fc28.x86_64

Don't really understand what changed...

Technical-Angel, 2018-11-14:
I did the same thing. Hooked up a Windows laptop and ran the firmware updater. No update found. 

Hooked up my laptop and finally got the extra providers, but the monitors still weren't working. 

Checked in Displays, and the extra monitors were now showing, but off. Turned them on, still nothing. Then I remembered another comment here, and and moved the resolution of the external monitors down to 720x576. The monitor connected to the HDMI port of the dock started working. 

I moved both monitors back up to the highest resolution setting, and the HDMI monitor stayed connected. 

I disconnected the HDMI cable, and this popped up on the DP1 monitor 
[image]

Later, while I was working, the connection to the external monitor just poofed. So, back at square one. 

[jmetcalf@lt-jmetcalf ~]$ systemctl status displaylink
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: active (running) since Wed 2018-11-14 10:24:36 CST; 1h 6min ago
  Process: 1259 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 1260 (DisplayLinkMana)
    Tasks: 71 (limit: 4915)
   Memory: 181.5M
   CGroup: /system.slice/displaylink.service
           └─1260 /usr/libexec/displaylink/DisplayLinkManager

Nov 14 10:24:36 lt-jmetcalf.bundystl.com systemd[1]: Starting DisplayLink Manager Service...
Nov 14 10:24:36 lt-jmetcalf.bundystl.com systemd[1]: Started DisplayLink Manager Service.
[jmetcalf@lt-jmetcalf ~]$ xrandr --listproviders
Providers: number : 3
Provider 0: id: 0x44 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 2 associated providers: 2 name:modesetting
Provider 1: id: 0x1cc cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x196 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

ampious, 2018-11-18:
Oh well, so much for that (below).

Start using the D6000 this morning and I get nothing but black screen with a mouse pointer on my laptop display and the external monitors don't even wake up.

I tried:
* before and after a kernel update;
* generic RPM and Rawhide RPM

One step forward 3 steps back...

> @Technical-Angel your experience is sounding similar to mine.
> 
> So after I posted my last comment, I started using my DA200 USB-C dock again with a HDMI cable connected. I was still connected to the D6000 via a USB-3 (for mouse/keyboard) and when I looked at "Displays" setting there were four monitors available: the interal laptop monitor, the HDMI-connected monitor (both ot those were actually working) and the two monitors provided by the D6000.
> 
> When I tried to enable the D6000 monitors the same thing was happenning as I described above, ie. partially black screen on Laptop monitor, general glitchiness, etc.
> 
> Now this morning when I plug my laptop in fresh, without the DA200, the D6000 displays are available and after a bit of fiddling with the "Display" settings I am now using them as you would expect to use them, and I was able to arrange them around my laptop screen.
> 
> The only weird thing now is that the mouse pointer "normal" size, ie. the same scale as the screen resolution when it is over the background desktop, but when I move it over a window it is much larger.
> 
> Looks funny but actually works fine. I'm looking up mouse scaling issues now to see if there is something I can do about it.

Technical-Angel, 2018-11-19:
[image]
I swapped out the HDMI cable, so both monitors were connected with a DisplayPort to HDMI cable. I'll report back if the displays stop working like they did the other day, but so far, VICTORY.

ssaavedra, 2019-02-28:
This thread has been out of activity for a time, and I'm considering it closed.
Watchers from the future, consider also #70 if you think this is related to you.

closed 2019-02-28 by ssaavedra