displaylink-rpm/displaylink-rpm #65: Xorg crash on Rawhide with v1.5.0.2

mdhill, 2018-07-27:
I have a ThinkPad USB 3.0 Pro dock. When I connect a Dell U3011 monitor (DVI) I can configure it in the GNOME Displays panel but when I switch it to ON in the panel, I get an Xorg crash.

I get a similar crash when I power on the dock with a Samsung S27F350 connected (VGA->DVI or HDMI->DisplayPort), or if I plug the Samsung in to an already-powered dock.

I'm not sure what's good diagnostic information, but journalctl shows these:
Jul 27 14:24:57 himon systemd-coredump[2940]: Process 30576 (Xorg) of user 1000 dumped core.
Jul 27 14:24:59 himon audit[3007]: ANOM_ABEND auid=4294967295 uid=42 gid=42 ses=4294967295 subj=system_u:system_r:xserver_t:s0-s0:c0.c1023 pid=3007 comm="Xorg" exe="/usr/libexec/Xorg" sig=11 res=1
...and...
Jul 27 14:25:03 himon /usr/libexec/gdm-x-session[3125]: gbm: Last dlopen error: /usr/lib64/dri/evdi_dri.so: cannot open shared object file: No such file or directory

Geodomus, 2018-10-23:
I'm getting the same crash on rawhide/F29, which sends me back to the login screen.

I'm unsure what to look for in journalctl though.

ssaavedra, 2018-11-05:
Can you check whether this is related to https://github.com/DisplayLink/evdi/issues/133 ? PR #68 may have solved this if so. Could you test tag [4.4.24](https://github.com/displaylink-rpm/displaylink-rpm/tree/4.4.24)?

mdhill, 2018-11-06:
No more crash for me with tag 4.4.24.

ssaavedra, 2018-11-06:
@Geodomus if you are still suffering from this, feel free to reopen ;)
Will close for now since the original problem is now solved.

closed 2018-11-06 by ssaavedra