displaylink-rpm/displaylink-rpm #66: Cannot Insert

clham, 2018-08-11:
Good Morning, I tried to install from the Travis build on to Fedora 28_64.
This feels related to #63 , but I'm not smart enough to know how.
Result of  ```journalctl -f -n 0``` is:

```Aug 11 12:27:55 magritte audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=displaylink comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Aug 11 12:27:55 magritte systemd[1]: dlm.service: Service hold-off time over, scheduling restart.
Aug 11 12:27:55 magritte systemd[1]: dlm.service: Scheduled restart job, restart counter is at 310.
Aug 11 12:27:55 magritte systemd[1]: Stopped DisplayLink Manager Service.
Aug 11 12:27:55 magritte audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=dlm comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Aug 11 12:27:55 magritte audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=dlm comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Aug 11 12:27:55 magritte systemd[1]: Starting DisplayLink Manager Service...
Aug 11 12:27:55 magritte sh[25641]: modprobe: ERROR: could not insert 'evdi': Operation not permitted
Aug 11 12:27:55 magritte sh[25641]: Module evdi/4.2.29 already installed on kernel 4.17.12-200.fc28.x86_64/x86_64
Aug 11 12:27:55 magritte sh[25641]: modprobe: ERROR: could not insert 'evdi': Operation not permitted
Aug 11 12:27:55 magritte systemd[1]: dlm.service: Control process exited, code=exited status=1
Aug 11 12:27:55 magritte systemd[1]: dlm.service: Failed with result 'exit-code'.
Aug 11 12:27:55 magritte systemd[1]: Failed to start DisplayLink Manager Service.
```

output of ```cat /var/log/displaylink/displaylink.log``` is 
```
Creating symlink /var/lib/dkms/evdi/1.5.0.2/source ->
                 /usr/src/evdi-1.5.0.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....(bad exit status: 2)
make -j4 KERNELRELEASE=4.17.12-200.fc28.x86_64 all INCLUDEDIR=/lib/modules/4.17.12-200.fc28.x86_64/build/include KVERSION=4.17.12-200.fc28.x86_64 DKMS_BUILD=1.....
cleaning build area....(bad exit status: 2)

DKMS: build completed.

evdi.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.17.12-200.fc28.x86_64/extra/
Adding any weak-modules

depmod.....

DKMS: install completed.

evdi.ko.xz:
Running module version sanity check.
 - Original module
   - This kernel never originally had a module by this name
 - Installation
   - Installing to /lib/modules/4.17.12-200.fc28.x86_64/extra/
Adding any weak-modules

depmod.....

DKMS: install completed.
```
Any thoughts or more info I can provide?

alecmillerbackstop, 2018-08-24:
i get the same error unplugging the USB displaylink and reconnecting it.   Upgraded ubuntu 16.04 lts to 18.04 lts on a Dell latitude e7470.  Was working fine in ubuntu 16.04

@:~$ journalctl -f -n 0
-- Logs begin at Wed 2018-08-01 22:59:42 CDT. --
Aug 24 09:42:16  kernel: xhci_hcd 0000:00:14.0: Cannot set link state.
Aug 24 09:42:16  kernel: usb usb2-port1: cannot disable (err = -32)
Aug 24 09:42:16  kernel: usb 2-1: USB disconnect, device number 5
Aug 24 09:42:16  kernel: evdi: [D] evdi_painter_disconnect:612 (dev=1) Disconnected from 000000008cf5287b
Aug 24 09:42:16  kernel: evdi: [D] evdi_detect:78 Painter is disconnected
Aug 24 09:42:16  upowerd[2082]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0
Aug 24 09:42:16  upowerd[2082]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1
Aug 24 09:42:17  org.gnome.Shell.desktop[5588]: Window manager warning: Overwriting existing binding of keysym 73 with keysym 73 (keycode 27).
Aug 24 09:42:21  colord-sane[24181]: io/hpmud/pp.c 627: unable to read device-id ret=-1
Aug 24 09:42:30  kernel: usb 2-1: new SuperSpeed USB device number 6 using xhci_hcd
Aug 24 09:42:30  kernel: usb 2-1: New USB device found, idVendor=17e9, idProduct=4300
Aug 24 09:42:30  kernel: usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 24 09:42:30  kernel: usb 2-1: Product: USB32DVIPRO
Aug 24 09:42:30  kernel: usb 2-1: Manufacturer: DisplayLink
Aug 24 09:42:30  kernel: usb 2-1: SerialNumber: 000100110001228
Aug 24 09:42:30  mtp-probe[24203]: checking bus 2, device 6: "/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1"
Aug 24 09:42:30  mtp-probe[24203]: bus: 2, device: 6 was not an MTP device
Aug 24 09:42:31  kernel: evdi: [D] evdi_painter_connect:573 (dev=1) Connected with 000000008c7be1c8
Aug 24 09:42:31  kernel: evdi: [D] evdi_detect:75 (dev=1) Painter is connected
Aug 24 09:42:31  kernel: evdi: [D] evdi_painter_crtc_state_notify:482 (dev=1) Notifying crtc state: 3
Aug 24 09:42:31  fwupd[8129]: failed to parse interface data: interface found, but not the correct length for functional data: 25 bytes: 09 21 01 c8 00 00 40 01 01 10 40 09 05 12 01 01 03 45 6c 6c 61 44 6f 63 6b
Aug 24 09:42:31  upowerd[2082]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1
Aug 24 09:42:31  upowerd[2082]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0
Aug 24 09:42:32  org.gnome.Shell.desktop[5588]: Window manager warning: Overwriting existing binding of keysym 73 with keysym 73 (keycode 27).
Aug 24 09:42:36  colord-sane[24253]: io/hpmud/pp.c 627: unable to read device-id ret=-1

NathanDeckinga, 2018-08-28:
I don't have a lot of additional information, but I wanted to confirm that my experience mirrored that of the original poster (clham).  I started off on Fedora 28 with KDE (the spin, not added later) and X11 (Wayland isn't quite ready to go with KDE so that was never in the picture) on a recent Dell XPS 13 9370.  

Disabling Secure Boot in BIOS/UEFI allowed this to work.

ssaavedra, 2018-08-30:
Usually the error on `modprobe evdi` with Secure Boot enabled should be `modprobe: ERROR: could not insert 'evdi': Required key not available`, and not `Operation not permitted`. That happens most likely when the user does not have permission to use `modprobe`.

If you still want to benefit from Secure Boot, you will have to follow steps to enroll a MOK (Machine Owner's Key) into your chain of trust. You can follow my comment here https://github.com/displaylink-rpm/displaylink-rpm/issues/48#issuecomment-358948794. For a more high-level explanation I have an answer on askubuntu that is totally transposable to Fedora: https://askubuntu.com/a/797442/12049

alecmillerbackstop, 2018-09-23:
just following up on my post from above.  I disabled wayland from the custom.conf file in /etc/gdm3/

WaylandEnable=false

and now it works.

ssaavedra, 2018-09-28:
I'll mark this as closed since everyone here seems to have worked this out.

closed 2018-09-28 by ssaavedra

MaurizioBella, 2020-06-13:
I had the same issue, my third monitoring didn't work. I was able to make it works thanks to the previously mentioned link. I'd like to give a shout-out to everyone who helped here

> $ lsusb
> Bus 001 Device 005: ID 17e9:4324 DisplayLink Plugable UGA-4KDP
> $ uname -a
> Linux mauri-ubuntu 5.4.0-37-generic #41-Ubuntu SMP Wed Jun 3 18:57:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Here the [link](https://askubuntu.com/questions/760671/could-not-load-vboxdrv-after-upgrade-to-ubuntu-16-04-and-i-want-to-keep-secur/768310#768310)

`sudo /usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 ./MOK.priv ./MOK.der $(modinfo -n evdi)`