displaylink-rpm/displaylink-rpm #67: Fedora 28 - RPM fails install

Office-Manager, 2018-09-18:
Hi all ,
I have a D1000 Dell adaptor but struggling to get it working running Fedora 28

```
# lsb_release -a
LSB Version:	:core-4.1-amd64:core-4.1-noarch
Distributor ID:	Fedora
Description:	Fedora release 28 (Twenty Eight)
Release:	28
Codename:	TwentyEight

# uname -a
Linux localhost.localdomain 4.16.3-301.fc28.x86_64 #1 SMP Mon Apr 23 21:59:58 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

```

I've tried using the latest RPM  `fedora-28-displaylink-1.5.0.2-2.x86_64.rpm`
But it's not installing for me. 

```
# sudo rpm -ivh fedora-28-displaylink-1.5.0.2-2.x86_64.rpm 
Verifying...                          ################################# [100%]
Preparing...                          ################################# [100%]
Updating / installing...
   1:displaylink-1.5.0.2-2            ################################# [100%]
Job for displaylink.service failed because the control process exited with error code.
See "systemctl status displaylink.service" and "journalctl -xe" for details.
warning: %post(displaylink-1.5.0.2-2.x86_64) scriptlet failed, exit status 1

# /sbin/modprobe evdi 
modprobe: FATAL: Module evdi not found in directory /lib/modules/4.16.3-301.fc28.x86_64
```

Is there any other files which I could give that may help debug ?
Any help would be appreciated

( As a side note I thought it might have something to do with me not having the nvdia drivers installed so I followed the instructions at https://www.if-not-true-then-false.com/2015/fedora-nvidia-guide/  -- but it seemingly failed since whenever I boot into that kernel i just get a black screen.... So hopefully I don't need to go into that kernel )

beardedbern, 2018-09-20:
Same issues here. Any help or guidance would be greatly appreciated.

rhmk, 2018-09-26:
To better resolve depencies I use dnf to install these packages:

`# dnf install https://github.com/displaylink-rpm/displaylink-rpm/releases/download/v4.2.29-2/fedora-28-displaylink-1.5.0.2-2.x86_64.rpm
`
This worked perfektly well and added the following depencies:
`===============================================================================================================================================
 Paket                                    Arch                      Version                               Paketquelle                     Größe
================================================================================================================================================
Installieren:
 displaylink                              x86_64                    1.5.0.2-2                             @commandline                    3.3 M
Installiere Abhängigkeiten:
 dkms                                     noarch                    2.6.1-1.fc28                          updates                          78 k
 elfutils-libelf-devel                    x86_64                    0.173-1.fc28                          updates                          50 k
 kernel-devel                             x86_64                    4.18.9-200.fc28                       updates                          13 M

Transaktionsübersicht
================================================================================================================================================
Installieren  4 Pakete
`
Rgds Markus

seanmiz, 2018-09-26:
I have the same results, even when trying the dnf method.

"FATAL: Module evdi not found in directory /lib/modules/4.17.14-202.fc28.x86_64"

ssaavedra, 2019-02-28:
You should have first uninstalled the rpm before reinstalling via dnf. If this is still happening, please open a new issue. I'm closing this, since it had no activity for months.

closed 2019-02-28 by ssaavedra