displaylink-rpm/displaylink-rpm #70: On Fedora 29, module loads but I get no providers

kodos96, 2018-11-07:
On a Fedora 29 system (a Dell Latitude 7490 with a Dell D6000 dock), the rpm installs, and the service starts successfully:

```
[root@localhost Downloads]# systemctl status displaylink.service 
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: active (running) since Tue 2018-11-06 16:12:44 PST; 4s ago
  Process: 8428 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 8429 (DisplayLinkMana)
    Tasks: 65 (limit: 4915)
   Memory: 16.1M
   CGroup: /system.slice/displaylink.service
           └─8429 /usr/libexec/displaylink/DisplayLinkManager

Nov 06 16:12:44 localhost.localdomain systemd[1]: Starting DisplayLink Manager Service...
Nov 06 16:12:44 localhost.localdomain systemd[1]: Started DisplayLink Manager Service.
```
But I get no providers... like literally 0, not even a builtin: 
```

xrandr --listproviders
Providers: number : 0
```
What am I doing wrong?

Technical-Angel, 2018-11-07:
Hello, 

I'm having a similar issue. I have an Inspiron 7348 2-in-1 running Fedora 29 and a d6000 dock. 

Everything installed correctly, but I can't get the external monitor to work via the dock. 

Unlike kodos96, I do get 1 provider using xrandr. 
[image]

I can make a separate thread, if necessary, but it seems to be related.

sorvani, 2018-11-07:
I setup @Technical-Angel 's system and used both the current release as well as built the rpm myself from https://github.com/displaylink-rpm/displaylink-rpm/pull/69  from @okayawright

kodos96, 2018-11-07:
So I realized I was still on xorg 1.20.1, I upped to 1.20.3 and built from  @okayawright 's fork with the workaround removed. Now, plugging and unplugging the dock results in the laptop screen flickering briefly, which it didn't before. Otherwise no change. Xrandr still shows no providers (with or without the dock plugged, or the displaylink service running, which seems very odd to me), and no additional screens available.

jpgimenez, 2018-11-08:
same here,
```
$ rpm -q xorg-x11-server-Xorg xorg-x11-server-Xwayland displaylink 
xorg-x11-server-Xorg-1.20.3-1.fc29.x86_64
xorg-x11-server-Xwayland-1.20.3-1.fc29.x86_64
displaylink-1.5.1-2.x86_64
```
```
$ systemctl status displaylink.service 
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: active (running) since Thu 2018-11-08 08:58:23 -03; 3min 19s ago
  Process: 16937 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 16938 (DisplayLinkMana)
    Tasks: 11 (limit: 4915)
   Memory: 4.7M
   CGroup: /system.slice/displaylink.service
           └─16938 /usr/libexec/displaylink/DisplayLinkManager

nov 08 08:58:23 linux.local systemd[1]: Starting DisplayLink Manager Service...
nov 08 08:58:23 linux.local systemd[1]: Started DisplayLink Manager Service.
```
```
$ xrandr --listproviders 
Providers: number : 0
```
I can use the external monitor, but only in 4:3 low resolutions. It was working with no issues on fedora 28.

sorvani, 2018-11-11:
Not working for me either. I used 1.5.1 and 1.5.1-2 (from pending pull request)
 
Results on my laptop.
```
[jbusch@lt-jared ~]$ sudo xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x44 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
[jbusch@lt-jared ~]$ sudo rpm -q xorg-x11-server-Xorg xorg-x11-server-Xwayland displaylink
xorg-x11-server-Xorg-1.20.3-1.fc29.x86_64
xorg-x11-server-Xwayland-1.20.3-1.fc29.x86_64
displaylink-1.5.1-2.x86_64
[jbusch@lt-jared ~]$ 
```

Technical-Angel, 2018-11-19:
[image]
I swapped out the HDMI cable, so both monitors were connected with a DisplayPort to HDMI cable. I'll report back if the displays stop working like they did the other day, but so far, VICTORY.

kodos96, 2018-11-20:
So it now seems to be working for me, using an RPM built from @okayawright 's branch - it takes some time (like a minute) and lots of screen flickering on my main screen before it recognizes the newly plugged in screens, and it seems to be finicky about the order you plug in/turn on monitors, but they do eventually become available. Still no output from "xrandr --listproviders" though, which seems odd.

jpgimenez, 2018-11-20:
is working after last update, kernel 4.19.2-300.fc29.x86_64
still `xrandr --listproviders` returns 0 
```
xrandr --listproviders 
Providers: number : 0
```

ZMcCrocklin, 2018-11-30:
Hello Friends,
     I am having this same issue. on an Dell Latitude 5580 with the doc, but I am also trying directly connected to the laptop. I'm on latest kernel: 4.19.4-300.fc29. I get a different error when trying to run xrandr:

`# xrandr --listproviders
No protocol specified
Can't open display :0`

I'm still not getting anything to the monitor.  I have 2 other monitors connected via HDMI & mini display port, and they're displaying just fine.

jch388, 2018-12-05:
I am also experiencing this on a Lenovo Thinkpad Yoga X380 running Fedora 29 4.19.5-300.fc29.x86_64. No output to external monitor (via display port).

$ xrandr --listproviders
Providers: number : 0

$ systemctl status displaylink.service 
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor >
   Active: active (running) since Tue 2018-12-04 19:16:16 CST; 54s ago
  Process: 1781 ExecStartPre=/sbin/modprobe evdi (code=exited, status=0/SUCCESS)
 Main PID: 1788 (DisplayLinkMana)
    Tasks: 33 (limit: 4915)
   Memory: 22.9M
   CGroup: /system.slice/displaylink.service
           └─1788 /usr/libexec/displaylink/DisplayLinkManager

Dec 04 19:16:16 thinklinux systemd[1]: Starting DisplayLink Manager Service...
Dec 04 19:16:16 thinklinux systemd[1]: Started DisplayLink Manager Service.

gnarlium, 2019-01-11:
Anyone still having this issue? I am having it as well

Technical-Angel, 2019-01-11:
While I have one provider, I have been unable to get the external monitors to connect for weeks. I just haven't had time to troubleshoot.

Technical-Angel, 2019-01-11:
[jmetcalf@lt-jmetcalf ~]$ xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x44 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting

gnarlium, 2019-01-11:
[jparlett@bcubs aws]$ xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

outputs=5 matches up with my `xrandr --current` output, which shows my 3 disconnected (and 2 connected) display outputs.

gnarlium, 2019-01-14:
To clarify, I can see the dock's displays, but can't enable them:
```
[jparlett@bcubs Downloads]$ xrandr --current|grep connected
eDP-1 connected (normal left inverted right x axis y axis)
DP-1 disconnected (normal left inverted right x axis y axis)
HDMI-1 connected primary 1920x1200+0+0 (normal left inverted right x axis y axis) 550mm x 340mm
DP-2 disconnected (normal left inverted right x axis y axis)
HDMI-2 disconnected (normal left inverted right x axis y axis)
```
^^^(Laptop screen off, connected to external monitor via HDMI on laptop)

But `xrandr --listmonitors` doesn't show the displays as available:
```[jparlett@bcubs Downloads]$ xrandr --listmonitors
Monitors: 1
 0: +*HDMI-1 1920/550x1200/340+0+0  HDMI-1```

ZMcCrocklin, 2019-01-14:
So it sees the laptop screen & the single HDMI monitor connected, but nothing else.  I'm going to have to go into the office to see what output I get, but I installed both the x86_64 & the src rpm files, then restarted & that got mine to work.  However, I will state that I'm using a DVI -> USB adapter, not a dock.

```[root@Revan:~]# xrandr --current | grep connected
eDP-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 344mm x 194mm
HDMI-1 disconnected (normal left inverted right x axis y axis)
DP-1 disconnected (normal left inverted right x axis y axis)
HDMI-2 disconnected (normal left inverted right x axis y axis)
DP-2 disconnected (normal left inverted right x axis y axis)
HDMI-3 disconnected (normal left inverted right x axis y axis)```

This is not representative of the monitors I normally have connected.  I have issues with my laptop pushing out to the HDMI & VGA ports when I'm working from home, and an issue with refresh rates on one of my monitors at work, so I have to push the display profiles out on startup with a shell script (DVI-I-1-1 is the DisplayLink adapter):

```[root@Revan:~]# cat /custom/setmonitors.sh
#!/bin/sh
xrandr | grep "DP-2" | grep " connected " && xrandr | grep "HDMI-1" | grep " connected "
if [ $? -eq 0 ]; then
    xrandr --output DP-2 --mode 1680x1050 --right-of eDP-1
    xrandr --output HDMI-1 --mode 1920x1200 --rate 60 --left-of eDP-1
else
    xrandr | grep "DP-1-1" | grep " connected " && xrandr | grep "DP-1-2" | grep " connected " && xrandr | grep "DVI-I-1-1" | grep " connected "
    if [ $? -eq 0 ]; then
	xrandr --output DP-1-1 --mode 1680x1050 --pos 1050x0
	xrandr --output DP-1-2 --mode 1920x1080 --pos 2970x193 --rotate left
	xrandr --output DVI-I-1-1 --mode 1680x1050 --pos 0x450 --rotate left --rate 60
    fi
fi```

ZMcCrocklin, 2019-01-15:
I ended up working from home today, so I'm going to show my output on my home setup.  Tomorrow I'll show what my work output looks like.

```
[root@Revan:~]# xrandr --current | grep " connected "
eDP-1 connected primary 1920x1080+1920+0 (normal left inverted right x axis y axis) 344mm x 194mm
HDMI-1 connected 1920x1200+0+0 (normal left inverted right x axis y axis) 519mm x 324mm
DP-2 connected 1680x1050+3840+0 (normal left inverted right x axis y axis) 473mm x 296mm

[root@Revan:~]# xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x48 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 6 associated providers: 0 name:modesetting

[root@Revan:~]# xrandr --listmonitors
Monitors: 3
 0: +*eDP-1 1920/344x1080/194+1920+0  eDP-1
 1: +DP-2 1680/473x1050/296+3840+0  DP-2
 2: +HDMI-1 1920/519x1200/324+0+0  HDMI-1
```

ssaavedra, 2019-02-28:
Has this issue been resolved with the latest version? Either way, this seems related to displaylinkmanager/evdi and not the packaging itself anyway. I left it open in case it could help others, but the proper place for this kind of support is probably the DisplayLink forums.
Feel free to open a new issue with updated information.

closed 2019-02-28 by ssaavedra