displaylink-rpm/displaylink-rpm #72: kernel 4.19

summerm78, 2018-11-20:
After the update to kernel 4.19 displaylink stop working. When trying to load the module I get this error

`/sbin/modprobe evdi
modprobe: FATAL: Module evdi not found in directory /lib/modules/4.19.2-300.fc29.x86_64`
Modul is available up to kernel 4.18.7-200

Maybe it is the same issue as descriped in https://github.com/DisplayLink/evdi/issues/139

RandieM, 2018-11-26:
I had the same problem. I upgraded to version 1.5.1, restarted the computer and the problem was solved. Hope it helps!

ssaavedra, 2018-12-16:
"Module evdi not found in directory /lib/modules/$(uname -r)" is probably an issue of dkms not firing and building the module for your kernel. You can try to build it manually (`dkms build evdi/VERSION` for the version you have currently installed, you will find a folder at `/usr/src/evdi-VERSION`), and if that fails, check the dkms logs.

summerm78, 2018-12-19:
Building and installing the modul manually works.

ssaavedra, 2019-01-29:
Closing due to lack of further feedback

closed 2019-01-29 by ssaavedra