displaylink-rpm/displaylink-rpm #75: fedora 29, kernel 4.19.13 - screen stays black

pgstef, 2019-01-16:
Hi,

I just jumped from f28 to f29 (4.19.13-300.fc29.x86_64) and now, my screen stays black. (Was working fine with f28)

I both installed "rpm -ivh fedora-29-displaylink-1.5.1-1.src.rpm fedora-29-displaylink-1.5.1-1.x86_64.rpm".

```
$ systemctl status displaylink.service
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: active (running) since Wed 2019-01-16 11:12:09 CET; 25s ago
 Main PID: 19679 (DisplayLinkMana)
    Tasks: 33 (limit: 4915)
   Memory: 12.7M
   CGroup: /system.slice/displaylink.service
           └─19679 /usr/libexec/displaylink/DisplayLinkManager

Jan 16 11:12:09 stefan.dalibo.local systemd[1]: Starting DisplayLink Manager Service...
Jan 16 11:12:09 stefan.dalibo.local systemd[1]: Started DisplayLink Manager Service.
```

From journalctl :

```
kernel: evdi: [D] evdi_painter_connect:574 (dev=1) Connected with 00000000aa539aae
kernel: evdi: [D] evdi_detect:85 (dev=1) Painter is connected
kernel: evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 3
kernel: evdi: [D] evdi_detect:85 (dev=1) Painter is connected
kernel: evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
kernel: evdi: [D] evdi_detect:85 (dev=1) Painter is connected
kernel: evdi: [D] evdi_painter_get_edid_copy:217 (dev=1) 00 ff ff
kernel: evdi: [D] evdi_painter_mode_changed_notify:507 (dev=1) Notifying mode changed: 1920x1080@60; bpp 32; 
kernel: evdi: [D] evdi_painter_mode_changed_notify:508 pixel format 875713112
kernel: evdi: [D] evdi_painter_crtc_state_notify:483 (dev=1) Notifying crtc state: 0
kernel: evdi: [D] evdi_painter_dpms_notify:470 (dev=1) Notifying dpms mode: 0
```

What can I check now?

Many thanks in advance,
Kind regards

linuxcrash, 2019-01-20:
I had the same issue with my Dell XPS 15 and D-3000 dock. 
What normally helps if the latest kernel updates break the display functionality is to do the following:
- Remove current displaylink rpm (sudo dnf remove displaylink)
- Clean up the local GIT repo and remove any compiled RPM in the x86_64 directory, remove the Displaylink USB Graphics Software for Ubuntu.zip, delete the vX.x.x.tar.gz file.
- Clean up the /var/tmp folder and remove the all EVDI and displaylink subfolders.
- Go back to your local GIT repo and run git pull to ensure you got the latest sources.
- Run sudo make rawhide command to let the make script download the latest EVDI stuff and recompile a new RPM file.
- Run sudo rpm -i displaylink-x.x.x-x.rawhide.xxxxx.rpm.
- Reboot the system.

This has fixed it for the latest kernel version (4.9.15-300) I'm running on. If it breaks again, repeat above steps to recompile the RPM with the latest EVDI release.

Hope this helps.

pgstef, 2019-01-21:
Hi,

Thanks for your answer. I just tried that.

Unfortunately, no change :-(

ssaavedra, 2019-02-28:
Have you tried the latest released version? Is this issue obsolete?

pgstef, 2019-03-01:
Hi,
Just updated now.
With `displaylink-1.6.0-1.x86_64` and kernel `4.20.12-200.fc29.x86_64`, still same behavior. The laptop doesn't see the screens as "connected".
Sorry :-(

sthuebner, 2019-03-01:
I have the exact same issue, running `displaylink-1.6.0-1.x86_64` and `kernel 4.20.12-200.fc29.x86_64` here as well. Hardware: Dell Precision 7510 laptop, Dell Universal Dock D6000, two Dell screens mounted to the Dock.

ptschack, 2019-03-04:
Same issue here. I'm using a freshly installed Fedora 29 on a ThinkPad T480s with a "hybrid USB-C with USB-A dock". Kernel Version is `4.20.13-200.fc29.x86_64`, I installed `fedora-29-displaylink-1.6.0-1.x86_64.rpm`.
When I connect a screen using the dock, the ThinkPad screen briefly blacks out, and it looks like the external monitor is looking for an input source. But it fails after a few seconds, the external monitor says "no input device found" and the fedora display settings app does not list any other displays than the internal one.

The displaylink service is loaded and active (checked via `systemctl status displaylink`).

The external monitor is connected to the dock using the DisplayPort connector.

ssaavedra, 2019-03-06:
If displaylink.service is running, lsusb detects your dock and evdi is loaded as a kernel module, then this issue is most likely not related to a packaging issue.
For those kind of issues, I recommend you go to the DisplayLink forums and state it there, since the actual developers of the DisplayLink technology may be of assistance :)

Since it's unlikely to be packaging-related I'll close this issue as there is nothing we can do from here.

closed 2019-03-06 by ssaavedra