displaylink-rpm/displaylink-rpm #76: Please join RPM Fusion project

kwizart, 2019-01-22:
Hi there,

I've seen that you provide packages for fedora and EL for displaylink adapters.
Please consider joining the rpmfusion.org project to provide theses packages there.

There will be a need to adapt the package to handle the kmod mechanism which will allow to provide pre-built kernel modules.

ssaavedra, 2019-01-29:
It's an interesting proposal. We are not currently providing EL releases, though; CentOS 7 is built but I'm not sure if it works.

As it currently is, this project does not totally comply with the Fedora Packaging Guidelines. For example, three packages should be generated: one for the evdi dkms sources, another one for the evdi library, and a separate one for the displaylink binaries themselves.

While this would make it easier in terms of installing each thing separately (and in terms of which license applies to which binaries), it puts a burden on version maintenance. The current "single package" generation is more appropriate for downloading the RPM and installing it, but is a stopper for introducing it into a proper DNF repository.

We cannot upload this on a COPR repository, though, since parts of the redistributed files are under a proprietary non-opensource license (the DisplayLink binaries).

Another issue would be switching from dkms to kmod, which is also an interesting task by itself.

If this is something that has enough popular attention, I can start a "proper-dnf" or "rpmfusion" branch in order to address these changes while we still support the current course of action until this is finished.

:+1: the issue if you are interested in that :)

kwizart, 2019-01-29:
Thx for the answer.

Having a separate branch seems appropriate.

ssaavedra, 2019-02-04:
I have just created a fedora-guidelines branch to adapt the specfile so that it fits the packaging guidelines, including separate packages for {a,}kmod-evdi and displaylink-lib.

ffgiff, 2019-06-25:
@kwizart I've put in a PR (#97) for building as an akmod. Would you mind giving some feedback on whether this would be right for RPMFusion? In particular, I've only done two packages, one for akmod and one for all the userland stuff, but this thread suggests it would be preferable to do three, akmod, libevdi and displaylink binaries.

kwizart, 2019-06-25:
@ffgiff 
This looks like a good start, thx for looking into this.
I would suggest to have evdi-kmod.spec and displaylink.spec (the later having a "virtual" Provides: evdi-kmod-common to satisfy the suer

Also please have a look on https://rpmfusion.org/Contributors
You will have to provide both a spec file and src.rpm to review (one for each package in a separate bugzilla entry, so we will use our own tooling to process the review).

Thx

firefoxlover, 2023-09-27:
I see its still not part of RPMFusion which is kinda sad. Awesome that this project is still going! Thanks a lot!

firefoxlover, 2023-09-27:
In the meantime, this COPR should work

https://copr.fedorainfracloud.org/coprs/crashdummy/Displaylink/

rapgro, 2023-10-22:
> I see its still not part of RPMFusion which is kinda sad. Awesome that this project is still going! Thanks a lot!

As RPMFusion is an official repository there's some process to review package to get imported. Maybe I can offer some help with https://rpmfusion.org/Contributors .

nathan-weinberg, 2024-04-12:
I am not a Fedora contributor/maintainer and I see that the https://github.com/displaylink-rpm/displaylink-rpm/tree/fedora-guidelines branch is basically abandoned, but if there is any interest from the project maintainers in trying to get this into Fedora project I'm happy to try and facilitate a connection to start that process

regulad, 2025-05-15:
Using kmod for RPMFusion support would also enable the use of the project on Fedora coreos-based systems, like Kinoite: https://github.com/coreos/rpm-ostree/issues/1091

DKMS not working under rpm-ostree currently prevents the use of this package.

regulad, 2025-05-18:
I have updated the akmod to work on Fedora 42. See progress here: #166