displaylink-rpm/displaylink-rpm #79: evdi issue?

beshad, 2019-01-26:
hi all, 

i have successfully insalled Displaylink on Ubuntu and Debian but doesn't matter what I do i can't get the service to start in Fedora: 
```
systemctl status displaylink.service
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2019-01-26 13:04:24 NZDT; 626ms ago
  Process: 5681 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)
```
and 

```
- Unit displaylink.service has begun starting up.
Jan 26 12:59:47 fedora-xps18 modprobe[5096]: modprobe: FATAL: Module evdi not found in directory /lib/modules/4.19.4-300.fc29.x86_64
Jan 26 12:59:47 fedora-xps18 systemd[1]: displaylink.service: Control process exited, code=exited status=1
Jan 26 12:59:47 fedora-xps18 systemd[1]: displaylink.service: Failed with result 'exit-code'.
Jan 26 12:59:47 fedora-xps18 systemd[1]: Failed to start DisplayLink Manager Service.
-- Subject: Unit displaylink.service has failed
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit displaylink.service has failed.
-- 
-- The result is failed.
Jan 26 12:59:47 fedora-xps18 audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=displaylink comm="systemd" exe="/usr/lib/syst>
```

this is my kernel: 
```
uname -r
4.19.4-300.fc29.x86_64
```
i tried cloning the repo and running make, as well as going into ci folder and run the script, and finally downloading the .rpm (25, 27 and 29) from releases. neither works. what am i missing?

sorvani, 2019-01-28:
I did this and it resolved it for kernel 4.20
 https://github.com/DisplayLink/evdi/issues/145

beshad, 2019-01-29:
i tried and still get this error"
```
$ dkms build evdi/1.5.1
Error! echo
Your kernel headers for kernel 4.19.4-300.fc29.x86_64 cannot be found at
/lib/modules/4.19.4-300.fc29.x86_64/build or /lib/modules/4.19.4-300.fc29.x86_64/source.
```
couldn't this be because i have dual boot Ubuntu/Fedora, and grud is on the Ubuntu side? so my 4.20 kernel is not loading and different to hedares?
i also have these:
```
$ systemctl --type=service
 displaylink.service                                                          loaded activating auto-restart DisplayLink Manager Service
● dkms.service                                                                 loaded failed     failed       Builds and install new kernel modules through DKMS
```
and
```
$ dnf info kernel-headers
Remi's RPM repository - PHP 7.1 - Fedora 29 - x86_64                                                    0.0  B/s |   0  B     00:32    
Failed to synchronize cache for repo 'remi-php71', ignoring this repo.
Last metadata expiration check: 22:49:46 ago on Mon 28 Jan 2019 19:58:15 NZDT.
Installed Packages
Name         : kernel-headers
Version      : 4.20.3
Release      : 200.fc29
Arch         : x86_64
Size         : 4.6 M
Source       : kernel-headers-4.20.3-200.fc29.src.rpm
Repo         : @System
```

ssaavedra, 2019-01-29:
You don't have `kernel-headers` installed for the current version of your `kernel` package. You can either install `kernel-headers-$(uname -r)`, or upgrade your `kernel` package to the same version as your kernel headers.

You can have multiple kernels (and kernel-headers) installed at once, but in order to build for your kernel, you must have the appropriate version of the headers.

Also, if you upgrade to Linux >= 4.20 you will run into #78 which as of this moment must be solved manually on your side unless someone can contribute the patch to this repo.

beshad, 2019-01-29:
thanks helpful info, i know it is unrelated to this repo issues, but how do you upgrade/downgrade/match the kernel-header version in Fedora with header?been googling and tried many commands but cant change this `dnf info kernel-headers` i managed  to start the dkms service but still unable to build evdi and i am blaming everything to this mismatch kernel and kernel-header (how did i end up in this!!!)

beshad, 2019-01-30:
i have now managed to match my kernel and kernel-headers (4.18.20) but still get this error:

```
sudo dkms install evdi/1.5.1
Error! echo
Your kernel headers for kernel 4.18.20-100.fc27.x86_64 cannot be found at
/lib/modules/4.18.20-100.fc27.x86_64/build or /lib/modules/4.18.20-100.fc27.x86_64/source.
 
```

same for `dkms build evdi/1.5.1`

markllama, 2019-01-30:
I just upgraded kernel and kernel headers to  _kernel-headers-4.20.4-200.fc29.x86_64_ and _kernel-4.20.4-200.fc29.x86_64_ .  

I've cleaned and rebuilt and I'm seeing the same issue

Looking in  `/var/lib/dkms/evdi/1.5.1/`

I don't see anything later than 4.,19

````
 sudo ls /var/lib/dkms/evdi
1.5.0.2				       kernel-4.18.18-300.fc29.x86_64-x86_64
1.5.1				       kernel-4.19.10-300.fc29.x86_64-x86_64
kernel-4.18.16-300.fc29.x86_64-x86_64  kernel-4.19.13-300.fc29.x86_64-x86_64
kernel-4.18.17-300.fc29.x86_64-x86_64
````

Unfortunately this has continued for the last 4 kernel updates so I finally, today can't backrev the kernel on boot and get displaylink to work.

markllama, 2019-01-30:
Yep This was https://github.com/displaylink-rpm/displaylink-rpm/issues/78 I followed it, updated the dkms_drv.h and rebuild evdi and it works for me now on 4.20.

Thanks!

beshad, 2019-01-30:
@markllama looking into `/var/lib/dkms/evdi` if there is no 4.18.20 in the list, does that mean i need to downgrade to another listed version?

beshad, 2019-01-31:
upgraded to 4.20 now. followed the instructions from #78 still i can't build dkms: 
```
$ dkms build evdi/1.5.1
Error! echo
Your kernel headers for kernel 4.20.5-250.vanilla.knurd.1.fc29.x86_64 cannot be found at
/lib/modules/4.20.5-250.vanilla.knurd.1.fc29.x86_64/build or /lib/modules/4.20.5-250.vanilla.knurd.1.fc29.x86_64/source.
```
so frustrated that i can't fix this!
```
$ ll /lib/modules/4.20.5-250.vanilla.knurd.1.fc29.x86_64 
total 17M
drwxr-xr-x.  6 root 4.0K Jan 31 19:21 .
drwxr-xr-x.  5 root 4.0K Jan 31 19:20 ..
-rw-r--r--.  1 root  318 Jan 26 23:00 bls.conf
lrwxrwxrwx.  1 root   55 Jan 26 23:00 build -> /usr/src/kernels/4.20.5-250.vanilla.knurd.1.fc29.x86_64
-rw-r--r--.  1 root 197K Jan 26 23:00 config
drwxr-xr-x.  5 root 4.0K Jan 31 19:20 extra
drwxr-xr-x. 14 root 4.0K Jan 31 19:20 kernel
-rw-r--r--.  1 root 1.1M Jan 31 19:21 modules.alias
-rw-r--r--.  1 root 1.1M Jan 31 19:21 modules.alias.bin
-rw-r--r--.  1 root 1.7K Jan 26 23:00 modules.block
-rw-r--r--.  1 root 8.2K Jan 26 23:00 modules.builtin
-rw-r--r--.  1 root  11K Jan 31 19:21 modules.builtin.bin
-rw-r--r--.  1 root 397K Jan 31 19:21 modules.dep
-rw-r--r--.  1 root 548K Jan 31 19:21 modules.dep.bin
-rw-r--r--.  1 root  381 Jan 31 19:21 modules.devname
-rw-r--r--.  1 root  153 Jan 26 23:00 modules.drm
-rw-r--r--.  1 root   69 Jan 26 23:00 modules.modesetting
-rw-r--r--.  1 root 2.7K Jan 26 23:00 modules.networking
-rw-r--r--.  1 root 134K Jan 26 23:00 modules.order
-rw-r--r--.  1 root  562 Jan 31 19:21 modules.softdep
-rw-r--r--.  1 root 446K Jan 31 19:21 modules.symbols
-rw-r--r--.  1 root 545K Jan 31 19:21 modules.symbols.bin
lrwxrwxrwx.  1 root    5 Jan 26 23:00 source -> build
-rw-------.  1 root 4.0M Jan 26 23:00 System.map
drwxr-xr-x.  2 root 4.0K Jan 26 22:58 updates
drwxr-xr-x.  2 root 4.0K Jan 31 19:20 vdso
-rwxr-xr-x.  1 root 8.4M Jan 26 23:00 vmlinuz
-rw-r--r--.  1 root  183 Jan 26 22:56 .vmlinuz.hmac
```

pad92, 2019-01-31:
Hello,

same issue with `displaylink-1.5.1-2.x86_64` with kernel `4.20.4-200.fc29.x86_64` on fedora 29

```
systemctl status displaylink.service       
● displaylink.service - DisplayLink Manager Service
   Loaded: loaded (/usr/lib/systemd/system/displaylink.service; static; vendor preset: disabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2019-01-31 09:19:11 CET; 2s ago
  Process: 18529 ExecStartPre=/sbin/modprobe evdi (code=exited, status=1/FAILURE)
```

```
sudo dkms build evdi/1.5.1                                                                                    1↵ 

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.20.4-200.fc29.x86_64 all INCLUDEDIR=/lib/modules/4.20.4-200.fc29.x86_64/build/include KVERSION=4.20.4-200.fc29.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.20.4-200.fc29.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.5.1/build/make.log for more information.
```

```
/var/lib/dkms/evdi/1.5.1/build/evdi_fb.c: In function ‘evdifb_create’:
/var/lib/dkms/evdi/1.5.1/build/evdi_fb.c:370:33: error: ‘FBINFO_CAN_FORCE_OUTPUT’ undeclared (first use in this function)
  info->flags = FBINFO_DEFAULT | FBINFO_CAN_FORCE_OUTPUT;
                                 ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/evdi/1.5.1/build/evdi_fb.c:370:33: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:291: /var/lib/dkms/evdi/1.5.1/build/evdi_fb.o] Error 1
make[2]: *** Waiting for unfinished jobs....
   ./tools/objtool/objtool orc generate  --module --no-fp --retpoline "/var/lib/dkms/evdi/1.5.1/build/evdi_main.o";
   ./tools/objtool/objtool orc generate  --module --no-fp --retpoline "/var/lib/dkms/evdi/1.5.1/build/evdi_stats.o";
   ./tools/objtool/objtool orc generate  --module --no-fp --retpoline "/var/lib/dkms/evdi/1.5.1/build/evdi_gem.o";
make[1]: *** [Makefile:1564: _module_/var/lib/dkms/evdi/1.5.1/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.20.4-200.fc29.x86_64'
make: *** [Makefile:22: all] Error 2
```

beshad, 2019-01-31:
issue seems to be with DKMS service in 4.20 and not Displaylink/evdi related at all, imy DKMS service is off and i can't start so no build for evdi

beshad, 2019-02-01:
uninstalled both displaylink and dkms, did a fresh install of dkms, unable to start the service:
```
$ sudo systemctl status dkms.service
● dkms.service - Builds and install new kernel modules through DKMS
   Loaded: loaded (/usr/lib/systemd/system/dkms.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2019-02-01 18:13:41 NZDT; 5min ago
     Docs: man:dkms(8)
  Process: 19959 ExecStart=/bin/sh -c dkms autoinstall --verbose --kernelver $(uname -r) (code=exited, status=4)
 Main PID: 19959 (code=exited, status=4)

Feb 01 18:13:41 fedora-xps18 systemd[1]: Starting Builds and install new kernel modules through DKMS...
Feb 01 18:13:41 fedora-xps18 sh[19959]: Error! Could not locate dkms.conf file.
Feb 01 18:13:41 fedora-xps18 sh[19959]: File: /var/lib/dkms/evdi/1.5.0/source/dkms.conf does not exist.
Feb 01 18:13:41 fedora-xps18 systemd[1]: dkms.service: Main process exited, code=exited, status=4/NOPERMISSION
Feb 01 18:13:41 fedora-xps18 systemd[1]: dkms.service: Failed with result 'exit-code'.
Feb 01 18:13:41 fedora-xps18 systemd[1]: Failed to start Builds and install new kernel modules through DKMS.

```

beshad, 2019-02-01:
Finally solved it, I think the combination of all the helps here plus some external Googling helped solving my problem, started by running `rpm -q kernel` & `dnf info kernel-headers` and then used `dnf remove` to uninstall any extra kernels from those lists. then ran `sudo dnf clean all && sudo dnf upgrade && reboot`. 
after that i managed to do `sudo dkms build evdi/1.5.1` and the `dkms install` next successfully. restarted dkms.service & displaylink.service and got my USB display lighting up. of course I had already followed #78 instructions. kernel is 4.20.5-200.fc29.x86_64 with exact matching headers, core and modules. Pleased that it worked out at the end but to be honest a wee bit disappointed with Fedora in general (I'm new to Fedora, coming from Debian/Ubuntu) that i had to spend so much time solving a problem which worked out of the box for others. anyways thank you so much guys for helping out :)

closed 2019-02-01 by beshad

nicobao, 2020-08-26:
Hi,
I had the following issues:
- two screens not detected
- reboot was freezing when connected to the docking station via usb

I run Fedora 32, had installed the latest version of DisplayLink drivers via the GitHub page, and had installed the latest kernel headers.
I had secure boot disabled in my BIOS and was running X11.

I also had the following error message in my journalctl:
`Jan 26 12:59:47 fedora-xps18 modprobe[5096]: modprobe: FATAL: Module evdi not found in directory /lib/modules/4.19.4-300.fc29.x86_64`
Figured it was because I needed to install evdi manually.
I cloned this repo: https://github.com/DisplayLink/evdi, cd on the directory, `make` and `sudo make install` and the magic happens, it works.

itec docking station documentation is lacking information about the necessity to download and install the evdi library manually:
https://i-tec.cz/fr/produkt/u3hdmidvidock-4/