displaylink-rpm/displaylink-rpm #81: Package evdi as an akmod

ssaavedra, 2019-02-04:
Add evdi as a different package for future contribution to RPMFusion. Related to old issue #15.

ffgiff, 2019-05-30:
@ssaavedra , I've started work on this in https://github.com/ffgiff/displaylink-rpm/tree/81-package-as-akmod. There are two working spec files to generate a package for the (a)kmod and a package for the libs. Still to be done is building from the Makefile instead of doing it all at the command line, and un-hard-coding evdi versions.
The libs .spec is largely based on the displaylink.spec as you'd expect, and the (a)kmod .spec is largely based on the nvidia-kmod one from rpmfusion.

ssaavedra, 2019-06-05:
Shouldn't the package name for this be an `akmod-evdi`? Usually kmods are installed as `kmod-evdi-$(uname -r)`

I don't currently have a rpmbuild testing machine handy, but this is great progress! I'm not much available this month, but I'll try to take a look next week.

ffgiff, 2019-06-07:
> Shouldn't the package name for this be an `akmod-evdi`? Usually kmods are installed as `kmod-evdi-$(uname -r)`

I'm following the example from the nvidia kmod. The way akmod works you end up with a `kmod-evdi-$(uname -r)` installed as you would expect, i.e. like this:
```bash
$ dnf list installed | grep evdi
akmod-evdi.x86_64                                 1.6.1-1.fc29                           @@System               
evdi-kmod-common.x86_64                           1.6.1-1.fc29                           @@System               
kmod-evdi-5.0.17-200.fc29.x86_64.x86_64           1.6.1-1.fc29                           @@commandline          
kmod-evdi-5.0.19-200.fc29.x86_64.x86_64           1.6.1-1.fc29                           @@commandline
```

> I don't currently have a rpmbuild testing machine handy, but this is great progress! I'm not much available this month, but I'll try to take a look next week.
That's fine, I'm busy too, but might have it building from the Makefile soon.

FWIW I'm building in a container from the fedora image, you could try that if you have a machine with podman/docker. One effect of this is that I'm not sure that the displaylink.spec needs the dependency on gcc-c++; it's not needed by my akmod spec files, which build fine without gcc-c++ installed in my build container (but this means I can't build the displaylink.spec in there).