displaylink-rpm/displaylink-rpm #83: Update packaging for kernel 4.20

ssaavedra, 2019-02-19:
Last version of Displaylink (5.1) should support kernel version 4.20 which was the issue at #78.

An update of evdi and DisplayLink is needed to generate the new version. Beware that the new version does not work with kernel < 4.2, so that needs to be added to the `Requires:`.

- [ ] Update evdi and DisplayLink on Makefile
- [ ] Update evdi and DisplayLink on CI build
- [ ] Update kernel requirement version in specfile

elguero, 2019-02-19:
I just saw that this issue was created after I submitted a pull request.

I think the kernel requirement in the specfile is already set for kernels > than 4.7.  Unless I am missing something?

ssaavedra, 2019-02-20:
That was probably me being cautious some time ago. Anyway, for recent Fedora versions 4.7 is going to be fine, let's keep it like that.

CentOS support is a problem anyway since they are still shipping 3.x kernels, but that is a different topic :-)

Since the version is now tagged and Travis just finished building the artifacts, this issue can now be closed :tada:

closed 2019-02-20 by ssaavedra

sorvani, 2019-02-22:
Clean install of Fedora 29 + Cinnamon and then installed the current 1.6 release and it all works good. THanks for this.