displaylink-rpm/displaylink-rpm #85: Add proper RHEL7/Centos7 support

ssaavedra, 2019-02-28:
It is a long standing issue (#13) that we don't properly support Centos/RHEL.

There is already some guidance in that thread about how we could fix Centos support for the next release. I'm pushing this as an issue if anybody wants to pitch in.

thechitowncubs, 2019-04-09:
Ya, that would be nice.

rbarakjsonar, 2020-06-23:
> It is a long standing issue (#13) that we don't properly support Centos/RHEL.
> 
> There is already some guidance in that thread about how we could fix Centos support for the next release. I'm pushing this as an issue if anybody wants to pitch in.

Could you point me to the guidance, and other resources, that would enable me to assess if I could tackle adding support for DisplayLink to CentOS 7?

rbarakjsonar, 2020-06-23:
Could you point me to the guidance, and other resources, that would enable me to assess if I could tackle adding support for DisplayLink to CentOS 7?

closed 2024-05-16 by elguero