displaylink-rpm/displaylink-rpm #86: Intermittent "Disable of device-initiated U1/U2 failed" causes screens to disconnect

peterjanes, 2019-03-12:
I believe I have a similar configuration as the original reporter of #52 (Dell Latitude 7490 laptop with two Dell monitors connected via a Dell D6000 docking station) and am seeing the same behaviour reported there (particular [this comment](https://github.com/displaylink-rpm/displaylink-rpm/issues/52#issuecomment-364023360_)) with similar logs. Using displaylink-1.6.0-1.x86_64 compiled from tag v5.1.26.

ssaavedra, 2019-03-13:
This is likely an upstream issue on DisplayLink, not a packaging issue. If the displaylinkmanager binary is outputting problems in the log, it's likely upstream, have you tried the DisplayLink forum?

peterjanes, 2019-03-13:
Thought that might be the case, but as I saw there was a previous issue here with some discussion I thought I'd start at the distribution-specific level. For what it's worth, there aren't any logs from the displaylink binary itself, just kernel messages about the hub's USB interface and then fallout from the failures.

peterjanes, 2019-03-14:
I was able to find this page, which solves the problem for me. https://displaylink.org/forum/showpost.php?p=84983
Agreed that the root cause isn't packaging; might be worth including these rules if it's a common problem with this dock (and I could try to provide a patch if you'd like), but it's fine by me if you want to just close this issue.

ssaavedra, 2019-03-21:
According to this other post, https://displaylink.org/forum/showpost.php?p=85116&postcount=19 the change on the pulseaudio driver should be enough.

I would be happy to add power management rules to udev if they are indeed useful, but if the pulseaudio workaround works then it's not udev's fault.

My first impulse would be not adding the PulseAudio workaround to the packaging because it will force the FLOSS part of the stack to consume more energy, when this should be fixed by DisplayLink by properly  marking the device tree as in-use.

At least we should document it on the README in a "Hardware-specific behavior" section or something like that, but "fixing" it by adding a power management workaround I'm not sure if it's the right decision, and I'm open to further input.

ssaavedra, 2019-03-21:
This answer may also provide another workaround, if you want to test it https://askubuntu.com/questions/185274/how-can-i-disable-usb-autosuspend-for-a-specific-device

peterjanes, 2019-03-21:
I blacklisted just about everything with no luck. However, the only-PA workaround does seem to be sufficient (hence using it in the new doc section).

ssaavedra, 2019-03-25:
I just merged both of your contributions (notice I just added a small change to the README to account for future possible issues with other devices; and I added you to the changelog).

I'm closing this, but **if anyone gets a better way to keep this device awake, reopen or create a new issue**, and we can consider putting it as a patch if it does not badly impact other hardware.

Thanks for your contribution!

closed 2019-03-25 by ssaavedra