displaylink-rpm/displaylink-rpm #89: Blackscreen after GRUB (Fedora x86-64 kernel 5.0.3-200)

elegos, 2019-03-29:
Hello!

I'm definitively running into a problem which I don't know if it's Fedora's or DisplayLink's fault. The problem is that I've got a black screen and nothing else after GRUB, or "mostly" after the update to kernel 5.0.3-200 (yes I tried updating displaylink-rpm to version 5.1.26-2):

1. select the 5.0.3-200 kernel
2. the DELL logo appears on my laptop's display (as usual)
3. black screen

I've removed quiet and added nomodeset to the kernel entry and now I'm able to unlock the hard drive, then the kernel shows all the services starting and then... nothing, it hangs there indefinitely.

As I found no issues on Fedora, I suspected it might have something to do with displaylink.

May you please help me debugging this problem? For now I'm able to run kernel 4.20 with no errors.

Follows the list of hardware and loaded (4.20) kernel modules, if it might help.

Thanks :)

```bash
lspci

00:00.0 Host bridge: Intel Corporation 8th Gen Core Processor Host Bridge/DRAM Registers (rev 07)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) (rev 07)
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 630 (Mobile)
00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 07)
00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th Gen Core Processor Gaussian Mixture Model
00:12.0 Signal processing controller: Intel Corporation Cannon Lake PCH Thermal Controller (rev 10)
00:14.0 USB controller: Intel Corporation Cannon Lake PCH USB 3.1 xHCI Host Controller (rev 10)
00:14.2 RAM memory: Intel Corporation Cannon Lake PCH Shared SRAM (rev 10)
00:15.0 Serial bus controller [0c80]: Intel Corporation Cannon Lake PCH Serial IO I2C Controller #0 (rev 10)
00:15.1 Serial bus controller [0c80]: Intel Corporation Cannon Lake PCH Serial IO I2C Controller #1 (rev 10)
00:16.0 Communication controller: Intel Corporation Cannon Lake PCH HECI Controller (rev 10)
00:17.0 SATA controller: Intel Corporation Cannon Lake Mobile PCH SATA AHCI Controller (rev 10)
00:1b.0 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #17 (rev f0)
00:1c.0 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #1 (rev f0)
00:1c.4 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #5 (rev f0)
00:1d.0 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #9 (rev f0)
00:1f.0 ISA bridge: Intel Corporation Device a30e (rev 10)
00:1f.3 Audio device: Intel Corporation Cannon Lake PCH cAVS (rev 10)
00:1f.4 SMBus: Intel Corporation Cannon Lake PCH SMBus Controller (rev 10)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Cannon Lake PCH SPI Controller (rev 10)
01:00.0 3D controller: NVIDIA Corporation GP107M [GeForce GTX 1050 Ti Mobile] (rev a1)
3b:00.0 Network controller: Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter (rev 32)
3c:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS525A PCI Express Card Reader (rev 01)
3d:00.0 Non-Volatile memory controller: SK hynix Device 1527```

```bash
lsusb

Bus 002 Device 005: ID 17e9:436e DisplayLink 
Bus 002 Device 004: ID 2109:0813 VIA Labs, Inc. 
Bus 002 Device 003: ID 17e9:436e DisplayLink 
Bus 002 Device 002: ID 2109:0813 VIA Labs, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 27c6:5395  
Bus 001 Device 003: ID 0cf3:e300 Qualcomm Atheros Communications 
Bus 001 Device 006: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 004: ID 2109:2813 VIA Labs, Inc. 
Bus 001 Device 002: ID 2109:2813 VIA Labs, Inc. 
Bus 001 Device 007: ID 0c45:671d Microdia 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub```

```bash
lsmod

Module                  Size  Used by
ccm                    20480  6
fuse                  135168  3
ipt_MASQUERADE         20480  3
nf_conntrack_netlink    53248  0
[... 114 lines cut ...]
i915                 2121728  44
kvmgt                  32768  0
mdev                   24576  1 kvmgt
vfio                   36864  1 kvmgt
kvm                   741376  62 kvmgt,kvm_intel
irqbypass              16384  1 kvm
i2c_algo_bit           16384  1 i915
drm_kms_helper        204800  2 evdi,i915
rtsx_pci_sdmmc         32768  0
mmc_core              176128  1 rtsx_pci_sdmmc
crct10dif_pclmul       16384  1
crc32_pclmul           16384  0
mxm_wmi                16384  0
crc32c_intel           24576  5
drm                   495616  24 drm_kms_helper,evdi,i915
nvme                   40960  3
ghash_clmulni_intel    16384  0
nvme_core              86016  5 nvme
serio_raw              20480  0
rtsx_pci               77824  2 rtsx_pci_sdmmc,rtsx_pci_ms
i2c_hid                32768  0
pinctrl_cannonlake     36864  0
wmi                    32768  6 intel_wmi_thunderbolt,dell_wmi,wmi_bmof,dell_smbios,dell_wmi_descriptor,mxm_wmi
video                  49152  3 dell_wmi,dell_laptop,i915
pinctrl_intel          28672  1 pinctrl_cannonlake```

pbaitule, 2019-04-10:
@elegos I'm assuming your external monitors don't display anything when booting using the newer kernel. If you do an `ls -al /var/lib/dkms/evdi/[1.5.1 or 1.6.0]/`, do you see your kernel version as a directory?

If not, do a `sudo dkms install evdi/[1.5.1 or 1.6.0]` once you have booted using the newer kernel version.

elegos, 2019-04-10:
@pbaitule resolved this morning patching the kernel (actually the source of kernel 5.0.7 for Fedora 29 is already patched, waiting for a release). It turned out it was a kernel 5.x bug with certain series 8000 intel integrated GPU.

https://bugs.freedesktop.org/show_bug.cgi?id=109959

closed 2019-04-10 by elegos