displaylink-rpm/displaylink-rpm #101: RHEL8/CentOS 8 support

JonasKs, 2019-08-13:
Any plans on supporting RHEL8, or does the #13 work arounds work in RHEL8 maybe?

ssaavedra, 2019-08-28:
Disclaimer: no "support" is given, nor should be expected. This is just a community effort that tries to package the DisplayLink driver in a way more amenable to rpm-based systems.

I am completely open to accepting patches to CentOS 8 (or even CentOS 7, which is a long-standing open issue too), if you have the time to debug what's happening (in terms of needed dependencies, perhaps), and I'm sure someone could review the PR, if you can come up with one.

However, me in particular, I am no longer using DisplayLink-enabled hardware (nor I'm using a CentOS system) to have an incentive to do this research by myself, but I could review changes authored by others.

I'm leaving this open in case you have the time and wants to communicate further :-)

JonasKs, 2019-09-01:
No worries mate, makes perfect sense. It’s not super critical for me, but if I go back to a DisplayLink adapter I’ll debug more and re open.

closed 2019-09-01 by JonasKs