displaylink-rpm/displaylink-rpm #103: USB devices on DisplayLink D3100 Dell not always working

baverhey, 2019-09-05:
fedora 30 workstation
USB devices work directly on the laptop dell Latitude 5590
But most times 70% not when insert in the DisplayLink pre boot. and certainly not when adding them after boot.

Screens work ;-)
 
USB devices:
Kensington Orbit
Logitech headset G430
Logitech M570 trackball

lsusb
Bus 002 Device 005: ID 17e9:436e DisplayLink USB3.0 Hub             
Bus 002 Device 004: ID 2109:0812 VIA Labs, Inc. VL812 Hub
Bus 002 Device 003: ID 17e9:436e DisplayLink 
Bus 002 Device 002: ID 2109:0812 VIA Labs, Inc. VL812 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 007: ID 8087:0a2b Intel Corp. 
Bus 001 Device 005: ID 0bda:5650 Realtek Semiconductor Corp. Integrated Webcam_HD
Bus 001 Device 003: ID 046d:0a4d Logitech, Inc. G430 Surround Sound Gaming Headset
Bus 001 Device 002: ID 2109:2812 VIA Labs, Inc. VL812 Hub
Bus 001 Device 010: ID 0a5c:5832 Broadcom Corp. 5880
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

The headset  G430 and  Logitech, Inc. Unifying Receiver  is now plugged  in in the laptop.

Booting with the usb devices in Display link helps, a few boots needed.

please advice
Kind regards,

vyscond, 2019-09-22:
I've tried this method on my ThinkPad L380 but it didn't had any effect. I can see the device through `lsusb`:
```
Bus 002 Device 003: ID 17e9:436f DisplayLink 
Bus 002 Device 002: ID 2109:0210 VIA Labs, Inc. USB3.0 Hub             
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 06cb:00a2 Synaptics, Inc. 
Bus 001 Device 005: ID 8087:0a2b Intel Corp. 
Bus 001 Device 003: ID 5986:2113 Acer, Inc Integrated Camera
Bus 001 Device 004: ID 1532:0035 Razer USA, Ltd USB2.0 Hub             
Bus 001 Device 002: ID 2109:2210 VIA Labs, Inc. USB2.0 Hub             
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
But can't use the video output ports (mouse works though).

elguero, 2020-05-14:
@baverhey Were you able to resolve this at all?

closed 2020-10-26 by elguero