displaylink-rpm/displaylink-rpm #104: Wrong Kernel Devel installed

vyscond, 2019-09-22:
I've installed the `fedora-30-displaylink-1.6.2-2.x86_64.rpm` using the dnfdragora Package Installer. The installation process was fine. The lsusb was able to identity the device as well but the HDMI nor VGA output were working.

After a few rounds looking up the closed issues i realised that this package install a .service file that wasn't mention on the README. I tried using systemctl to see the status it was saying that something was not right: `systemctl status displaylink.service`

Since the two main commands to pull the logs were producing nothing to clarify what was the issue i start looking at the /var/log folder. There i was also able to find this log file: `var/log/displaylink/displaylink.log` (also not explicitly mentioned on the README but very helpful to debug).

There i was able to recover this:

```
Creating symlink /var/lib/dkms/evdi/1.6.2/source ->
                 /usr/src/evdi-1.6.2

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.2.14-200.fc30.x86_64 cannot be found at
/lib/modules/5.2.14-200.fc30.x86_64/build or /lib/modules/5.2.14-200.fc30.x86_64/source.
Warning: I do not know how to handle 5.0.9-301.fc30.x86_64.
Error! echo
Your kernel headers for kernel 5.2.14-200.fc30.x86_64 cannot be found at
/lib/modules/5.2.14-200.fc30.x86_64/build or /lib/modules/5.2.14-200.fc30.x86_64/source.

------------------------------
Deleting module version: 1.6.2
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/evdi/1.6.2/source ->
                 /usr/src/evdi-1.6.2

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.2.14-200.fc30.x86_64 cannot be found at
/lib/modules/5.2.14-200.fc30.x86_64/build or /lib/modules/5.2.14-200.fc30.x86_64/source.
```

After that i was starting to look up the paths tha message said they were missing:
```
$ ls /lib/modules/5.2.14-200.fc30.x86_64/build 
lrwxrwxrwx. 1 root 39 Sep 10 09:32 /lib/modules/5.2.14-200.fc30.x86_64/build -> /usr/src/kernels/5.2.14-200.fc30.x86_64
```

and

```
$ ls /lib/modules/5.2.14-200.fc30.x86_64/source 
lrwxrwxrwx. 1 root 5 Sep 10 09:31 /lib/modules/5.2.14-200.fc30.x86_64/source -> build
```

Links seemed to be fine. Then i proceed to checkout the pointed path:
```
$ sudo ls /usr/src/kernels/
5.2.15-200.fc30.x86_64
```
And that's our problem. We only have the source for 5.2.15 instead of having 5.2.14. So i decided to trying to re-install the rpm and weirdly i got this on the prompt:

```
$ sudo dnf remove displaylink
Dependencies resolved.
=================================================================================================================================================================================================================================
 Package                                                      Architecture                                  Version                                                   Repository                                            Size
=================================================================================================================================================================================================================================
Removing:
 displaylink                                                  x86_64                                        1.6.2-2                                                   @@commandline                                         10 M
Removing unused dependencies:
 dkms                                                         noarch                                        2.6.1-3.fc30                                              @fedora                                              219 k
 elfutils-libelf-devel                                        x86_64                                        0.177-1.fc30                                              @updates                                              32 k
 kernel-devel                                                 x86_64                                        5.2.15-200.fc30                                           @updates                                              53 M
 zlib-devel                                                   x86_64                                        1.2.11-18.fc30                                            @updates                                             138 k

Transaction Summary
=================================================================================================================================================================================================================================
Remove  5 Packages

Freed space: 64 M
Is this ok [y/N]:
```

I was expecting to see an attempt to remove the version 5.2.14 and not the 5.2.15. So maybe the RPM is expecting 5.2.14 but somehow is resolving to 5.2.15?

alexberry, 2019-11-01:
I doubt it's going to make a difference, as dragora seems to just be an interface for dnf, but I find the following does the job fine for me when installing displaylink (including it's dependencies)

```
#!/bin/bash

# Set this to your fedora version, so long as it is available in displaylink-rpm's releases
fedoraver=30
arch=x86_64

# If not installed, install jq and curl to parse the json releases feed on github
for dep in curl jq
do
  if ! which ${dep} > /dev/null 2>&1 
  then
    dnf install ${dep}
  fi
done

# Look up the url that contains a list of assets/releases for this project
assetsurl=$(curl -s https://api.github.com/repos/displaylink-rpm/displaylink-rpm/releases/latest | jq -r .assets_url)

# Lookup up the url of the latest matching rpm listed in assets

rpmurl=$(curl -s ${assetsurl} | jq -r '.[].browser_download_url' | grep ${fedoraver} | grep ${arch})

# Install the latest version of displaylink by looking it up on github's release feed
dnf install ${rpmurl}
```

elguero, 2020-02-10:
Perhaps you need to clear out what DKMS has stored.

I would suggest uninstalling the RPM.  Clear out /var/lib/dkms/evdi and then install the RPM package.

vyscond, 2020-04-15:
Tried again on clean setup:

- Running `sudo dnf install fedora-31-displaylink-1.6.4-1.x86_64.rpm `

```
Last metadata expiration check: 0:11:05 ago on Wed 15 Apr 2020 02:00:14 PM -03.
Dependencies resolved.
==================================================================================================================================================================================================================================================================================
 Package                                                               Architecture                                           Version                                                                          Repository                                                    Size
==================================================================================================================================================================================================================================================================================
Installing:
 displaylink                                                           x86_64                                                 1.6.4-1                                                                          @commandline                                                 3.6 M
Installing dependencies:
 dkms                                                                  noarch                                                 2.8.1-4.20200214git5ca628c.fc31                                                  updates                                                       76 k
 elfutils-libelf-devel                                                 x86_64                                                 0.179-1.fc31                                                                     updates                                                       27 k
 kernel-devel                                                          x86_64                                                 5.5.15-200.fc31                                                                  updates                                                       13 M

Transaction Summary
==================================================================================================================================================================================================================================================================================
Install  4 Packages

Total size: 17 M
Total download size: 13 M
Installed size: 60 M
Is this ok [y/N]: y
Downloading Packages:
(1/3): dkms-2.8.1-4.20200214git5ca628c.fc31.noarch.rpm                                                                                                                                                                                             75 kB/s |  76 kB     00:01    
(2/3): elfutils-libelf-devel-0.179-1.fc31.x86_64.rpm                                                                                                                                                                                               17 kB/s |  27 kB     00:01    
(3/3): kernel-devel-5.5.15-200.fc31.x86_64.rpm                                                                                                                                                                                                    667 kB/s |  13 MB     00:19    
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Total                                                                                                                                                                                                                                             561 kB/s |  13 MB     00:23     
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                                                                                                                                                                                          1/1 
  Installing       : kernel-devel-5.5.15-200.fc31.x86_64                                                                                                                                                                                                                      1/4 
  Running scriptlet: kernel-devel-5.5.15-200.fc31.x86_64                                                                                                                                                                                                                      1/4 
  Installing       : elfutils-libelf-devel-0.179-1.fc31.x86_64                                                                                                                                                                                                                2/4 
  Installing       : dkms-2.8.1-4.20200214git5ca628c.fc31.noarch                                                                                                                                                                                                              3/4 
[... cut ...]
Warning: I do not know how to handle 5.5.11-200.fc31.x86_64.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....(bad exit status: 2)
make -j8 KERNELRELEASE=5.5.15-200.fc31.x86_64 all INCLUDEDIR=/lib/modules/5.5.15-200.fc31.x86_64/build/include KVERSION=5.5.15-200.fc31.x86_64 DKMS_BUILD=1....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.5.15-200.fc31.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.6.4/build/make.log for more information.
```

elguero, 2020-04-16:
@vyscond It looks like this is a different issue.  The version you are trying to install does not work with kernel 5.5.

Please see https://github.com/displaylink-rpm/displaylink-rpm/issues/119#issuecomment-592207142

closed 2020-04-16 by elguero