displaylink-rpm/displaylink-rpm #112: Kernel 5.3.12

mattdioso, 2019-12-02:
Has anyone noticed any issues with running this on kernel 5.3.12-300.fc31.x86_64? 

I recently upgraded from Fedora 30 to 31 and was forced to this kernel version. I tried installing from source and from rpm successfully and got it to recognize all of my screens but once I started to change the display settings, everything would freeze. My screens froze and I couldn't move the mouse.

The same behavior would sometimes happen right when I plug in my dock.

I tried downgrading to kernel 5.3.11 (which seemed to be the kernel when the last update was put out) but dkms forces me to stay on 5.3.12.

vischlum, 2019-12-02:
Are you using X11 or Wayland? Cause I have the same issue (freezing when changing the screen layout) with Wayland, but not with X11. You can check with `echo $XDG_SESSION_TYPE`.

trevh86, 2019-12-02:
I have the same issue on Wayland too. I'm running a Dell Lattitude 7490 if that's relevant? I presume this is a niche problem by the lack of exposure it seems to have?

I also switched to X11 but it crashed on me randomly sometimes too.

mattdioso, 2019-12-02:
I disabled Wayland in `/etc/gdm/custom.conf` and setting `WaylandEnable=false` and saw better results and was actually able to change my display settings but saw a lot of lag on the external monitors but not on my laptop.

hadarsharon, 2019-12-02:
I am running Fedora 31 with kernel version 5.3.13 on X11 (haven't tried Wayland for a while now actually), also have set WaylandEnable=false and can use the external monitors OK when the laptop monitor is also turned on -- however once I turn off my laptop monitor I start experiencing horrible lag on the external monitors (until I re-enable the laptop monitor, then everything is back to normal).
Needless to say I have never experienced it before upgrading from Fedora 30 to Fedora 31.

elguero, 2019-12-05:
Just to chime in, I have been experiencing lag on external monitors ever since upgrade to Fedora 31.

linuxcrash, 2019-12-06:
I can report the same behaviour. Running on latest updated Fedora 31 with Kernel 5.3.14-300 using the rawhide version of displaylink (updated 6.12.2019) and suddenly I have a massive lag on my external monitors in X11 sessions.
Additionally if I logout from the X11 session my system completely freezes and I have to hard power reset it (pressing the power button for 10s). Randomly it also just crashes the X11 session and then when restarting X11 it also freezes the same way.

I did not have this issues before in Fedora 30.

My hardware is an Dell XPS 15 9550 using the Displaylink D3000 USB 3 dock with two 27" Dell monitors.

fran-pastor, 2019-12-10:
> I am running Fedora 31 with kernel version 5.3.13 on X11 (haven't tried Wayland for a while now actually), also have set WaylandEnable=false and can use the external monitors OK when the laptop [...]

exactly the same issue for me
Fresh Fedora 31 Install
Kernel 5.3.15
Dell Latitude 5590

lix0st, 2019-12-17:
> I am running Fedora 31 with kernel version 5.3.13 on X11 (haven't tried Wayland for a while now actually), also have set WaylandEnable=false and can use the external monitors OK when the laptop [...]

I have exactly the same issue. But for me it freezes on X11 with and without the laptop monitor enabled.

Fedora 31 (latest updates)
Kernel: 5.3.16-300
Lenovo X1 Yoga (4th generation)
Dock: ThinkPad Hybrid USB-C with USB-A

richardfearn, 2019-12-28:
Same issue here... Wayland session, Fedora 31, kernel 5.3.16-300, on a ThinkPad T440s, connected to a Dell D6000 and two external monitors.

When I log in, the laptop's built-in display and the two external monitors all seem to be working. But when I try to turn off the built-in display using GNOME Settings, everything freezes.

I saw this in the journal:

```
Dec 27 23:00:26 thinkpad gnome-shell[3813]: *** BUG ***
Dec 27 23:00:26 thinkpad gnome-shell[3813]: In pixman_region32_init_rect: Invalid rectangle passed
Dec 27 23:00:26 thinkpad gnome-shell[3813]: Set a breakpoint on '_pixman_log_error' to debug
Dec 27 23:00:26 thinkpad gnome-shell[3813]: *** BUG ***
Dec 27 23:00:26 thinkpad gnome-shell[3813]: In pixman_region32_init_rect: Invalid rectangle passed
Dec 27 23:00:26 thinkpad gnome-shell[3813]: Set a breakpoint on '_pixman_log_error' to debug
```

which led me to this:

https://gitlab.gnome.org/GNOME/gnome-shell/issues/1606

and this:

https://gitlab.gnome.org/GNOME/mutter/merge_requests/961

which give me some hope this might get fixed via a gnome-shell and/or mutter update fairly soon...

richardfearn, 2019-12-28:
Some discussion of the display settings issue on the DisplayLink Forum: https://displaylink.org/forum/showthread.php?p=89046

spyter, 2020-02-18:
I am having the same issues using x11 w/kernel 5.3 and 5.4. As soon as I disable my laptop screen to use the two monitors, it lags so bad it's unusable and I am forced to hard reboot. 

Has anyone found a work around yet?

PR #131 (2020-05-08): Fixes related to evdi devel branch changes
The changes are:
  * Updates related to the new evdi driver version 1.7.0.
  * Update to the released Displaylink driver 5.3.1.
  * The minimum kernel supported in evdi 1.7.0 is now 4.15. Adjusted the spec
    accordingly.
  * Remove Travis build for Fedora 27 since the kernel is no longer supported.
  * Add Fedora 32 build to Travis.
  * Fix support for DL-6xxx devices. The firmware image was not being copied from
    the DisplayLink driver package.
  * Adjust how we use dkms inside the rpm to follow recommended way outlined in the
    documentation. [... cut]
commit 4a9f425b (2020-05-08): Fixes related to evdi devel branch changes

The changes are:
  * Updates related to the new evdi driver version 1.7.0.
  * Update to the released Displaylink driver 5.3.1.
  * The minimum kernel supported in evdi 1.7.0 is now 4.15. Adjusted the spec
    accordingly.
  * Remove Travis build for Fedora 27 since the kernel is no longer supported.
  * Fix support for DL-6xxx devices. The firmware image was not being copied from
    the DisplayLink driver package.
  * Adjust how we use dkms inside the rpm to follow recommended way outlined in the
    documentation.

Resolves #112, resolves #121, re [... cut]
closed 2020-05-08 by elguero